iiab/iiab #1811: Nod-RED failed to `npm install` ("ERR! 409 Conflict - GET https://registry.npmjs.org/node-red-dashboard")

holta, 2019-07-03:
@m-anish are there ways to make error-prone `npm install` of Node-RED and similar more reliable in future?  Error details pasted in below.

Context: `./iiab-install --reinstall` was run on Raspbian Stretch Lite on RPi 3.  This is a BIG-sized IIAB 7.0 pre-release that was freshly installed just over a week ago.

Re-running Ansible indeed got past the problem, but these transient errors drive newbie installers quite crazy, preventing people from reliably/quickly getting to a working IIAB.

(So if further improvement is possible in future let's keep this in mind.  I have no idea if `yarn` is really that much better than `npm` as some people say!?)

> TASK [nodered : Globally 'npm install' 3 Node-RED packages: node-red, node-red-admin, node-red-dashboard] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["npm", "install", "-g", "--unsafe-perm", "node-red", "node-red-admin", "node-red-dashboard"], "delta": "0:00:07.032984", "end": "2019-07-02 20:32:43.733918", "msg": "non-zero return code", "rc": 1, "start": "2019-07-02 20:32:36.700934", "stderr": "npm ERR! code E409\nnpm ERR! 409 Conflict - GET https://registry.npmjs.org/node-red-dashboard\n\nnpm ERR! A complete log of this run can be found in:\nnpm ERR!     /root/.npm/_logs/2019-07-03T00_32_43_640Z-debug.log", "stderr_lines": ["npm ERR! code E409", "npm ERR! 409 Conflict - GET https://registry.npmjs.org/node-red-dashboard", "", "npm ERR! A complete log of this run can be found in:", "npm ERR!     /root/.npm/_logs/2019-07-03T00_32_43_640Z-debug.log"], "stdout": "", "stdout_lines": []}

> root@box:~# cat /root/.npm/_logs/2019-07-03T00_32_43_640Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'install',
1 verbose cli   '-g',
1 verbose cli   '--unsafe-perm',
1 verbose cli   'node-red',
1 verbose cli   'node-red-admin',
1 verbose cli   'node-red-dashboard' ]
2 info using npm@6.9.0
3 info using node@v10.16.0
4 verbose npm-session eab8cffe5d843e39
5 silly install loadCurrentTree
6 silly install readGlobalPackageData
7 http fetch GET 409 https://registry.npmjs.org/node-red-dashboard 656ms
8 silly fetchPackageMetaData error for node-red-dashboard@latest 409 Conflict - GET https://registry.npmjs.org/node-red-dashboard
9 http fetch GET 304 https://registry.npmjs.org/node-red-admin 891ms (from cache)
10 silly pacote tag manifest for node-red-admin@latest fetched in 940ms
11 http fetch GET 304 https://registry.npmjs.org/node-red 969ms (from cache)
12 silly pacote tag manifest for node-red@latest fetched in 1055ms
13 timing stage:rollbackFailedOptional Completed in 6ms
14 timing stage:runTopLevelLifecycles Completed in 1141ms
15 verbose stack Error: 409 Conflict - GET https://registry.npmjs.org/node-red-dashboard
15 verbose stack     at res.buffer.catch.then.body (/usr/lib/node_modules/npm/node_modules/npm-registry-fetch/check-response.js:104:15)
15 verbose stack     at process._tickCallback (internal/process/next_tick.js:68:7)
16 verbose statusCode 409
17 verbose pkgid node-red-dashboard@latest
18 verbose cwd /opt/iiab/iiab
19 verbose Linux 4.19.42-v7+
20 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "-g" "--unsafe-perm" "node-red" "node-red-admin" "node-red-dashboard"
21 verbose node v10.16.0
22 verbose npm  v6.9.0
23 error code E409
24 error 409 Conflict - GET https://registry.npmjs.org/node-red-dashboard
25 verbose exit [ 1, true ]

m-anish, 2019-07-08:
Okay. Can we shift this to 7.1. I don't think I can fix this within this week (or three) :)

holta, 2019-12-21:
Seems to have resolved itself over the past 5.5 months.

closed 2019-12-21 by holta