iiab/iiab #1812: Investigate @mitra42's suggestions re: severe recent slowdowns (which are related to 'internetarchive' ?)

holta, 2019-07-03:
This can be deferred to IIAB 7.1 if necessary, but several severe recent slowdowns in IIAB 7.0 pre-releases have been observed, that might possibly be related:

- MySQL (MariaDB 10.1.40) times out on every boot [never used to happen] #1802
- `yarn add @internetarchive/dweb-archive @internetarchive/dweb-mirror` can take 15+ minutes to install [used to by 5 minutes] https://github.com/iiab/iiab/pull/1806#issue-293486566
- See @mitra42's recommendations for monitoring these possible patterns here: https://github.com/iiab/iiab/pull/1780#issuecomment-507451425

holta, 2019-07-03:
From the user's perspective, slowness is a bug, so I'm escalating this one (whether we solve for IIAB 7.0 or 7.1!)

holta, 2019-08-09:
Tangentially, see also this list that @mitra42 is compiling during his profiling to uncover where slowdowns are happening during (re)installs of 'internetarchive' etc: (similar to https://github.com/iiab/iiab/issues/1915#issuecomment-517874297)

Package dependency warnings {META}
internetarchive/dweb-mirror#234

holta, 2019-08-14:
@mitra42 @jvonau please LMK if the mitigations implemented by PR #1920 #1921 #1923 #1927 are now sufficient to release IIAB 7.0, which needs to happen ASAP (Thanks!)

mitra42, 2019-08-14:
@holta  if you are seeing it install correctly then I don't see any reason to hold up 7.0

mitra42, 2019-10-04:
I think this can be closed now PR #1982  should significantly improve install speed.

closed 2019-10-04 by holta