iiab/iiab #1815: box.lan does not resolve at the command-line (IIAB installed on Debian 10 VM)

holta, 2019-07-03:
@jvonau do you have suggestions or ideas?

Live at 10.8.0.30 for now:

```
root@box:~# more /etc/iiab/iiab.env
# This is a configuration file for IIAB
# It can be sourced in a shell script or read into an application
IIAB_BASE_PATH=/opt/iiab
IIAB_DIR=/opt/iiab/iiab
IIAB_RELEASE=7.0
IIAB_REVISION=0
IIAB_GATEWAY_ENABLED=False
LAN_IP=172.18.96.1
OS=debian
OS_VER=debian-10
WWWROOT=/library/www/html
STAGE=9
FQDN=box.lan
IIAB_USB_LIB_SHOW_ALL=True
IIAB_WAN_DEVICE="enp0s3"
IIAB_LAN_DEVICE="none"
HOSTAPD_ENABLED=True
HTTPCACHE_ON=False
```

holta, 2019-07-03:
@jvonau writes:
> quick fix is to edit /etc/hosts...  as /etc/resolv.conf is pointing upstream..

@holta writes:
> What playbook should do this?

@jvonau writes:
> hostname.yml in 0-init

holta, 2019-07-03:
@jvonau has raised concerns about the existence of `ifupdown` on Debian 10 "Buster", wondering if this might break some aspects of `systemd` ?  See these Debian 10 dependencies:.

```
root@box:/home/iiab-admin# apt-cache depends ifupdown
ifupdown
  Depends: libc6
  Depends: lsb-base
  Depends: adduser
  Depends: iproute2
  Breaks: systemd
 |Recommends: isc-dhcp-client
  Recommends: <dhcp-client>
    dhcpcanon
    dhcpcd5
    pump
    isc-dhcp-client
  Suggests: ppp
  Suggests: rdnssd
  Replaces: systemd
```

ASIDE: `ifupdown` also exists on Ubuntu 18.04 (Desktop version) FWIW.  But does not exist on Ubuntu 18.04's Server version:

```
root@box:~# apt -a list ifupdown
Listing... Done
ifupdown/bionic-updates 0.8.17ubuntu1.1 amd64
ifupdown/bionic 0.8.17ubuntu1 amd64
```

holta, 2019-07-04:
Curiously, I don't believe Raspbian Buster is affected by these Debian 10 Buster issues?

(But @jvonau & others can correct me if I'm wrong!)

holta, 2019-07-07:
@jvonau is hoping to solve this in coming days — by comparing fresh "OS only" installs of Debian 10 Buster (released yesterday) and Raspbian Buster (released 2019-06-24) to see what is different.

(PR #1829 is only tangentially related, i.e. Debian 10 brings a few subtle networking changes.)

jvonau, 2019-07-08:
running iiab-network results in box.lan resolving via /etc/hosts.

jvonau, 2019-07-08:
hostname returns box.lan while ansible is returning localhost.localdomain when running runrole after the install clobbering box.lan:

TASK [0-init : Set iiab_fqdn: box.lan] ***************************************************************
ok: [127.0.0.1]

TASK [0-init : Set FQDN_changed when iiab_fqdn != ansible_fqdn (localhost.localdomain)] **************
ok: [127.0.0.1]

TASK [0-init : Set hostname if FQDN_changed] *********************************************************
included: /opt/iiab/iiab/roles/0-init/tasks/hostname.yml for 127.0.0.1
<snip>
TASK [0-init : Put hostnames "127.0.0.1 localhost.localdomain localhost box box" in /etc/hosts] ******
changed: [127.0.0.1]
<snip>
changed: [127.0.0.1] => (item={u'option': u'FQDN_changed', u'value': True})

Why is ansible returning "ansible_fqdn": "localhost.localdomain", 
during ansible -m setup -i ansible_hosts localhost --connection=local
and hostname returns box.lan?

jvonau, 2019-07-08:
answer is ansible is reading the first entry in /etc/hosts for ansible_fqdn so should the first entry be {{ iiab_hostname }}.{{ iiab_domain }}?

holta, 2019-07-08:
@jvonau fixed this in PR #1847 -- and as a bonus, amazingly this also fixes name resolution of `box.lan` and `box` starting a few seconds after IIAB's installer begins!

closed 2019-07-08 by holta