iiab/iiab #1821: @holta will this work on a pi 3b+?

kmw, 2019-07-05:
@holta thanks for the offer, but I can source a SD if need be :) I'm based in Cape Town, South Africa, btw.

I did have a slightly larger USB thumb drive available, so I installed on there and it worked, except for the tune2fs bit. Below are my notes:

1. Install debian buster on a USB thumb drive
1. Enable SSH (`sudo touch /boot/ssh`)
1. Run `curl d.iiab.io/install.txt | sudo bash`
1. Update password: busterf13ndly
1. Choose edit /etc/iiab/local_vars.yml
1. Change `kalite_install` and `kalite_enabled` to `False`
1. Run `sudo iiab`
1. System rebooted after updating OS
1. Run `sudo iiab` again

> Detected RPi: Lower reserve disk (SD card) space from ~5% to 1%
> 
> tune2fs 1.44.5 (15-Dec-2018)
> tune2fs: No such file or directory while trying to open /dev/mmcblk0p2
> Couldn't find valid filesystem superblock.

10. Change mmcblk0p2 to sda2 `sudo sed -i "s/mmcblk0p2/sda2/g" /usr/sbin/iiab`
1. Run `sudo iiab` again
1. Run until it got to

> INTERNET-IN-A-BOX (IIAB) SOFTWARE INSTALL IS COMPLETE
> 
> (1A) A couple minutes after you reboot (below) try to connect any laptop to
> Wi-Fi hotspot "Internet in a Box".  If this works, verify that you can browse
> to http://box or http://box.lan or http://172.18.96.1
>
> (1B) IF THOSE 3 DON'T WORK, try http://box.local from any device connected to
> your usual network.  Or try http://localhost from your IIAB itself!
>
> (1C) IF ALL 5 ABOVE DON'T WORK, ask the person who set up the network/router
> in your building for the IP address of your IIAB, so you can browse to it
> using (something like) http://192.168.0.100
>
> (2) ADD CONTENT using http://box.lan/admin (changing "box.lan" to be as above!)
> PLEASE READ "What are the default passwords?" and "How do I customize my
> Internet-in-a-Box home page?" at http://FAQ.IIAB.IO
>
> (3) If you're installing IIAB over Wi-Fi (instead of Ethernet) remember to run
> 'iiab-hotspot-on' at the VERY END, when you're ready to ACTIVATE YOUR IIAB's
> INTERNAL WI-FI HOTSPOT.  CAUTION: this permanently kills your IIAB's Internet
> (over Wi-Fi anyway) until you later run 'iiab-hotspot-off'.  CAVEAT: these
> two commands only work with Raspberry Pi as of 2019-06-27.
>
> HIT [Enter] TO REBOOT, CONFIRMING YOU'VE "photographed" THE ABOVE PARAGRAPHS: 

13. reboot on the pi 3
1. Connect to the IIAB access point - working!
1. Kill second partition of 4GB SD card I have
1. Delete all content from first partition (/boot)
1. Copy [bootcode.bin](https://raspberrypi.stackexchange.com/questions/62965/boot-zero-w-from-usb-device) to first partition to boot from USB
1. Boot Pi Zero W 1.3 from USB!!
1. Connect to IIAB access point
1. go to http://box.lan
1. http://box.lan/kiwix/ works! Only test content
1. http://box.lan/munin/ works!
1. http://box.lan/osm-vector-maps/maplist/ works! Only world map though
1. http://box.lan/usb/ shows actual USB content
1. http://box.lan/awstats/awstats.pl works. Take some time to load.
1. http://box.lan/admin/ works!
1. I didn't install kalite

_Originally posted by @dirkcuys in https://github.com/iiab/iiab/issues/1799#issuecomment-508701485_

holta, 2019-07-05:
@kmw is IIAB's stock installer (http://download.iiab.io) for RPi 3 B+ insufficient somehow?

What specifically are you trying to accomplish?

Borrowing from @dirkcuys #1799 in which area(s) ?

dirkcuys, 2019-07-06:
@Kennith

Do you mean to run iiab from a USB drive? Up to step 14 was on a pi 3b+. At
that point I had a working install booting from USB on a 3b+

holta, 2019-07-06:
@kmw please see the 43rd question among 48 here:

    "Can I install IIAB onto a USB drive?"
    http://FAQ.IIAB.IO

Am closing this ticket for now, but plz feel free to re-open it if that doesn't answer your question!

closed 2019-07-06 by holta