iiab/iiab #1822: Failure at Task [Cmdsrv: Create a link python can use to import iiab-make-kiwix-lib functions]

holta, 2019-07-06:
@tim-moody can you confirm this Report below, by NCR in Michigan?

Sounds like vars `kiwix_install: True` and `kiwix_enabled: True` are effectively mandatory for now?

And that we should ask him to re-enable these in `/etc/iiab/local_vars.yml` for now?

_CONTEXT: he pays per GB downloaded, so needs to be very judicious in what he enables for download!_

> 1.7 GB, 50 min to Failure of MIN-size using Buster on Pi 3 with current updates.
Connected via serial port to TPLink 3020 as AP client to ZTE HotSpot to Boost Mobile tower with maximum number bars (5)
>
> Failure at Task [Cmdsrv: Create a link python can use to import iiab-make-kiwix-lib functions].  Rerun of sudo iiab fails same task. (src file does not exist:  src : /usr/bin/iiab-make-kiwix-lib.py)
>
> May have made error in edit /setting kiwix install / enable to false in edit of /etc/iiab/local_vars.yml.  Unsure if script echoing the
>
> [copying] of /etc/iiab/local_min_vars.yml  to local_vars.yml or note for me to manually copy and edit.

holta, 2019-07-08:
@tim-moody should we declare this "wontfix" ?

Or is this something that should be worked on in the IIAB 7.1 release cycle?

tim-moody, 2019-07-08:
kiwix_install: False will break things. Not worth fixing.

holta, 2019-07-08:
> kiwix_install: False will break things. Not worth fixing.

Ok, I'll document this in local_vars.yml so people don't fall in this trap as often going forward.

holta, 2019-07-09:
Mitigated with documentation, in PR #1848

closed 2019-07-09 by holta