iiab/iiab #1830: Lokole admin user "settings" page does not work

m-anish, 2019-07-07:
### Expected behavior and actual behavior.

Admin settings page for lokole does not work. 

### Steps to reproduce the problem.

* Install `lokole` as part of IIAB install
* Login as administrator with Admin/changeme
* Access the Settings tab
* You get `Unexpected error. Please contact your administrator. ` message
* Output of systemctl status lokole is below
```

● lokole.service - Provides the Lokole Server
   Loaded: loaded (/etc/systemd/system/lokole.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-07-06 13:40:11 IST; 24h ago
 Main PID: 1144 (bash)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/lokole.service
           ├─1144 /bin/bash /home/lokole/state/webapp.sh
           ├─1202 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/gunicorn --timeout=300 --workers=4 --bind=unix:/home/lokole/state/lokole_gunicorn.sock --log-level=error opwen_email_client.webapp:app
           ├─2020 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/gunicorn --timeout=300 --workers=4 --bind=unix:/home/lokole/state/lokole_gunicorn.sock --log-level=error opwen_email_client.webapp:app
           ├─2034 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/gunicorn --timeout=300 --workers=4 --bind=unix:/home/lokole/state/lokole_gunicorn.sock --log-level=error opwen_email_client.webapp:app
           ├─2042 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/gunicorn --timeout=300 --workers=4 --bind=unix:/home/lokole/state/lokole_gunicorn.sock --log-level=error opwen_email_client.webapp:app
           └─2155 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/gunicorn --timeout=300 --workers=4 --bind=unix:/home/lokole/state/lokole_gunicorn.sock --log-level=error opwen_email_client.webapp:app

Jul 06 13:40:11 box systemd[1]: Started Provides the Lokole Server.
Jul 07 14:05:19 box bash[1144]: [2019-07-07 14:05:19 +0530] [2020] [ERROR] TypeError: expected str, bytes or os.PathLike object, not NoneType
Jul 07 14:05:24 box bash[1144]: [2019-07-07 14:05:24 +0530] [2042] [ERROR] TypeError: expected str, bytes or os.PathLike object, not NoneType
Jul 07 14:05:52 box bash[1144]: [2019-07-07 14:05:52 +0530] [2042] [ERROR] TypeError: expected str, bytes or os.PathLike object, not NoneType
Jul 07 14:06:03 box bash[1144]: [2019-07-07 14:06:03 +0530] [2034] [ERROR] TypeError: expected str, bytes or os.PathLike object, not NoneType

```

### Specifications like the version of the IIAB, operating system version, or hardware details.

Ubuntu 18.04.2 LTS, IIAB from Master branch, Intel NUC hardware, Gateway mode

holta, 2019-07-07:
@c-w & @aidan-fitz can you possibly look into this?

(Ideally if you have time in coming days for this team in North India, that is taking this work into mountain villages quite shortly, in the coming week or so ;)

holta, 2019-07-07:
Just FYI I've verified this error with IIAB running on 4 different OS's: Ubuntu 18.04/Desktop, 18.04/Server, Raspbian Buster Lite, Debian 10 &mdash; in each case http://box/lokole -> username/password: Admin/changeme -> `Settings` (button on top, in the middle) yields:
```
Unexpected error. Please contact your administrator.
```
Note: Lokole's `Settings` button does not appear for regular Lokole accounts.  It appears to be something for Administrators only?

aidan-fitz, 2019-07-08:
On it right now. Did this failure occur in 0.4.2 @m-anish @holta?

holta, 2019-07-08:
> On it right now. Did this failure occur in 0.4.2 @m-anish @holta?

Sorry I don't know if it also occurred in 0.4.2

(Certainly it's definitely occurring in 0.4.3)

m-anish, 2019-07-08:
0.4.2 as per the main/defaults.yml file

aidan-fitz, 2019-07-08:
Okay, I could not reproduce the bug in the standalone version of Lokole, but I do see it in Lokole with IIAB. Doing a `git bisect` to find the commit where the bug first occurs.

aidan-fitz, 2019-07-08:
Based on the `git bisect`, the bug appears at 9fda1fb64f61aeb3340b2a09ca7d7a1eba3dfa43. We jumped several versions from 0.1.26 to 0.1.35 then, so next step is to figure out which version of Lokole introduced the bug.

aidan-fitz, 2019-07-08:
I made a dummy branch of the IIAB repository in my own fork, populated it with every version of Lokole from 0.1.26 to 0.1.35, and then ran a bisect to determine which version first introduced the error. The result:
```
$ sudo git bisect good
c123c6c34e2484af1fa598261e60de1f67ac27e7 is the first bad commit
commit c123c6c34e2484af1fa598261e60de1f67ac27e7
Author: Aidan Fitzgerald <acf67@cornell.edu>
Date:   Sun Jul 7 23:12:15 2019 -0700

    0.1.33 -> 0.1.34
    
    Signed-off-by: Aidan Fitzgerald <acf67@cornell.edu>

:040000 040000 36f27ea497d52527820f7b03324715b011c72f77 c4f7fc2369cbacc78a7d42466e73d4546acc7d03 M	roles
```

The bug was first introduced between 0.1.33 and 0.1.34.

aidan-fitz, 2019-07-08:
The next step is to find the exact commit in ascoderu/opwen-webapp that introduced the bug. We can [tell pip to install a specific commit](https://stackoverflow.com/questions/13685920/install-specific-git-commit-with-pip) and make a commit on a dummy branch for each commit in opwen-webapp between `0.1.33` and `0.1.34`. Luckily we only have to do this for one commit.

https://github.com/ascoderu/opwen-webapp/compare/0.1.33...0.1.34

aidan-fitz, 2019-07-08:
Based on more git bisecting, I have found that the pull request ascoderu/opwen-webapp#122 (which created merge commit ascoderu/opwen-webapp@f920763a69e7e32841dfe3903959afc19050a3ba, the last commit before version 0.1.34 was released) introduced the bug.

m-anish, 2019-07-08:
Thanks for working on this. Please tag me when you think you have a fix, or suggest a workaround. I'll pull that in and test. We leave in about a week so ideally, to get this working, I'll want to close it out in the next ~3 days. 

Lokole is not a blocker for us, but a good-to-have.

aidan-fitz, 2019-07-08:
On second thought, the settings page only has network settings, which don't apply to the IIAB edition of Lokole. We really should just be hiding the settings page from offline users.

This is what the settings page looks like:
[image]

aidan-fitz, 2019-07-08:
I've submitted pull request https://github.com/ascoderu/opwen-webapp/pull/197, which I think will resolve this issue. For now, just ignore the settings page. :)

holta, 2019-07-08:
Resolved by PR #1844; @m-anish please confirm!

closed 2019-07-08 by holta