iiab/iiab #1831: Instant submission of Content Pack descriptions/logos, for upstream/sidestream community review

holta, 2019-07-07:
Instant submission of Content Pack descriptions/logos for your IIAB home page is a nearly universal desire among implementers, so that implementers/educators around the planet can rapidly & efficiently share "menu item definitions" — putting community action into high gear — without GitHub bureaucracy.

Naturally there will still be _some_ community/grassroots review process to triage submissions & suggestions, so while submission should be instant & transparent to all, publication itself may not be instant if course!

Facilitating "sidestream" workflows should also be considered, among more offline communities, where a fully hierarchical approval process is certainly not always appreciated or appropriate.

In Any Case: the general community workflow to describe+depict Content Packs (and IIAB Apps, etc?) needs to be much closer to "WIKI" (Hawaiian word for fast!) than a technocratic process like GitHub PR's, which unintentionally-but-inherently deflates educators enthusiasm to contribute — _in those very areas where their talents & excitement are most suited to helping!_

Unleashing these fervent grassroots desires the be heard is a goal worth pursuing — regardless when-and-if diverse larger plans for decentralized Sneakernets-of-Alexandria take root or not !!

Refs: #828 #1002

holta, 2019-07-13:
Tangentially related:

#1862 Logo corners are chopped: can this be prevented?

holta, 2019-07-18:
Several design ideas from 2019-02-10 are listed & linked here:
https://github.com/iiab/iiab/issues/1002#issuecomment-462183401

holta, 2020-04-27:
Am moving this to the IIAB 7.2 milestone as @tim-moody has worked on this a lot in recent weeks.  Not at all an easy task!

Hopefully with IIAB 7.2 and 7.3 the "community review" process (as mentioned in the original subject line above) can become more visible and tactile...seeding collaboration (intentionally and unintentionally) starting with basic de-spamming techniques and associated UX...growing from there...inviting solidarity around proven content bouquets etc.

tim-moody, 2020-04-27:
> Instant submission of Content Pack descriptions/logos for your IIAB home page is a nearly universal desire among implementers,

Not sure what instant means as the upload of a couple of menu defs takes maybe 10 seconds. But that is doable now. Check the upload checkbox when you save. After you're finished editing click sync menu defs and its done.

Of course there is no community review.

jvonau, 2021-07-21:
done?

holta, 2021-07-21:
> done?

A complex affair for sure, that's always improving!

A public discussion area for this topic would be great &mdash; whether this ticket or another.