iiab/iiab #1832: Tracking the best systemd unit file(s) for Kolibri?

holta, 2019-07-07:
@benjaoming wrote @ https://github.com/iiab/iiab/pull/1828#issuecomment-509012372:
> In light of this issue and the fact that IIAB is shipping its own systemd service, I would recommend that you open an issue to have Kolibri installed through the Debian package instead? Interactive with dpkg,

Just so I understand, are you suggesting something like either of the following?
https://docs.ansible.com/ansible/latest/modules/apt_module.html
https://docs.ansible.com/ansible/latest/modules/dpkg_selections_module.html

IIAB has had endless trouble installing certain .deb packages directly in the case of Calibre, and likewise the lastest PPA's for Ansible, but perhaps the process is becoming more sane in recent years ??  Not that pip/npm/yarn are perfect by any stretch of the imagination, when normal `apt` installs are not possible :-)

> whether as an uninstall, reinstall or fresh install will manage the systemd job, starting, stopping and reloading it when necessary.

If https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri.service.j2 is not the latest we all definitely need to know yes!

Conversely where should we find the latest/recommended systemd unit file Kolibri that you recommend in general?  Certainly IIAB has to manage diverse such system unit files, which sometimes need delicate modifications on slow-booting / older machine etc, especially when many service are running, that @jvonay and others have greatly helped with~

holta, 2019-07-07:
Quick Background on the narrow issue of systemd unit files: even "official" unit files e.g. for major packages like dnsmasq very often contain showstopper bugs, so IIAB has learned that we have to be extremely judicious in choosing/refining the best strategies here!

(Accepting system unit files from upstream without extensive testing+review is definitely not an effective strategy in the end, despite our best wishes that has NOT worked out ;-)

jvonau, 2019-08-25:
Closest [upstream](https://github.com/learningequality/kolibri-server/blob/master/debian/kolibri-server.service) unit file. We do not currently use kolibri-server as the webserver is not apache but nginx.

benjaoming, 2019-08-26:
@jvonau it's part of the Debian package source, which is separate. https://github.com/learningequality/kolibri-installer-debian/blob/master/debian/startup/kolibri.service

jvonau, 2019-09-07:
Think then the question for IIAB is do you use the upstream ppa repo or continue to install via pip.

holta, 2019-10-21:
@jvonau can we close this now?

holta, 2019-12-21:
@jvonau LMK if this needs to be re-opened.

closed 2019-12-21 by holta