iiab/iiab #1833: Lost connection to MySQL server at 'sending authentication information', system error: 32 "Broken pipe"

holta, 2019-07-07:
Is this something we should be concerned with, e.g. on Raspbian Stretch Lite on RPi 3 where it blocked IIAB installation / reinstallation?

FWIW re-running IIAB's installer i.e. `cd /opt/iiab/iiab; ./iiab-install --reinstall` solved the problem here:

> TASK [mysql : Update MySQL root password for localhost root accounts, if mysql_enabled] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "unable to connect to database, check login_user and login_password are correct or /root/.my.cnf has the credentials. Exception message: (2013, 'Lost connection to MySQL server at \\'sending authentication information\\', system error: 32 \"Broken pipe\"')"}

holta, 2019-07-15:
Moving to the [IIAB 7.1 Milestone](https://github.com/iiab/iiab/milestone/6).

Let's close this in the coming month if it's not reproducible or recurring.

holta, 2019-08-13:
Does this have anything to do with...?

https://github.com/iiab/iiab/issues/1714 "MySQL database password-changing failure during fresh install of IIAB on Ubuntu 19.04 [and 19.10 pre-releases?]"

georgejhunt, 2019-09-04:
If restarting the install caused the problem to be overcome, I don't think it is a permissions problem.  If that's the case, "become" will not change the problem, or correct it. 

Perhaps a delay might be a better thing to try.

holta, 2019-09-04:
Agree.  If this is a common enough flaw to be worth solving.

Whereas #1833 happens every time (on Ubuntu 19.04 and presumably 19.10 / 20.04 too?) so might be our higher priority.

holta, 2019-10-19:
This is not a recurring/common problem, so am closing.

Please re-open as nec.

closed 2019-10-19 by holta