iiab/iiab #1849: @m-anish & @tim-moody ask: can systemd now replace Monit?

holta, 2019-07-08:
Please weigh in as to whether [Monit](https://github.com/iiab/iiab/tree/master/roles/monit) should be deprecated in the IIAB 7.1 release cycle, much as Debian 10 "Buster" has now done:

#1842 Ugly workaround so that IIAB installs on Debian 10 Buster (which suddenly removed monit)

Thanks you to @m-anish & @tim-moody for raising this question during the IIAB community/team call earlier this morning (http://minutes.iiab.io)

ASIDE: the Monit project is not dead, as you can see here: https://mmonit.com/monit/changes/

As such if someone can find out why a community-centric project like Debian decided to drop Monit in very recent weeks, that might help us understand the situation?

holta, 2019-07-08:
@m-anish wrote an email to the Debian maintainer of `monit` asking if there are more details...explaining options going forward?

e.g. if https://packages.debian.org/buster/mon (or otherwise) might be an alternative to Monit going forward, even if `mon` is not a drop-in replacement to be added into https://github.com/iiab/iiab/tree/master/roles/monit/tasks apparently?

holta, 2019-07-08:
@m-anish's email to Debian's monit maintainer Sergey B Kirpichev, who responded that backports of MongoDB might later be possible:
>> We don't see monit in the debian 10 package repos and wonder if there are any plans to
>> make it available on it?
>
> Maybe backports.  See release notes.

also this -- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=887350#26

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=927775

holta, 2020-09-22:
FYI the 'monit' package would (appear) to be returning to Debian 11 / Bullseye when that is released (around) mid-2021:

https://packages.debian.org/bullseye/monit
https://packages.debian.org/source/bullseye/monit

A reminder that the 'monit' package was suddenly removed from Debian 10 / Buster during the very final days prior to its mid-2019 release, as confirmed by the sudden disappearance of these 2 pages:

https://packages.debian.org/buster/monit
https://packages.debian.org/source/buster/monit

Please also see this file, which will be modified very shortly in coming days:

https://github.com/iiab/iiab/blob/master/roles/monit/tasks/main.yml

Ref: #2536

holta, 2021-12-31:
FWIW Monit is once again part of Debian:

https://packages.debian.org/bullseye/monit

Not that I know anybody who truly uses it & depends on it :/

tim-moody, 2021-12-31:
I think I used to include munin when building images. More recently I think I confused it with monit and included that instead. 

To the extent that monit is used to restart monitored services, I think it is correct that systemd unit files can have a parameter to do that. So probably it no longer serves a purpose.