iiab/iiab #1854: Is /etc/network/interfaces.d/patch_auto needed?

holta, 2019-07-09:
@jvonau as mentioned in your PR #1829 (WAN detection) is /etc/network/interfaces.d/patch_auto sometimes needed?

How should this be placed automatically if so?

jvonau, 2019-07-09:
Backstory is 58ce24bd4c92e6b3e900b2e64ddd397ed57643a8 was introduced early in Debian-10's testing sequence as -10 was close to being the same as -9 but might of changed as time progressed. Think we need to confirm that a stock Debian-10 install raises the network correctly upon boot without adding patch_auto. Should that prove to be unneeded on a pre-iiab install then test removing patch_auto from an iiab prep'd install in both Appliance & Gateway Modes to confirm no longer needed.

holta, 2019-07-09:
@jvonau: I've installed a fresh/clean Debian 10.0.0 (Server not Desktop) VM @ 192.168.0.100 for now, if that helps you confirm?

holta, 2019-07-09:
Earlier @jvonau wrote:
> just reboot [vanilla Raspbian Buster Lite on RPi 4 @ .182], if the network come up were good, data point one.

.182 reboots fine.

> On the [similar] iiab installed machine [.181] remove patch_auto and reboot,  if the network come up were good, data point 2 . last would be to check the behavior with br0 present.

.181 [10.8.0.14] is being rebuilt with a new BIG-sized IIAB on a fresh microSD...so experiments 2 & 3 will have to wait for later this evening.

holta, 2019-07-09:
Correction to the above- this is about "real" Debian 10.

And both initial experiments are confirmed:
1) vanilla Debian 10.0.0 .162 (formerly .100) reboots fine
2) Debian 10 w/ BIG-sized IIAB @ .161 reboots fine, event after /etc/network/interfaces.d/patch_auto was removed.

holta, 2019-07-11:
@jvonau which playbook/file should delete `/etc/network/interfaces.d/patch_auto` ?

Is 'real' Debian 10+ the only OS where this needs to happen?

holta, 2019-07-18:
Thanks to @jvonau this is solved by PR #1886 (now merged).

Please test on a fresh/clean install of Debian 10.0.0 Buster those who can!

closed 2019-07-18 by holta