iiab/iiab #1856: nodejs fails to install on updated buster because of suite change testing->stable

georgejhunt, 2019-07-10:
### Expected behavior and actual behavior.

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.

holta, 2019-07-10:
I have not seen the in extensive testing of "real" Debian 10.0.0 VM's and Raspbian Buster Lite on various RPi's.

Perhaps b/c I ran "apt update" and "apt -y dist-upgrade" manually a priori, and accepted the testing->stable change when prompted?

Which Buster?

Which HW/VM?

Are you running `curl http://d.iiab.io/install.txt | sudo bash` and is it chocking?

georgejhunt, 2019-07-10:
Maybe it's just because I was working on an rpi image I created a couple of weeks ago. I was just running iiab-install --reinstall.  Perhaps running the d.iiab.io curl script, with it "apt-get dist-upgrade", would have takes car of it.

holta, 2019-07-10:
Good question.

We should try some installed without any "apt update" & "apt -y dist upgrade" a prior.

Better to put ourselves in the shoes of a total newcomer / newbie, and follow the install instructions at http://download.iiab.io quite literally, to see if that does or does not work !

holta, 2019-07-10:
FWIW i personally have never needed to run the command `apt update --allow-releaseinfo-change` on "real" Debian 10.0.0 Buster or on Raspbian Buster "Lite" or "With Desktop"

...presumably b/c I manually ran "apt update" and then typed in "y" when prompted to accept the upstream repo changes from...
1) "testing" -> "stable"
and
2) "" -> "10"

georgejhunt, 2019-07-11:
1. On clean "dd" of  buster lite, 6/20/19 release. "apt-get update" fails with an error message that points to "man apt-secure(8)", which in turn fails to mention "apt-get update --allow-releaseeinfo-change".
2. Lots of confusion mentioned at https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=245022.
3. In addition, "raspi-config" change-keyboard gets corrupted fonts, and I cannot select my keyboard.
4. Perhap we should delay release  of 7.0 until buster quiets down.

holta, 2019-07-11:
Which exact RPi are you running?

1) On RPi 4 it's reasonably self-explanatory *IF* we're talking about a knowledgable Linux person, who types in the following: (on Rasp Buster Lite 2019-06-20)

```
pi@raspberrypi:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N]
```

2) Does `apt -y update` work for you, to completely solve the problem, as it does on RPi 4?  (So we can tune IIAB's installer, to do most/all of this stuff automatically, for newbies that will indeed turn away in disgust!)

holta, 2019-07-11:
> 4. Perhap we should delay release of 7.0 until buster quiets down.

While I've not faced the problems you're facing (for whatever reason, possibly things are smoother on RPi 4?) yes let's hope they release another version of Raspbian Buster @ https://raspberrypi.org/downloads/raspbian/ in coming days/weeks, to make _everybody's_ life much easier!

georgejhunt, 2019-07-11:
>    1. Perhap we should delay release of 7.0 until buster quiets down.

confirmed that on a rpi3, "sudo apt update" provides an interactive Y/N
question about migrating to the stable release (which is not the case using
"apt-get update").

I am still not able to get out of the UK keyboard mapping. Raspi-config
still has square blocks displayed on the keyboard selection screen. Once
before, I was able to select a keyboard (usually I select generic 101 keys
-- which is not now displayed at all as an option), but it must have been a
lucky guess, when I chose a keyboard make/manufacturer.

holta, 2019-07-11:
Thanks much @georgejhunt 

I've warned  the ~100 people on the xsce-devel mailing list, until a more proper solution is in place.

holta, 2019-07-11:
@georgejhunt I've applied this Temporary Workaround:

https://github.com/iiab/iiab-factory/pull/70

If possible can you try again, i.e. a clean/fresh install of http://d.iiab.io on Rasp Buster?

georgejhunt, 2019-07-11:
> iiab/iiab-factory#70 <https://github.com/iiab/iiab-factory/pull/70>

I'm stuck because I cannot follow the following instructions at d.iiab.io:

#                 curl d.iiab.io/install.txt | sudo bash

because I cannot find the "|" on my generic keyboard, and the "sudo
raspi-config" does not work as it has in the past. I cannot find the
"generic 101" keyboard listed as an option.

When I googled keyboard configuration, it suggests:

   1.

   sudo dpkg-reconfigure keyboard-configuration

which brings me to the same screen which no longer displays the generic
keyboards.

Really frustrating. Not what I would want to impose on newbies.

holta, 2019-07-11:
> Really frustrating. Not what I would want to impose on newbies.

@georgejhunt can you report the full details to Raspbian's forums (https://www.raspberrypi.org/forums/) to make sure this is fixed without delay?

Is the Raspbian sub-forum the most appropriate perhaps?
https://www.raspberrypi.org/forums/viewforum.php?f=66

holta, 2019-07-11:
1) @georgejhunt can you try Raspbian's brand new kernel & new `raspi-config` after `apt update` and `apt -y dist-upgrade && reboot` ?

> raspberrypi-kernel/testing 1.20190709-1 armhf [upgradable from: 1.20190620+1-1]
> raspi-config/testing 20190709 all [upgradable from: 20190620]

2) We should also try Raspbian With Desktop, to see if keyboards/newbies will fare better on that (most popular of all) Raspbians ?

Unfortunately I cannot attach a screen to my RPi 4's, as its 2 microHDMI ports are highly non-standard :(

holta, 2019-07-11:
Before-and-after kernel versions, just for the record...
```
root@box:~# uname -a
Linux raspberrypi 4.19.50-v7l+ #895 SMP Thu Jun 20 16:03:42 BST 2019 armv7l GNU/Linux

root@box:~# uname -a
Linux box.lan 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

georgejhunt, 2019-07-11:
I tried upgrade on the lite image  to kernel  4.19.57-v7+ #1244 SMP , but it did not correct the raspi-config keyboard problem.  

Then I tried 6/20/19 desktop. The startup screen leads the user to change password, select locale and keyboard. At the completion of the startup interaction, I opened a terminal, and ran 'sudo raspi-config', which showed the generic keyboards.  The kernel was still 4.19.50.
So maybe for most of our users, this is not a  blocker, the problem seems limited to the lite version.

holta, 2019-07-11:
> So maybe for most of our users, this is not a blocker, the problem seems limited to the lite version.

Still very serious.

Thanks for the very comprehensive reportage.

Is there no trace of this raspi-config failure to configure keyboards within RPi's forums?

Not that we're responsible for every upstream bug &mdash; but still we should report/upvote this so that maintainers are well aware of this serious problem &mdash; can you perhaps file an issue against https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config if not in RPi's forums?

holta, 2019-07-12:
Huge thanks to @georgejhunt for reporting this upstream:
https://github.com/RPi-Distro/raspi-config/issues/97

The response so far (in the above tkt) suggests we might have to wait a month-or-few until this is repaired?  As quoted here:
> I believe this is an upstream Debian issue that should be fixed in the next point release.

holta, 2019-07-12:
@georgejhunt can you give the new 2019-07-10 Raspbian Buster Lite image a shot, to see if it solves one (or possibly even both) of these issues?

https://www.raspberrypi.org/downloads/raspbian/

holta, 2019-07-12:
It might possibly also contain the new firmware that can lower the RPi 4 temperature by 5 degrees Celsius or so?

http://downloads.raspberrypi.org/raspbian/release_notes.txt

georgejhunt, 2019-07-12:
The 7.10.2019 lite release solves the buster testing->stable issue, but
does not solve the keyboard issue.

holta, 2019-10-06:
> The 7.10.2019 lite release solves the buster testing->stable issue, but does not solve the keyboard issue.

@georgejhunt does the new [Raspbian 2019-09-26](https://raspberrypi.org/downloads/raspbian/) solve your above keyboard issue?

holta, 2019-12-21:
> > The 7.10.2019 lite release solves the buster testing->stable issue, but does not solve the keyboard issue.
> 
> @georgejhunt does the new [Raspbian 2019-09-26](https://raspberrypi.org/downloads/raspbian/) solve your above keyboard issue?

I'm going to assume this is resolved 6 months later.

@georgejhunt  plz re-open as nec!

closed 2019-12-21 by holta