iiab/iiab #1860: Do we care about Elgg 3.0.5+ which fail as part of IIAB?

holta, 2019-07-11:
[Elgg 2.3.13 LTS](https://elgg.org/about/download) works fine with IIAB.

But the newer, non-LTS version ([Elgg 3.0.5](https://elgg.org/about/download), also available from http://download.iiab.io/packages/) fails when installing it with IIAB, i.e. http://box/elgg shows a 500 error in the browser, and the following in `/var/log/apache2/error.log` : 

> [Wed Jul 10 23:58:25.968002 2019] [mpm_prefork:notice] [pid 22169] AH00169: caught SIGTERM, shutting down
[Wed Jul 10 23:58:26.142015 2019] [ssl:warn] [pid 22559] AH01909: box:443:0 server certificate does NOT include an ID which matches the server name
PHP Warning:  PHP Startup: stem: Unable to initialize module\nModule compiled with module API=20151012\nPHP    compiled with module API=20170718\nThese options need to match\n in Unknown on line 0
[Wed Jul 10 23:58:26.222786 2019] [ssl:warn] [pid 22560] AH01909: box:443:0 server certificate does NOT include an ID which matches the server name
[Wed Jul 10 23:58:26.232295 2019] [mpm_prefork:notice] [pid 22560] AH00163: Apache/2.4.29 (Ubuntu) mod_fcgid/2.3.9 OpenSSL/1.1.1 mod_wsgi/4.5.17 Python/2.7 configured -- resuming normal operations
[Wed Jul 10 23:58:26.232342 2019] [core:notice] [pid 22560] AH00094: Command line: '/usr/sbin/apache2'

jvonau, 2021-07-21:
Close?

holta, 2021-07-21:
I don't think we care about Elgg 3.x now that Elgg 4.x Betas have been circulating for ~2 months now:

https://elgg.org/about/download

But possibly this tkt is the best placeholder for this difficult topic.

holta, 2021-09-24:
@jvonau wrote: [2021-07-21]
> Close?

Yes.  Let's close this now that Elgg 4.0.0 is officially released.

Anyone that wants to invest in making Elgg 4.0.0 work with IIAB — please follow up here:

- #2981

Thanks!

closed 2021-09-24 by holta