iiab/iiab #1866: Grid-style IIAB home page (with columns?) for "libraries" with 50+ Content Packs & Apps?

holta, 2019-07-14:
@tim-moody how challenging would it be to (also!) offer a grid-like output for those library-like communities drowning in 50+ Content Packs & Apps on their IIAB home page?

This is becoming more prevalent as 256GB and 400GB microSD cards are now becoming very affordable.

A colorful grid harking to a tablet's screen showcasing apps would seem to be the most desirable in many of these situations &mdash; even if mouseover "hovering" for descriptions/details is much, much harder on such mobile-like platforms, so design issues are complex!

(In any case, the grid-display at http://box/kiwix is illustrative, even if not nearly colorful enough, nor curated enough, to be broadly useful except as an ugly index.  _However:_ its 3 columns are not terribly far off from what [many] info-intensive library-like communities seem to want+need, when faced with a growing mountain of IIAB content packs / apps, typically 50 or more...unlike times past where ~30 was far mor typical!)

Somewhat Related:
#1831 Instant submission of Content Pack descriptions/logos, for upstream/sidestream community review
#1862 Logo corners are chopped: can this be prevented?

holta, 2019-07-15:
FWIW another example of grid-style layout of Content Packs is the Internet Archive's Decentralized Web: 

- https://dweb.archive.org (can take a full minute-or-so to load)

_(Certainly this one is much more colorful than things like http://iiab.me/kiwix !)_

avni, 2019-07-16:
Here is another example which enables toggling between a grid and list view: 
https://learn.partners.org/org/mgh-research-institute/card/
https://learn.partners.org/org/mgh-research-institute/date/

I think the mobile version of this can be improved slightly but this shows displaying a large number of items in an easily digestible format - with search and filter capability. Clicking on an individual item takes you to a page with additional detail. This site layout was done using Bootstrap 4.3.1.

holta, 2019-07-27:
The 81 Content Packs & IIAB Apps etc (https://paste.ubuntu.com/p/gJ5WTwkyFQ/) on the main page of the Tanzania 400GB IIAB that shipped yesterday make it obvious we really need some design innovation here.  Scrolling down through 81 things is absolutely overwhelming, on mobile especially.

One idea that @metasj suggested was collapsable/minizable sections.

Just as one example of that, here are 4 textbook Content Packs that really should be clustered somehow?
- [Boundless](http://oer2go.org/viewmod/en-boundless)
- [OpenStax](http://oer2go.org/viewmod/en-openstax)
- [CK-12](http://oer2go.org/viewmod/en-ck12)
- [Saylor Academy](http://oer2go.org/viewmod/en-saylor)

CLARIF: the RACHEL/OER2Go modules linked to above are not the only versions &mdash; ZIM files and Kolibri Channels also exist for several of them &mdash; i.e. for those who prefer a more searchable/modern Learning Pack format, possibly including pop quizzes etc to make this fun, keeping kids engaged!

holta, 2019-08-01:
@tim-moody observes that a grid-style layout might help on desktop (large screens) but does nothing for mobile (i.e. small screens).

He asks if a multi-tab interface might possibly help on mobile? (i.e. small screens)

avni, 2019-08-01:
Can you share a screenshot example of a mobil multi-tab interface?

llovemixo, 2020-01-14:
Suggest use of menu of menus