iiab/iiab #1873: How to delete menu item using Admin Console ?

holta, 2019-07-15:
NCR in Michigan asks:

> Can drag and drop to right list, but appears (after reviewing documentation) that can only delete via linux modules ?
>
> Still looking at ways to add small amount of data later when internet available. Default is to block internet access, including iiab-admin, when using hot spot. Dropping in keyboard / mouse seems only way to then access internet (ignoring virtual networked computers SSH).  The "testing to stable"  issue dropped a monkey wrench in the middle of this.
>
> Thinking wifi dongle carried with SD from Pi 4 to Pi Zero W is path.   The Zero can handle a passive two position hub for wireless keyboard dongle and wifi dongle.

holta, 2019-07-16:
At the end of the day, NCR is still stuck, unable to remove Content Packs from his IIAB home page (http://box).

He tried http://box.lan/admin > `Content Menus` > `Content Item List` > (drag content to the right, trying to remove it) > `Save Menu`

And many other attempts.

@tim-moody can you provide the essential instructions, to remove content from an IIAB home page (http://box) using Admin Console?

tim-moody, 2019-07-16:
I'm not clear on the question. Is the wish to remove a menu item from the menu or a menu item definition from the file system? The latter can indeed only be done in the Linux file system. The former can be done with drag and drop.

holta, 2019-07-17:
I'm now understanding from NCR that he was trying to accomplish **both aspects** of removing a Content Pack, i.e.:
1. Remove menu item from http://box (the IIAB home page) and...
2. Remove (possibly voluminous) content from the disk.

He now understands that there are 2 separate steps involved.

Where is the best place to document this?
(NCR struggled to find answers here within documentation over 2 days, so we should find a place for this-)

holta, 2019-07-24:
@tim-moody NCR is dedicated here over weeks, also now trying http://box.lan/admin > Install Content > Manage Content > Remove Selected Content

What are the best ways to make these features more broadly discoverable for others/all...clarifying the non-intuitive steps...e.g. are Screencast Videos and http://FAQ.IIAB.IO maybe both needed to explain that content deletion needs to be done entirely separately from...menu item deletion?

holta, 2019-07-25:
Content Deletion needs a fuller explanation in places like http://FAQ.IIAB.IO & screencast/video, but in the meantime this excerpt from https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#manage-content is a start:

> To Copy or Remove internal content click on the left panel. To use the USB, click on the right.
>
> Check the box beside the content of interest and click Copy Installed to ... to Copy or Remove Selected Content to Remove.
>
> Because copying happens in the background you can watch its status on Display Job Status on the Utilities menu to see when it has completed or any problems encountered.