iiab/iiab #1874: Mounting disks ?

mitra42, 2019-07-15:
### Expected behavior and actual behavior.
I'd expect that plugging a USB hard drive into the USB port would make it available

### Steps to reproduce the problem.
Plug in USB hard drive 
Look in /media/*

### Specifications like the version of the IIAB, operating system version, or hardware details.
IIABv7, Seagate Hard disk with a NTFS file system

I'm not succeeding in getting hard disks to work on IIAB.
 I've manually added `max_usb_current=1` to `/boot/config.txt` as suggested on https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=101084 but that didn't help.

/etc/usbmount/usbmount.conf also looks correct for NTFS according to various suggestions I found. 

Has anyone worked on this?

holta, 2019-07-15:
@mitra42 does the top of http://FAQ.IIAB.IO help, where it explains "Can teachers display their own content?"

Short Version:

By default most all disks/drives of almost any kind should be auto-mounted to /library/www/html/local_content and visible to all (publicly!) at http://box/usb

Please do not rely on this much older/earlier doc which is unmaintained:
https://github.com/iiab/iiab/blob/master/roles/usb-lib/README.rst

mitra42, 2019-07-15:
Yes - it helps - somewhere I got the idea that IIAB mounted drives in one place - on /media/pi (as a link to the /media/usb/* location. The funny thing is that this *DOES* happen when you insert a USB stick but not when its rebooted.   I'll change my code to look in each of the /media/usb*/ locations.

holta, 2019-07-15:
An NTFS-formatted external disk worked for me a few months ago (or maybe it was 2018) exactly the same way a USB memory stick / thumb drive &mdash; with all content mounting in `/library/www/html/local_content`

That was using an RPi 3 B+ running Raspbian Stretch Lite.  Now of course we favor RPi 4 running Raspbian Buster &mdash; whether the "Lite" version or the "With Desktop" version &mdash; both are common.

What is your exact HW & your exact OS?

holta, 2019-07-15:
@mitra42 FYI for comparison a USB NTFS hard disk just auto-mounted successfully to `/media/usb0` on IIAB 7.0 pre-release running Raspbian Buster on RPi 4, and its (HDD) content is available publicly over the IIAB/RPi's hotspot, with URL http://box/usb

So I'm unable to reproduce the symptom you describe.

Is it possible your modifications (to `/boot/config.txt` or otherwise?) is interfering with IIAB's default behavior to mount USB disks?

mitra42, 2019-07-15:
I've got a RPI3, uname -a reports: 
```
Linux box.lan 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux 
```
Which was downloaded from Rasbian in the last few days. Its the desktop version. 
The only change to /boot/config.txt was to add the line to increase max power draw on the USB's, and that was after I started seeing the problem.

mitra42, 2019-07-15:
The Hard disk was a NTFS drive, since that failed I reformated as a exFAT with no change to behavior

holta, 2019-07-15:
> The Hard disk was a NTFS drive, since that failed I reformated as a exFAT
> with no change to behavior
>

Myself I cannot reproduce this.

External (USB) NTFS hard disk auto-mounts within seconds to /media/usb0 on
IIAB 7.0 pre-release on latest Raspbian Buster Lite on RPi 4, as mentioned
above.

>

mitra42, 2019-07-15:
Thanks for trying to replicate, let me try reformatting my disk back to NTFS, maybe was a problem with this specific disk.

holta, 2019-07-15:
FYI likewise with a RPi 3 B+ running the latest IIAB 7.0 pre-release on Raspbian With Desktop: an external USB NTFS hard disk was just now tested to auto-mount to /usb/usb0

There must be something oddly anomalous and/or non-standard about your setup.

Let us know if you can identify what that is.

As this auto-mounting functionality has been tested in very diverse situations (with different external USB disks, etc, in many countries schools) in recent years, where schools rely on it, so that their materials appear nearly instantly across the entire classroom, using URL http://box/usb

mitra42, 2019-07-15:
Will do ....Looks like I dont have a way to format NTFS (can't do on a mac) but I'm trying a few other variations since sounds like problem might be here :-)

holta, 2019-07-15:
> Will do ....Looks like I dont have a way to format NTFS (can't do on a mac) but I'm trying a few other variations since sounds like problem might be here :-)

Weird.  NTFS is ~20 years ago now, so I imagined it would be de facto open, even if exFAT might still be encumbered by patents?

mitra42, 2019-07-15:
Maybe its because its so old, that its rarely seen - OSX can read NTFS but not write (or format).

I think I nailed it - I found another drive (Toshiba 2TB), and that worked, and then on a hunch I think the problem was power related, switching to different USB power source, and reducing it to just a single drive , and neither keyboard nor display, seemed to fix it even for the first drive, which must just be a power hog. 

Thanks for helping me narrow down the issue

holta, 2019-07-18:
@mitra42 I'll assume this is resolved as others have not been able to reproduce it, but please re-open and explain as nec?  Thanks!

closed 2019-07-18 by holta