iiab/iiab #1875: Mounting USB sticks - doesnt always get to /media/pi

mitra42, 2019-07-15:
### Expected behavior and actual behavior.
Plug in exfat ( I believe) USB stick

### Steps to reproduce the problem.
it seems to get into /media/pi/MITRA64GB when inserted, but 
sometimes on rebooting it only appears in /media/usb0/MITRA64GB

### Specifications like the version of the IIAB, operating system version, or hardware details.
IIAB v7

holta, 2019-07-15:
Does https://github.com/iiab/iiab/issues/1874#issuecomment-511503003 help?

mitra42, 2019-07-15:
Yes - it helps - somewhere I got the idea that IIAB mounted drives in one place - on /media/pi (as a link to the /media/usb/* location. The funny thing is that this DOES happen when you insert a USB stick but not when its rebooted. I'll change my code to look in each of the /media/usb*/ locations.

holta, 2019-07-15:
@mitra42 can I close this thumbdrive question?  (Or is there an ongoing/remaining issue?)

mitra42, 2019-07-16:
I guess its only an ongoing issue if anyone is expecting to find USB's at /media/pi/xxx which is where for some reason I'd expected to find them on IIAB (I forget where I read that, or if I discovered it myself) and is only now true if the USB is inserted after restart and the PI isn't restarted. ... (I've discarded the test for disks at /media/pi/xxx now so its not an issue for me)

holta, 2019-07-16:
Ok, good enough, thanks @mitra42 for explaining.

closed 2019-07-16 by holta