iiab/iiab #1877: Consistent Lokole docs across these 6 places

holta, 2019-07-15:
As summarized in #1865 & PR #1876, these 6 areas of Lokole documentation could perhaps be brought up to date and refined, if a volunteer can assist here?

0. Lokole's main doc for use with IIAB: https://github.com/iiab/iiab/blob/master/roles/lokole/README.rst
1. Lokole's main README, including the [Production setup](https://github.com/ascoderu/opwen-webapp/blob/master/README.rst#production-setup) section at the bottom:
   https://github.com/ascoderu/opwen-webapp/blob/master/README.rst
2. The 10-page Lokole-IIAB User Guide:
   https://github.com/iiab/iiab/blob/master/roles/lokole/The%20Lokole-IIAB%20User's%20Manual.pdf
3. Introductory Summaries at http://FAQ.IIAB.IO
   - Question/Answer 3:
   http://FAQ.IIAB.IO#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F
   - Question/Answer 9:
   http://FAQ.IIAB.IO#What_services_.28IIAB_apps.29_are_suggested_during_installation.3F
4. Getting Started summaries that appear automatically on http://box IIAB home pages, for IIAB implementers who've chosen to install Lokole:
   - https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-lokole.json#L6
   - https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-lokole.html
5. What appears in /etc/iiab/iiab.ini that is very rarely read, but prob still should be accurate!
   https://github.com/iiab/iiab/blob/master/roles/lokole/tasks/install.yml#L144

_(Caveat: not every doc above is preaching to the same choir of course, so the focus/audience/expectation of each can in fact be quite different!)_

holta, 2019-10-21:
@m-anish

cc @aidan-fitz @c-w 

Now that you used Lokole successfully in recent months in North India, which docs above (across these 6 places) are most important (or most in need of improvement) in your opinion?!

c-w, 2019-10-21:
@m-anish I'd be super keen on hearing your feedback on using Lokole on IIAB in North India! cc @nzola

holta, 2023-01-04:
@nzola may hopefully be able to work on this in coming weeks, thanks to @mabuelhagag's work that came together here:

- #3458

nzola, 2023-01-04:
> @nzola may hopefully be able to work on this in coming weeks, thanks to @mabuelhagag's work that came together here:
> 
> * [Enable Lokole during MEDIUM/LARGE IIAB installs #3458](https://github.com/iiab/iiab/pull/3458)

Thanks @holta for enabling Lokole for IIAB installs. Yes II will look into the docs and do my best