iiab/iiab #1879: ISSUE WHILE SAVING PERMITTED URLS

vinaybissa, 2019-07-16:
### Expected behavior and actual behavior.
Expected Behavior could be while saving the permitted urls it should whitelist those urls, but it return with error that SET -WHLIST generate a command error
### Steps to reproduce the problem.
After logging in to admin console and setting the permitted url it does not allow to save those url it returns with a error alert.
### Specifications like the version of the IIAB, operating system version, or hardware details.

holta, 2019-07-16:
Thanks @vinaybissa for the report.

Can you provide the exact clicks/actions you took within IIAB's Admin Console (http://box.lan/admin) so others can reproduce this?

vinaybissa, 2019-07-17:
Step 1 - I opened the admin console with credentials (http://box.lan/admin)
Step 2 - I Clicked on Configure from Menus
Step 3 - From left navigation pane, i clicked on Edit Permitted Urls
Step 4 - In the section where i can enter the permitted urls, i entered one url (eg. http://www.domian.com)
Step 5- Then i Clicked on Update Permitted URls list

After this step i got error in alert box like this (SET-WHLIST FAILED and reported Unexpected error in Command SET-WHLIST)

m-anish, 2019-07-17:
Is squid/dansguardian installed?

I haven't been down that road in a while, so it may well be broken. 

Can you paste the output of:
```

/etc/iiab/iiab.env
/etc/iiab/iiab.ini
/etc/iiab/local_vars.yml
```

m-anish, 2019-07-17:
tagging @jvonau @tim-moody

m-anish, 2019-07-17:
FWIW, it looks like it may easily be possible with dnsmasq (if we're not already using it)
https://superuser.com/questions/625885/dns-block-all-websites-except-for-whitelisted-sites

vinaybissa, 2019-07-17:
Hi Anish - Seems SQUID/ DANSGUARDIAN Both are installed, PFA for your reference.

Regards
Vinay

vinaybissa, 2019-07-17:
############ OUTPUT OF IIAB.ENV###########################
# This is a configuration file for IIAB
# It can be sourced in a shell script or read into an application
IIAB_BASE_PATH=/opt/iiab
IIAB_DIR=/opt/iiab/iiab
IIAB_RELEASE=7.0
IIAB_REVISION=0
IIAB_GATEWAY_ENABLED=True
LAN_IP=172.18.96.1
OS=ubuntu
OS_VER=ubuntu-18
WWWROOT=/library/www/html
STAGE=9
FQDN=box.lan
IIAB_USB_LIB_SHOW_ALL=True
IIAB_WAN_DEVICE="enp0s20u1"
IIAB_LAN_DEVICE="br0"
HOSTAPD_ENABLED=True
HTTPCACHE_ON=False

############ OUTPUT OF IIAB.INI###########################
[location]
iiab_base = /opt/iiab
iiab_dir = /opt/iiab/iiab
[version]
distribution = Ubuntu
arch = x86_64
iiab_base_ver = 7.0
iiab_branch = master
iiab_commit = 78e9a1f2c4e70245e99c9d1c7a079ec4361a4603
install_date = 2019-07-16T10:53:17Z
install_xo = none
[runtime]
iiab_stage = 9
iiab_base_ver = 7.0
iiab_revision = 0
runtime_php = /usr/lib/php
runtime_branch = master
runtime_commit = 78e9a1f2c4e70245e99c9d1c7a079ec4361a4603
runtime_date = 2019-07-17T09:40:05Z
ansible_version = 2.8.2
kernel = 4.15.0-54-generic
memory_mb = 5846
swap_mb = 2047
product_id = 3C611F00-732C-11E3-9115-F44D306D8FD5
gw_active = True
internet_available = True
is_rpi = False
first_run = False
local_tz = Asia/Kolkata
no_NM_reload = False
is_F18 = False
FQDN_changed = False
[iiab-admin]
name = iiab-admin
description = "Admin User"
iiab_admin_user = iiab-admin
[openvpn]
name = OpenVPN
description = OpenVPN enables live/remote support by connecting machines anywhere on the Internet, via a middleman server, using Virtual Private Network (VPN) techniques to create secure connections.
enabled = False
handle = 
cron_enabled = False
server = xscenet.net
server_virtual_ip = 10.8.0.1
server_port = 1194
[mysql]
name = MySQL
description = "MySQL is a widely used free and open source (GPLv2) database, offered by most web hosting services, on a diversity of platforms."
enabled = True
[... 465 lines cut ...]
# teamviewer_enabled: False

# Unmaintained
# docker_install: False
# docker_enabled: False

# Unmaintained
# schooltool_install: False
# schooltool_enabled: False

# Unmaintained
# debian_schooltool_install: False
# debian_schooltool_enabled: False

# Unmaintained (consider Calibre or Calibre-Web above?)
# pathagar_install: False
# pathagar_enabled: False

# Unmaintained
# sugar_stats_install: False
# sugar_stats_enabled: False

# Unmaintained
# xovis_install: False
# xovis_enabled: False

holta, 2019-07-17:
Thank @vinaybissa 

I hope @tim-moody can confirm today, and possibly offer alternatives.

m-anish, 2019-07-17:
@vinaybissa instead of copy pasting the contents, you can also attach a file on github :)

holta, 2019-07-17:
> @vinaybissa instead of copy pasting the contents, you can also attach a file on github :)

Usually this is easiest on everyone:

```
pastebinit < /etc/iiab/iiab.env
pastebinit < /etc/iiab/iiab.ini
pastebinit < /etc/iiab/local_vars.yml
```

In fact the command `/use/bin/iiab-diagnostics` handles this and more, if ever necessary!

tim-moody, 2019-07-17:
it will be looking for
/etc/squid3/sites.whitelist.txt
doubt this will be there unless squid is installed an enabled

vinaybissa, 2019-07-18:
@tim-moody  Squid is installed and enabled but the files its looking for at this location
/etc/squid/sites.whitelist.txt not at what you mentioned earlier

tim-moody, 2019-07-18:
OK. the problem is that iiab-admin-console maintains a varible squid_service that has 'squid3' for ubuntu, whereas iiab uses the variable proxy (could have a better name) that has 'squid'. 

 iiab-admin-console should use proxy

holta, 2019-07-18:
@vinaybissa can you review https://github.com/iiab/iiab-admin-console/pull/243 do see if that solves your problem?

As @tim-moody has now included into IIAB's mainline code.

Specifically, try the following:
- sudo su -
- cd /opt/iiab/iiab-admin-console
- git pull
- ./install

And then try again... (does this help?)

vinaybissa, 2019-07-19:
@holta  I have done that after installing the admin-console, now i am getting new error when opening admin-console, ERROR - [ GET-VARS FAILED and reported IIAB-CMDSRV failed to initialize]

vinaybissa, 2019-07-19:
@holta  & @tim-moody  - Issue has been resolved, but i am confused with the flow of given access to certain urls, to do this i have done steps as given below
Step 1- Logged in to Admin Console
Step 2 - Clicked on Configure -> Internet Filtering/Caching -> Checked (Restrict Web Page Access & Block all HTTPS Access ) so that i can access only few websites
Step 3- Then i clicked on Edit Permitted URL ( there are already many urls in the list)
Step 4-  I have added one more url at the end of the list.
Step 5-  Then i clicked upon Update Permitted Url list
Step 6 - Then i clicked save configuration
Step 7 - Then i clicked Install Configured Options
Step 8 - As it takes time to be successful state, so i waited for few minutes then it becomes successful.
Step 9 - After than i reboot the server
Step 10 - I have now internet access on my server, as i blocked all https sites so as usual they are not running, but i have given access to certain urls, but they are also not running.
Step 11- I reviewed the Edit Permit Url Section, as i can see the url i added in the end of list, they are not their only default entries are their, After running the Install Configured Option the list i have provided will be removed and the default entries remains.

tim-moody, 2019-07-19:
@vinaybissa can you try repeating steps 3 to 5 without the remaining steps and let us know what happened. I think step 7 reset your list.

vinaybissa, 2019-07-19:
@tim-moody By doing steps 3 to 5 i can see the latest added entries but still newly added urls are not accessible.

tim-moody, 2019-07-19:
@vinaybissa can you reboot just to be sure.

vinaybissa, 2019-07-19:
@tim-moody  rebooted the system, entries are updated, but still i am not able to access these url

tim-moody, 2019-07-19:
looks like a bug, squid or iiab. someone will have to investigate.

tim-moody, 2019-07-19:
to be clear are you trying to permit https://www.domain.com or http://www.domain.com?

what happens if you try to permit neverssl.com?

vinaybissa, 2019-07-19:
@tim-moody  i have added both type of entries ( 2 entries of https and 1 entry for http protocol) but none of them are working

tim-moody, 2019-07-19:
@vinaybissa any hint in /var/log/squid/access.log

tim-moody, 2019-07-19:
can you attache /etc/squid/sites.whitelist.txt

jvonau, 2019-07-19:
Can you post the contents of /etc/iiab/config_vars.yml please?

holta, 2019-07-19:
@vinaybissa running the `iiab-diagnostics` command to give us a more complete picture, as follows:
```
iiab-diagnostics /etc/squid/sites.whitelist.txt /var/log/squid/access.log
```
(If so, send us the resulting URL, thanks!)

vinaybissa, 2019-07-20:
@jvonau  here is the content of this file
http://paste.ubuntu.com/p/N3nmQNnHdM/

vinaybissa, 2019-07-20:
@tim-moody Please find attached for your reference
[sites.whitelist.txt](https://github.com/iiab/iiab/files/3413305/sites.whitelist.txt)

vinaybissa, 2019-07-20:
@holta  for your review
http://paste.ubuntu.com/p/3yfkWDCKXf/

vinaybissa, 2019-07-20:
Only thing i observed is it blocks all http urls weather they are in the sites.whitelist.txt or not. If i block all https web access from admin console then it blocks all https which is ok but the https url which i mentioned in sites.whitelist.txt should run but they are also not opening

tim-moody, 2019-07-20:
I think this logic is faulty:

{% if gw_squid_whitelist %}
http_access deny !whitelist
{% endif %}

http_access deny !Safe_ports
http_access allow school
http_access allow localhost
http_access deny all

to me this says block if not in whitelist and then later block all
it doesn't say allow whitelist

tim-moody, 2019-07-20:
@vinaybissa you could try editing /etc/squid/squid-iiab.conf and remove or comment out

http_access deny all

jvonau, 2019-07-21:
Think @tim-moody is on to something:
> 1563520151.855    729 127.0.0.1 TCP_DENIED/403 4251 GET http://quizacademy.org/ - HIER_NONE/- text/html

The source would be Dansgaurdian as shown by 127.0.0.1 as the source ip address in the log. The issue I see is in [squid-iiab.conf](https://github.com/iiab/iiab/blob/master/roles/network/templates/squid/squid-iiab.conf.j2) where the acl  for localhost is # out (line 38 #acl localhost src 127.0.0.1/32) so there would be no 'allow' present for that traffic.

The other issue is when 'save configuration' writes the whitelist file but as soon as 'Install Configured Options' is run [4-server-options](https://github.com/iiab/iiab/blob/master/roles/4-server-options/tasks/main.yml) is where [squid.yml](https://github.com/iiab/iiab/blob/master/roles/network/tasks/squid.yml) is referenced and we supply a new template for whitelist.txt, wiping out the user's custom settings.

jvonau, 2019-07-21:
> but the https url which i mentioned in sites.whitelist.txt should run but they are also not opening

The block https is done in iiab-gen-iptables also. 
`if [ "$gw_block_https" == "True" ]; then $IPTABLES -A FORWARD -p tcp ! -d {{ lan_ip }} --dport 443 -j DROP fi`

That makes blocking all or none, I don't think squid is currently setup to handle https requests, and would be missing the iptable rule for the interception.

holta, 2019-07-21:
@vinaybissa, I've merged in @jvonau's change here, so that you can immediately test if and what this improves:

#1879 Don't force templates, only create once in squid.yml

To test this, I believe you need to run the following:
```
sudo su -
cd /opt/iiab/iiab
git pull
./runrole 4-server-options
```
And it probably doesn't hurt to then run the following too:
```
./iiab-network
```
And then test filtering of `http` sites (remembering that all `https` sites are blocked for now!)

tim-moody, 2019-07-21:
this fixes a secondary problem, but the primary

holta, 2019-07-22:
> this fixes a secondary problem, but the primary

What should we move towards releasing here?

@vinaybissa can you provide feedback on the several questions & recommendations @tim-moody & @jvonau posed over the weekend?

holta, 2019-07-24:
> @vinaybissa you could try editing /etc/squid/squid-iiab.conf and remove or comment out
> 
> http_access deny all

@vinaybissa what is the result please, when you try this?

(Not sure if a reboot is required or not!)

vinaybissa, 2019-08-01:
> @vinaybissa you could try editing /etc/squid/squid-iiab.conf and remove or comment out
> 
> http_access deny all

@tim-moody - i have tried this now its filtering url having http, but what about https url ?

vinaybissa, 2019-08-01:
> > @vinaybissa you could try editing /etc/squid/squid-iiab.conf and remove or comment out
> > http_access deny all
> 
> @vinaybissa what is the result please, when you try this?
> 
> (Not sure if a reboot is required or not!)

@holta . Its working fine now for filtering http urls, but is there any way to filter https url ?

jvonau, 2019-08-01:
> > @vinaybissa you could try editing /etc/squid/squid-iiab.conf and remove or comment out
> > http_access deny all
> 
> @tim-moody - i have tried this now its filtering url having http, but what about https url ?

Did you apply the git pull/fix as recommended above first? Disabling 'deny all' is not really recommended by [upstream](http://www.squid-cache.org/Doc/config/http_access/). About https, as explained above. Did you have a feature request and time to test? In the end which websites do you want direct access to?

vinaybissa, 2019-08-01:
> Did you apply the git pull/fix as recommended above first? Disabling 'deny all' is not really recommended by [upstream](http://www.squid-cache.org/Doc/config/http_access/). About https, as explained [...]

@jvonau - i have done with process which you described, now filtering is working for http urls, but still issue with https (Interestingly Youtube and sugarlabs.org which are hosted on https are working, rest https are not working)

holta, 2019-08-01:
Thanks @vinaybissa!  Can you also:
1. tell us if the youtube videos are actually playing?
2. reboot
3. test both kinds of https sites (those that were blocked and those that were not like youtube.com & sugarlabs.org)
4. re-run /usr/bin/iiab-diagnostics and then post the resulting URL ?

holta, 2019-12-21:
> Thanks @vinaybissa! Can you also:
> 
> 1. tell us if the youtube videos are actually playing?
> 2. reboot
> 3. test both kinds of https sites (those that were blocked and those that were not like youtube.com & sugarlabs.org)
> 4. re-run /usr/bin/iiab-diagnostics and then post the resulting URL ?

@vinaybissa can you confirm where you're at?

Or should I close this now?

Thanks!

holta, 2021-08-24:
@vinaybissa IIAB's Squid implementation was overhauled over the last 10 days, so please do try it out if this is still useful to you?  

Thanks for your input if you have a moment!

Quick summary of the changes:

1. https://github.com/iiab/iiab/blob/master/roles/network/templates/squid/squid.conf.j2
2. PR #2948
3. PR #2963

(Finally, feel free to open this ticket if necessary!)

PR #1893 (2019-07-21): don't force templates, only create once in squid.yml
bug fix #1879
closed 2021-08-24 by holta