iiab/iiab #1881: http://box should show footnote metadata as Articles: 5,733,761 (NOT Articles: 5733761)

holta, 2019-07-16:
Content Pack metadata from 2017 and 2018 was far more readable to educators, implementers, administrators and students alike &mdash; allowing all to quickly understand their content/choices.

_How hard would it be to restore these commas, in an automated way this time?_

Across most/all major metadata fields shown by IIAB home pages, for example:
```
Size: 78.1G, Articles: 5733761, Media: 4680359, Date: 2018-10-24
```
Which really should be restored to appear as follows:
```
Size: 78.1G, Articles: 5,733,761, Media: 4,680,359, Date: 2018-10-24
```
I'm not so concerned about some Hispanic/Francophone countries that prefer appearances as follows:
```
Size: 78.1G, Articles: 5 733 761, Media: 4 680 359, Date: 2018-10-24
```
FWIW the official notation/tradition in most of these countries/regions is to keep 4-digit numbers all together e.g. '1984' (yes, if possible!) instead of '1,984' or '1 984' (avoid, if possible!)

holta, 2019-07-24:
Proposed Solution for IIAB 7.0:
iiab/iiab-admin-console#249

tim-moody, 2019-07-24:
'{:,}'.format(float(string))

holta, 2019-07-24:
> '{:,}'.format(float(string))

I did pretty much exactly that yesterday in Python, but this prevents people from changing their locales.

Then I also realized Sneakernets-of-Alexandria dynamic / on-the-fly content types become substantially more resilient with a Javascript solution.

tim-moody, 2019-07-24:
It is fragile to apply formatting to a string without knowing its semantic value:
January 15, 2019
License number 12345
1.2345K
10^2.3456
there is no assurance that a future arbitrary field will be handled correctly and the current code wouldn't support it anyway
the problem of locales is not addressed by either approach, but at least format is supposed to know about locale

holta, 2019-07-24:
1) In my tests against many dates/years and decimal points etc on IIAB home pages, I found no such manglings with https://github.com/iiab/iiab-admin-console/pull/249 as borrowed from https://stackoverflow.com/questions/2901102/how-to-print-a-number-with-commas-as-thousands-separators-in-javascript

But you're right if such exist, this should definitely be cleaned up.  It's likely fair to mandate that all footnote numbers will continue to be sizes or dates for now.  Unless a future reference librarian use case really demands ISBN or some such, but we're not facing anything like that today.

2) Locales will be a very real issue in future.  Educational systems get incredibly particular about the formats of dates and numbers, even though I personally think commas are quite good enough for IIAB 7.0.

But wherever possible, we should put every design we can on those paths that most respect local appropriation / authentic local feelings of genuine ownership.

holta, 2019-07-25:
PR iiab/iiab-admin-console#250 implements this with more safeguards, and across all browser locales.

closed 2019-07-25 by holta