iiab/iiab #1885: IIAB's Captive Portal can be extremely frustrating on recent versions of Android

holta, 2019-07-18:
Would it be better to keep IIAB's Captive Portal off by default for now, as more and more people adopt these recent versions of Android?

(Or might there perhaps be an option for Captive Portal to "turn itself off" when it encounters recent versions of Android like 8.x and higher perhaps?)

Recap: several phones were tested today, each facing these same UX dead-ends (IIAB's main page http://box loads but then its menu items do not appear...and then clicking in the top-right of Android's "Sign In" banner [which many people do not find, to accept the offline connection] causes the whole page to disappear).  This included Android 8.1.0 on a Moto phone and and a version of Android that's apparently more recent on a Pixel phone.

Background: http://FAQ.IIAB.IO Item 12 ("My Android device says "Connected, no Internet" and won't browse http://box")

tim-moody, 2019-07-18:
if you get to the home page, you need to click on ellipsis and select real browser. not sure when this started, but probably 8 or 9 for sure.

holta, 2019-07-18:
> if you get to the home page, you need to click on ellipsis and select real browser. not sure when this started, but probably 8 or 9 for sure.

Much more serious and negative things are happening, that did not happen in the past.

When testing on 2 different phones and 3 different Raspberry Pi's (RPi 3 B+ and 4) today &mdash; this has now become a confounding UX dead-end &mdash; and much more confusing than nothing at all:

- Clicking the ellipsis on the right side of Android 8.1.0's small "Sign in" banner on top gives 2 options (clicking the lower of the 2 options, i.e. "Use the network as is", causes visible-but-unpopulated IIAB page to disappear)
- Browsing to pages like http://mit.edu does nothing after that
- Browsing to pages like http://box and http://box.lan and http://172.18.96.1 temporarily works
- The entire connection is lost within about 30 seconds &mdash; Android reverts to the 4G/LTE mobile data connection (if available).  Not sure what's causing this last quagmire (tested on 3 different Raspberry Pi IIAB's) but it's quite a dramatic regression compared to 2018.

As such turning off Captive Portal (by default) would appear to be quite urgent for IIAB 7.0's release.  Unless these nightmarish Android 8/9 experiences can be contained somehow?  Likely also affecting Android 10 arriving in "August 2019."

holta, 2019-07-18:
Worked around with PR #1887 but needs more investigation &mdash; to try to understand why WiFi connection was being dropped entirely after 30-45 seconds (certainly on Android 8.1.0) &mdash; _does this disabling of captive_portal_install/enabled variables in local_vars.yml put an end to that serious glitch?_

holta, 2019-07-25:
A more positive story on Android 9 &mdash; @tim-moody faced a "Sign in" (?) banner on top, with an ellipsis (vertical 3 dots) on the right which offered him 3 options:

- "Disconnect"
- "Connect"
- "Open in Browser"

He clicked on this 3rd option (Open in Browser) which successfully opened http://box