iiab/iiab #1907: android not rendering main menu

mitra42, 2019-07-31:
### Expected behavior and actual behavior.
I'm using a cheap chinese pad, which appears to be running "Firefox OS", although it says during boot up that its Android 4.1.  It is connected directly to the WiFi point of an IIABv7. It has the Internet Archive code, but before I get there I'm going in the browser to http://box.lan

It redirects to http://box.lan/home 
BUT the menu displays a bunch of places "Attempting to load en-oob" and so on. 

### Steps to reproduce the problem.
I don't know how /if this is reproducable, the same URL works fine in my iPhone, *but* I did see someone else have the same problem at Dweb camp, but I didn't have a chance to investigate further. 

I also can't get it to run the Internet Archive server, so there may indeed be problems with the OS (e.g. Javascript might be off by default)

Note - I don't particularly care about solving this problem - I'm just testing this pad, but its the second time (on different devices) I've seen the exact same bug, 

I'm happy to help pin this down but I've never used this OS and cant even figure out how to get out of the browser to find out things like version numbers.

### Specifications like the version of the IIAB, operating system version, or hardware details.
IIAB v7 or RPI 3+
Cheap tablet on Firefox OS

holta, 2019-07-31:
Thanks @mitra42 

Firefox OS was discontinued more than 4 years ago according to https://en.wikipedia.org/wiki/Firefox_OS but let's try to uncover if this a broader pattern that needs addressing ?

mitra42, 2019-07-31:
Totally - it might not be a platform worth supporting. 
I think Dinesh (the person who had the other failure I saw) will be coming by the Archive tomorrow or Friday and I'll check if he was using Firefox_OS as well, or if it was Android.

jvonau, 2021-08-29:
#2739