iiab/iiab #1911: BIG-sized IIAB thrashes (memory/swap issues) on 1GB RPi 3 and RPi 4

holta, 2019-08-01:
Huge thanks to @georgejhunt who hopes to investigate this in coming days!

I'm definitely willing to defer our IIAB 7.0 release by several days if it turns out progress is possible here &mdash; i.e. re-tuning what's contained within BIG-sized IIAB's so that it runs smoothly &mdash; as the reasonably solid demo experience that most all newcomers expect!

_(Is it possible 1 or 2 of the new IIAB Apps from recent months is/are causing this?)_

georgejhunt, 2019-08-02:
Maybe everything is working as it should:

- Time to boot for BIG is 1:54 on 1GB, but after everything settles down, there is 430MB Available, and there is 269 MB in the Swap used, and used is 418MB.
- Time to boot for Big is 0:48 on 4GB, there is no swap used, used is 714MB, and available is 3096MB

I would expect data in swap to return to main memory when a specific application is used. The 430MB available seems like it might be enough for normal function for each of the applications one at a time.

These results are without internet_archive, which I cannot install properly.

holta, 2019-08-02:
Many thanks @georgejhunt 

I'm asking others to test BIG-sized IIAB installs on Lite/Headless & Desktop OS's both, over the course of the wkd especially

Also so we can all come to resolution on #1915.

So that final release can happen in the coming 10 days.

georgejhunt, 2019-08-02:
Looking at the items in swap, there is not a single application that stands out as a memory hog.
Candidates for removal might be gitea, lokole. But turning them off would not be enough!

root@box:~# smem -t -p
  PID User     Command                         Swap      USS      PSS      RSS 
 1884 pi       /usr/bin/ssh-agent -s          0.06%    0.00%    0.00%    0.10% 
  942 root     /usr/bin/vncserver-x11-serv    0.04%    0.00%    0.00%    0.18% 
  943 root     /bin/bash /home/lokole/stat    0.03%    0.00%    0.00%    0.17% 
 1172 postgres postgres: logger               0.21%    0.00%    0.01%    0.13% 
 1883 pi       xcompmgr -aR                   0.05%    0.00%    0.01%    0.20% 
 1262 www-data /usr/sbin/apache2 -k start     0.67%    0.00%    0.01%    0.16% 
 1866 pi       /bin/sh /usr/lib/raspi-conf    0.01%    0.01%    0.01%    0.11% 
 1024 root     /usr/sbin/syslogd --no-forw    0.01%    0.01%    0.01%    0.14% 
  795 uuidd    /usr/sbin/uuidd --socket-ac    0.02%    0.01%    0.01%    0.21% 
  386 root     /usr/sbin/rngd -r /dev/hwrn    0.01%    0.01%    0.01%    0.14% 
  672 root     /usr/bin/hciattach /dev/ser    0.01%    0.01%    0.01%    0.06% 
  395 avahi    avahi-daemon: chroot helper    0.04%    0.00%    0.01%    0.12% 
  975 vnstat   /usr/sbin/vnstatd -n           0.01%    0.01%    0.02%    0.15% 
  371 nobody   /usr/sbin/thd --triggers /e    0.01%    0.01%    0.02%    0.20% 
  697 root     /usr/bin/bluealsa              0.07%    0.01%    0.02%    0.26% 
  348 root     /usr/sbin/cron -f              0.02%    0.01%    0.02%    0.24% 
 1085 dnsmasq  /usr/sbin/dnsmasq -x /run/d    0.03%    0.02%    0.02%    0.19% 
 1181 postgres postgres: walwriter            0.52%    0.01%    0.03%    0.19% 
 1183 postgres postgres: stats collector      0.19%    0.01%    0.03%    0.19% 
  377 root     /usr/sbin/alsactl -E HOME=/    0.02%    0.03%    0.03%    0.17% 
  681 root     /usr/lib/bluetooth/bluetoot    0.05%    0.02%    0.03%    0.30% 
 1081 root     /bin/login -f                  0.04%    0.03%    0.03%    0.22% 
 1180 postgres postgres: background writer    0.52%    0.01%    0.04%    0.23% 
 1806 pi       /usr/bin/ssh-agent x-sessio    0.00%    0.03%    0.04%    0.14% 
 1144 root     /opt/iiab/kiwix/bin/kiwix-s    0.27%    0.04%    0.04%    0.04% 
  362 avahi    avahi-daemon: running [box-    0.02%    0.03%    0.04%    0.27% 
  979 mosquitto /usr/sbin/mosquitto -c /etc    0.09%    0.02%    0.04%    0.42% 
  439 root     /sbin/dhcpcd -q -b             0.02%    0.05%    0.05%    0.19% 
 1388 pi       (sd-pam)                       0.26%    0.02%    0.05%    0.24% 
  464 lp       /usr/lib/cups/notifier/dbus    0.04%    0.04%    0.05%    0.33% 
 1708 pi       /usr/bin/dbus-daemon --sess    0.03%    0.03%    0.05%    0.35% 
  367 root     /usr/sbin/hostapd -P /run/h    0.00%    0.05%    0.06%    0.25% 
 1020 ntp      /usr/sbin/ntpd -p /var/run/    0.05%    0.06%    0.06%    0.27% 
  443 root     /usr/bin/monit -c /etc/moni    0.04%    0.06%    0.06%    0.22% 
 1677 root     /usr/bin/vncagent service 1    0.04%    0.05%    0.06%    0.32% 
  462 lp       /usr/lib/cups/notifier/dbus    0.02%    0.05%    0.06%    0.34% 
 1477 pi       -bash                          0.11%    0.04%    0.06%    0.29% 
 1821 pi       /usr/lib/gvfs/gvfsd            0.05%    0.05%    0.07%    0.55% 
 2011 pi       /usr/lib/gvfs/gvfs-goa-volu    0.00%    0.05%    0.07%    0.52% 
 1184 postgres postgres: logical replicati    0.54%    0.01%    0.07%    0.29% 
  954 debian-transmission /usr/bin/transmission-daemo    0.14%    0.05%    0.07%    0.44% 
  354 messagebus /usr/bin/dbus-daemon --syst    0.04%    0.05%    0.07%    0.34% 
  376 root     /sbin/wpa_supplicant -u -s     0.00%    0.06%    0.07%    0.31% 
 2015 pi       /usr/lib/gvfs/gvfs-mtp-volu    0.00%    0.06%    0.08%    0.54% 
 1826 pi       /usr/lib/gvfs/gvfsd-fuse /r    0.05%    0.05%    0.08%    0.63% 
 1182 postgres postgres: autovacuum launch    0.54%    0.02%    0.08%    0.32% 
 1531 www-data /usr/sbin/apache2 -k start     1.48%    0.05%    0.09%    0.39% 
 2006 pi       /usr/lib/gvfs/gvfs-gphoto2-    0.00%    0.07%    0.09%    0.56% 
 1018 root     /usr/sbin/lightdm              0.06%    0.07%    0.09%    0.61% 
 1025 postgres /usr/lib/postgresql/11/bin/    0.53%    0.05%    0.09%    0.62% 
 1998 pi       /usr/lib/menu-cache/menu-ca    0.00%    0.06%    0.09%    0.63% 
 1075 root     /usr/sbin/sshd -D              0.04%    0.04%    0.09%    0.50% 
  449 root     /usr/lib/policykit-1/polkit    0.04%    0.07%    0.09%    0.65% 
  433 root     /usr/sbin/cups-browsed         0.09%    0.07%    0.09%    0.53% 
 1012 root     /lib/systemd/systemd-udevd     0.11%    0.09%    0.10%    0.32% 
  369 root     /lib/systemd/systemd-logind    0.05%    0.06%    0.10%    0.52% 
 1528 www-data /usr/sbin/apache2 -k start     1.46%    0.06%    0.10%    0.42% 
 1179 postgres postgres: checkpointer         0.51%    0.06%    0.11%    0.35% 
 2019 pi       /usr/lib/gvfs/gvfs-afc-volu    0.00%    0.09%    0.11%    0.63% 
 1681 root     lightdm --session-child 14     0.01%    0.08%    0.11%    0.73% 
 1699 pi       /usr/bin/lxsession -s LXDE-    0.16%    0.06%    0.12%    0.89% 
 1850 pi       lxpolkit                       0.08%    0.08%    0.12%    0.80% 
  364 root     /usr/sbin/cupsd -l             0.11%    0.11%    0.12%    0.48% 
  958 root     /library/lokole/venv/bin/py    1.41%    0.08%    0.13%    0.64% 
[... cut ...]
  945 gitea    /library/gitea/bin/gitea-1.    6.06%    4.68%    4.68%    4.68% 
  940 pi       node-red                       0.38%    4.41%    4.70%    5.27% 
-------------------------------------------------------------------------------
  101 18                                     62.03%   42.92%   47.43%   85.07% 
root@box:~# root@box:~# free
              total        used        free      shared  buff/cache   available
Mem:         946668      424964      286364       29836      235340      426648
Swap:        511996      268544      243452
root@box:~#

holta, 2019-08-02:
> But turning them off would not be enough!

Not be enough to achieve what goal exactly?

georgejhunt, 2019-08-02:
I think the X2 startup time between 1GB and 4GB is what I would like to
change. I have not really tested any of the applications, but the gui is
really sluggish, right after boot.

It takes about 10 seconds to bring up chrome on a 1 GB, and about 6 seconds
on a 4GB.

I didn't initially understand why the difference. But when I used smem to
list what's in swap, I find over 100MB  of the newly started
chromium-browser in swap.  So maybe once it starts using swap, it puts as
much as it can there all the time. After starting chromium-browser, there
is still about 300MB free.

I think the OS is making all the correct decisions, and we just need to
warn users that BIG is not a good fit for 1GB rpi.

holta, 2019-08-02:
> I think the X2 startup time between 1GB and 4GB is what I would like to change.

> I think the OS is making all the correct decisions, and we just need to warn users that BIG is not a good fit for 1GB rpi.

Do you know if some of these slownesses apply to 1GB machines even without IIAB?

Whereas up til now Kids On Computers has been very satisfied with 1GB client machines in school labs...

...should they now consider 2GB RPi 4's if these become more widely available?  (Pricing is not irrelevant if course, as 4GB RPi 4's and pricier computers are just not possible for every kid that might want one.

georgejhunt, 2019-08-02:
I remember that the 7/10/19 desktop was pretty snappy, before I applied
IIAB.

Is that your question?

holta, 2019-08-02:
> I remember that the 7/10/19 desktop was pretty snappy, before I applied IIAB. Is that your question?

Certainly that helps.

As there's a flood of NGO's that (always) want client-machine recommendations alongside their IIAB installs.  The better information we can give them upfront, the less hassles we will have later.

holta, 2019-08-08:
@georgejhunt discovered that IIAB install onto 1GB RPi 4 was dead-slow on Raspbian Desktop after reboot, likely due to ~400MB swap.

Even despite there being ~300MB RAM free.

Conclusion/Theory: the moment we begin using swap, things go downhill.

Idea 1: @tim-moody asks: should we possibly turn off swap entirely, after IIAB's software install completes?  @georgejhunt would like to investigate this radical notion in coming days/weeks if he has time.

Idea 2: Should we turn off services like internetarchive, Gitea, Kolibri ?

holta, 2019-08-08:
Idea 3: IIAB's installer could possibly become more intelligent &mdash; strongly dissuading people from installing these if we (IIAB's 1-line install script) detected that they have only 1GB RAM!

georgejhunt, 2019-08-08:
A data point (on pi4 1GB):

1. I turned off swap "dphy-swapfile swapoff" (also setting /etc/dphy-swapfille ->"CONF_MAXSWAP=0")
2. After turning off swap, boot time to GUI was 5 min.
3. Turned swap back on, boot time was 98 seconds, cursor had hard time keeping up with mouse.
4. Disabled gitea, and kolibri. Boot time was 52 sec. and cursor movement was adequate.

holta, 2019-08-09:
Tangentially, see also this list that @mitra42 is compiling while profiling to uncover where slowdowns are happening during (re)installs of 'internetarchive' etc:

Package dependency warnings {META}
internetarchive/dweb-mirror#234

mitra42, 2019-08-09:
THat's the warnings (which don't actually slow things down) the next one https://github.com/internetarchive/dweb-mirror/issues/235 covers trying to speed things up. (and there will be another task covering splitting the dependencies to remove transports not used in dweb-mirror from that dependency (e.g. IPFS) but that's a more complex process that will take a few weeks.

jvonau, 2019-08-10:
A little feedback on watching runrole internetarchive while on a second ssh session that was really slow to respond to keyboard input, little top observation after testing #1921 on .14 that has 'large' installed.
> Tasks: 179 total,   3 running, 176 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.8 us,  2.3 sy, 34.6 ni, 61.8 id,  0.5 wa,  0.0 hi,  0.1 si,  0.0 st
MiB Mem :    924.5 total,    289.6 free,    288.0 used,    346.8 buff/cache
MiB Swap:    500.0 total,     45.4 free,    454.6 used.    559.0 avail Mem  

The keyboard response returned to normal after runrole completed, but still shows about the same amount of swap being used.

jvonau, 2019-08-10:
> root@box:/opt/iiab/internetarchive# systemctl status internetarchive
● internetarchive.service - Internet Archive Universal Library service
   Loaded: loaded (/etc/systemd/system/internetarchive.service; enabled; vendor
   Active: active (running) since Sat 2019-08-10 05:19:28 CDT; 51min ago
 Main PID: 23926 (node)
    Tasks: 11 (limit: 1599)
   Memory: 57.4M
   CGroup: /system.slice/internetarchive.service
           └─23926 /usr/bin/node ./internetarchive -sc

> root@box:/opt/iiab/internetarchive# systemctl status kolibri
● kolibri.service - Kolibri
   Loaded: loaded (/etc/systemd/system/kolibri.service; enabled; vendor preset:
   Active: active (running) since Thu 2019-08-08 13:06:43 CDT; 1 day 17h ago
 Main PID: 789 (kolibri)
    Tasks: 55 (limit: 1599)
   Memory: 48.4M
   CGroup: /system.slice/kolibri.service
           └─789 /usr/local/kolibri/bin/python2 /usr/local/kolibri/bin/kolibri
 
> ● gitea.service - Gitea (Git with a cup of tea)
   Loaded: loaded (/etc/systemd/system/gitea.service; enabled; vendor preset: e
   Active: active (running) since Sat 2019-08-10 02:56:39 CDT; 3h 17min ago
 Main PID: 31783 (gitea-1.9.0)
    Tasks: 14 (limit: 1599)
   Memory: 49.8M
   CGroup: /system.slice/gitea.service
           └─31783 /library/gitea/bin/gitea-1.9.0 web -c /etc/gitea/app.ini

top sorted 
> Tasks: 178 total,   1 running, 177 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1.2 us,  0.7 sy,  0.0 ni, 98.0 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    924.5 total,    279.1 free,    281.4 used,    364.0 buff/cache
MiB Swap:    500.0 total,     54.0 free,    446.0 used.    564.4 avail Mem 
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND         
31783 gitea     20   0  981308  41964   3840 S   0.0   4.4   1:10.06 gitea-1.9.0     
 1021 mysql     20   0  731088   1684      0 S   0.0   0.2   2:41.06 mysqld          
  789 kolibri   20   0  511376  45932   2276 S   4.3   4.9  80:45.42 kolibri         
  880 mongodb   20   0  328076  35484  34824 S   0.7   3.7  16:45.40 mongod          
 2167 root      20   0  298872   4776   1708 S   0.0   0.5   2:52.39 BonJour         
24001 www-data  20   0  225288   9380   1948 S   0.0   1.0   0:00.04 apache2         
24002 www-data  20   0  225288   9396   1964 S   0.0   1.0   0:00.04 apache2         
24003 www-data  20   0  225288   9392   1960 S   0.0   1.0   0:00.03 apache2         
24004 www-data  20   0  225288   9396   1964 S   0.0   1.0   0:00.05 apache2         
23975 root      20   0  223460  18672  12888 S   0.0   2.0   0:00.33 apache2         
 2036 root      20   0  216660   6852   2856 S   0.0   0.7   3:06.99 kalite          
23926 root      20   0  185600  65536  25540 S   0.3   6.9   0:11.53 node            
 2409 root      20   0  165264  10288   5960 S   0.0   1.1   0:41.53 node            
  868 pi        20   0  156080   2900    764 S   0.0   0.3   0:13.70 node-red        
 1194 root      20   0  128960   1440    856 S   0.0   0.2   0:26.72 iiab-cmdsrv     
  319 root      20   0  107508   3756    796 S   0.0   0.4   0:17.38 python          
 1210 postgres  20   0   89812   1836   1552 S   0.0   0.2   0:09.62 postgres        
 1212 postgres  20   0   89700   1028    844 S   0.0   0.1   0:00.99 postgres        
 1207 postgres  20   0   89520    764    640 S   0.0   0.1   0:00.21 postgres        
  923 postgres  20   0   89412   1212   1008 S   0.0   0.1   0:16.12 postgres        
 1208 postgres  20   0   89412    552    428 S   0.0   0.1   0:04.77 postgres

holta, 2019-08-16:
@georgejhunt can tell if PR
https://github.com/iiab/iiab/pull/1923 mitigates the worst of the swap issues in 1GB RAM machines like the RPi 4?

georgejhunt, 2019-08-16:
No

I doubt that it will have much effect. My experience is that the swapping
itself is the slow down, rather than :"out of memory" cannibalization  of
running programs, which would occur if swap was too small.

mitra42, 2019-10-04:
PR #1982 should fix any memory issues with internetarchive install (though its not clear from the thread above if that was the issue during install and / or if the topic being addressed is memory thrashes during install, during startup, or during operation. )

holta, 2019-10-04:
Looking good.  Am closing this issue for now.  Feel feel free to reopen as nec if it flares up again!

closed 2019-10-04 by holta