iiab/iiab #1915: internet_archive fails to install

georgejhunt, 2019-08-01:
three attempts. 
First two - using the "sudo iiab" approach.
The last time I became root, and tried again.
Log:
  
2019-08-01 12:26:27,590 p=9208 u=root |  TASK [internetarchive : Run yarn install to get needed modules (CAN TAKE ~15 MINUTES)] ***
2019-08-01 14:37:10,832 p=9208 u=root |   [WARNING]: Consider using 'become', 'become_method', and 'become_user' rather
than running sudo

2019-08-01 14:37:37,172 p=9208 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["sudo", "yarn", "add", "@internetarchive/dweb-archive", "@internetarchive/dweb-mirror"], "delta": "1:26:33.015192", "end": "2019-08-01 13:53:00.948828", "msg": "non-zero return code", "rc": -9, "start": "2019-08-01 12:26:27.933636", "stderr": "warning @internetarchive/dweb-archive > @internetarchive/dweb-transports > gun > text-encoding@0.7.0: no longer maintained\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > boom@7.3.0: This module has moved and is now available at @hapi/boom. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > joi@14.3.1: This module has moved and is now available at @hapi/joi. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > ipfs-mfs > joi@14.3.1: This module has moved and is now available at @hapi/joi. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi@18.1.0: This module has moved and is now available at @hapi/hapi. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi > boom@7.3.0: This module has moved and is now available at @hapi/boom. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi > joi@14.3.1: This module has moved and is now available at @hapi/joi. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi@16.7.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > boom@5.2.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > joi@11.4.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > wolkjs > @trust/webcrypto > text-encoding@0.6.4: no longer maintained\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > gun > emailjs > emailjs-mime-codec > text-encoding@0.7.0: no longer maintained\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > boom > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi-pino > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > joi > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > boom > hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > joi > hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > ipfs-unixfs-importer > left-pad@1.3.0: use String.prototype.padStart()\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > joi > topo@3.0.3: This module has moved and is now available at @hapi/topo. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi > topo@3.0.3: This module has moved and is now available at @hapi/topo. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > topo@2.0.2: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > joi > topo@2.0.2: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > joi > topo > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > libp2p-webrtc-star > hapi > topo > hoek@4.2.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the lat [... cut ...] ompatibility check. Excluding it from installation.", "[3/4] Linking dependencies...", "[4/4] Building fresh packages..."]}
2019-08-01 14:37:38,179 p=9208 u=root |  PLAY RECAP *********************************************************************
2019-08-01 14:37:38,269 p=9208 u=root |  127.0.0.1                  : ok=45   changed=12   unreachable=0    failed=1    skipped=29   rescued=0    ignored=0   

I will disable the install, and try again. But I don't want to loose this, so I will submit it

holta, 2019-08-02:
@georgejhunt was this on a 1GB RAM Raspberry Pi 4 ?

@georgejhunt you mean the https://github.com/iiab/iiab/tree/master/roles playbook right?  e.g. no underscore as far as I know, as per subject line of this ticket.

Tagging @mitra42 in case he knows more.

holta, 2019-08-02:
@georgejhunt was this on Raspbian Lite 2019-07-10 ?

Or Raspbian With Desktop 2019-07-10 ?

mitra42, 2019-08-02:
Those warnings are super annoying but outside of our control (packages with warnings in their dependencies)  the fervent issue is not a problem as it’s an optional dependency. I’ve posted bug reports for several of these issues.   I’m on my phone st the moment so it’s hard to read but  I can’t actually see something in the logs causing a fail. I did a fresh install without IIAB yesterday on an RPI4 and can hopefully try  installing IIAB on it tomorrow

georgejhunt, 2019-08-02:
My exploration was with desktop, 1GB, since memory was the hypothesis.

holta, 2019-08-02:
> My exploration was with desktop, 1GB, since memory was the hypothesis.

Good idea.

@mitra42 please also consider testing on 'Raspbian Buster with desktop' from 2019-07-10 if possible?

jvonau, 2019-08-02:
@mitra42 can you post links to the upstream issues that you reported, so we can track the progress of the fixes please.

holta, 2019-08-02:
BIG-sized IIAB installs need to continue to install in a quite reliable/stable way for newbies as part of our final [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) release in coming days, making this as rock-solid as we can get anyway, per community expectations we've built up over the past 2 years.

So FYI we have a few days left but just advance warning that I may temporarily toggle `internetarchive_install` & `internetarchive_enabled` to false in https://github.com/iiab/iiab/blob/master/vars/local_vars_big.yml#L324-L327 e.g. as an interim measure, until IIAB 7.1 pre-releases become mainline in the weeks to follow.

Thankfully many people are attempting BIG-sized IIAB install over the coming weekend and will hopefully report back their latest results-

mitra42, 2019-08-02:
@jvonau - here are the links to the postings about warnings I've done so far, though looking at @georgejhunt 's post there may be new ones there.  I periodically do this, but its usually a) stuff out of our control and b) just warnings - usually repeated due to the web of dependencies in npm.
https://github.com/ipfs/js-ipfs/issues/2225
https://github.com/ipfs/js-ipfs-mfs/issues/54
https://github.com/ipfs/js-ipfs-multipart/issues/14

mitra42, 2019-08-02:
@holta - its fine if you want to toggle it off if quality isn't what you need by time you crank a release.

mitra42, 2019-08-03:
@georgejhunt @holta 
I just ran a complete build successfully from scratch. 

RPI4, 2GB, Raspbian Desktop (2019-07-10), BIG install

I followed the instructions on https://github.com/internetarchive/dweb-mirror/blob/master/INSTALLATION-iiab-raspberry.md exactly - i.e. downloading Raspbian, then running the pipe through bash.

So I'm not sure what I'm doing different from you ... 
I'm thinking that maybe there was a network fail at some point that caused something to be unavailable ?

holta, 2019-08-03:
@mitra42 excellent news.  1GB RAM developing world RPi's is our target but every data point helps!

I'll try to gather a few more data points Saturday from others...

holta, 2019-08-03:
@georgejhunt 3 other people cannot reproduce this, including on a 1GB RPi 4 &mdash; can you?

@jvonau suggests:
> see if a --reinstall or a pass with cmd-serve triggers the error. Might work the first time but fail on the second pass 

@tim-moody mentions:
> There's a boot setting that says how much memory for GPU vs CPU. For lite we don't need more GPU than the minimum. Don't know about desktop.

holta, 2019-08-04:
Observations installing BIG-sized onto 1GB RPi 4:
- `internetarchive` installed successfully
- load average was extremely high (often more than 20) for about 5min after boot, and `internetarchive` was partly responsible (often consuming about 50% of the CPU, according to `htop`)
- subsequent boots were not quite so terribly slow, but still slower than we'd really like during those 1st five minutes after boot, so better profiling could be extremely helpful, to shed light on:

> Investigate @mitra42's suggestions re: severe recent slowdowns (which are related to 'internetarchive' ?)
> https://github.com/iiab/iiab/issues/1812

mitra42, 2019-08-04:
If internetarchive is running, and *if* the user has setup some files to crawl then it could take a while to try and crawl them, but on a fresh user all it gets is the tiles for the top menu - and then subsequent crawls are very fast - certainly not full-power for 5 mins. 

I've got a clean install, but its at the office, so I'll have to check on Monday if I can repeat this problem.

holta, 2019-08-04:
> I've got a clean install, but its at the office, so I'll have to check on Monday if I can repeat this problem.

Is it a 1GB RAM RPi (3 or 4) hopefully?

> @holta Can you confirm if you are seeing this problem while the box has, or doesn't have, an internet connection ?

All recent tests have been done with a live Internet connection attached (Ethernet).

mitra42, 2019-08-04:
Its a 2GB RPI 4, so it wont be an exact match, but it should tell us if its hammering the CPU for 5 mins as was reported.

mitra42, 2019-08-04:
Also ... I'm not following all the issues, but I saw elsewhere that you are concerned about the first 5 mins, is that because other services are starting up ? 
Note that any changes I make should *not* require any change to the IIAB code itself, it will all be in the dweb-mirror npm packages that are pulled down during installation

holta, 2019-08-04:
> Also ... I'm not following all the issues, but I saw elsewhere that you are concerned about the first 5 mins, is that because other services are starting up ?

Whatever the reason(s), BIG-sized IIAB have been quite a lot slower during the first 5min after boot in recent weeks (especially on mainline 1GB RAM Raspberry Pi 3's and 4's??)

I don't quite understand the reason(s) why but this was not nearly as severe in the past, and needs digging into now (teachers/students need it responsive at the beginning of the lesson !)

(i.e. internetarchive & other intensive services' startup profiles need to be quantified during those initial minutes, however we can!)

mitra42, 2019-08-05:
@holta  I just ran on my clean install on a RPI 4 with 2GB. The server is running according to htop with 3.7% of memory and 0% of CPU. 

I ran the crawl - which runs at startup, the CPU stayed at 0% and it completed in under a minute.
I then did `sudo rm -r /library/archiveorg/*` which empties the cache, and reran the crawl, again it never left 0% and took about a minute to complete. 

It would be good to see if the slowdown is after a reinstall of IIAB on a machine which already has a substantial crawl defined from previous usage, could anyone seeing this send me their `/root/dweb-mirror.config.yaml` a default one on a new install looks like 
```
apps:
  crawl:
    tasks:
      - identifier:
          - home
        level: details
        search:
          - sort: '-downloads'
            rows: 30
            level: tile
```

holta, 2019-08-08:
@georgejhunt suggests we closed this as others cannot reproduce this, and was likely caused by a networking glitch.

(But please shout if this should be re-opened!)

closed 2019-08-08 by holta

holta, 2019-08-09:
> @jvonau - here are the links to the postings about warnings I've done so far, though looking at @georgejhunt 's post there may be new ones there. I periodically do this, but its usually a) stuff out of our control and b) just warnings - usually repeated due to the web of dependencies in npm.
> [ipfs/js-ipfs#2225](https://github.com/ipfs/js-ipfs/issues/2225)
> [ipfs/js-ipfs-mfs#54](https://github.com/ipfs/js-ipfs-mfs/issues/54)
> [ipfs/js-ipfs-multipart#14](https://github.com/ipfs/js-ipfs-multipart/issues/14)

@jvonau see also this list that @mitra42 is compiling:

Package dependency warnings {META}
https://github.com/internetarchive/dweb-mirror/issues/234

mitra42, 2019-08-11:
@jvonau if you saw this as well - and can repeat, it would be great to get a dump WITHOUT the warnings, I can't actually find what failed among all the warnings

jvonau, 2019-08-11:
I'm thinking this is more of an OOM issue and related to #1911, as I triggered the kind of warnings/failure above while running yarn update under ansible while testing #1921 on 181-rpi4-1gb-desk-0803-BIG. The ansible failure went away when some system resources were freed up before the second try with no other changes.

Those warnings should subside as you update your sources to the version where the upstream fix is applied.

mitra42, 2019-08-11:
Sorry - I'm not understanding this (and guessing that OOM means Out Of Memory).

I'm not sure how updating sources to the changes you made in https://github.com/iiab/iiab/pull/1921 would fix this for a fresh install (when neither nodejs, yarn nor internetarchive will be already installed), so do you mean something else for "upstream" fix?

jvonau, 2019-08-12:
> Sorry - I'm not understanding this (and guessing that OOM means Out Of Memory).
>
Yes, Out Of Memory as this was on similar hardware/OS as George and Adam's VM in issue #1812 has 1GB. Just trying to triage the trigger of the ansible failure and it's looking like a memory resource issue to me. 
 
> I'm not sure how updating sources to the changes you made in #1921 would fix this for a fresh install (when neither nodejs, yarn nor internetarchive will be already installed), so do you mean something else for "upstream" fix?

Yes something else, upstream from us would your repo and it's dependencies, nice to be be fixed as yarn upgrade on the command line looks scary but I feel that it is not the trigger for an ansible failure as I noted above.

mitra42, 2019-08-13:
Thanks - that is clear, we've posted the warnings as per internetarchive/dweb-mirror#234 but unfortunately as these get fixed we can expect more to occur.