iiab/iiab #1916: wikihow_en_2016-11.zim prevents Admin Console from reinstalling ?

holta, 2019-08-02:
@tim-moody this is one of the most popular non-standard ZIM files that many people want to install.  How should we standardize this?

Currently it's manually installable (instructions below) but then that would appear to break your next reinstall of Admin Console, per the error below (can you confirm that's what's happening?)

Specifically: when using the ZIM file contained within http://oer2go.org/viewmod/en-wikihow = http://oer2go.org/zipped-modules/en-wikihow.zip I did as was done at http://iiab.me here:

- I downloaded the above RACHEL/OER2Go module then renamed/moved its ZIM file from `/library/www/html/modules/en-wikihow/data/content/en-wikihow.zim` to `/library/zims/content/wikihow_en_2016-11.zim`
- I place `en-wikihow_en` into the "menu_items_1" list at `/library/www/html/home/menu.json`
- I verified that https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-wikihow_en.json is installed at `/library/www/html/js-menu/menu-files/menu-defs/en-wikihow_en.json`
- It now works!  Showing up on IIAB home page http://box leading to the ZIM content just like http://iiab.me/kiwix/wikihow_en_2016-11/
- FYI renaming `wikihow_en_2016-11.zim` to `en-wikihow_en_2016-11.zim` did not work (http://box fails to display it if you do that).
- And The Kicker... `cd /opt/iiab/iiab-admin-console; ./install` now fails as follows:

> TASK [js-menu : Add any IIAB roles that are installed to home/menu.js] *********
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/opt/admin/cmdsrv/scripts/iiab_update_menus.py", "delta": "0:00:04.195277", "end": "2019-08-02 22:51:27.039014", "msg": "non-zero return code", "rc": 1, "start": "2019-08-02 22:51:22.843737", "stderr": "Traceback (most recent call last):\n  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 285, in <module>\n    main()\n  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 73, in main\n    put_kiwix_enabled_into_menu_json()\n  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 142, in put_kiwix_enabled_into_menu_json\n    menu_item = create_menu_def(perma_ref, default_name)\n  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 172, in create_menu_def\n    zim_lang = item['language']\nKeyError: 'language'", "stderr_lines": ["Traceback (most recent call last):", "  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 285, in <module>", "    main()", "  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 73, in main", "    put_kiwix_enabled_into_menu_json()", "  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 142, in put_kiwix_enabled_into_menu_json", "    menu_item = create_menu_def(perma_ref, default_name)", "  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 172, in create_menu_def", "    zim_lang = item['language']", "KeyError: 'language'"], "stdout": "Updating kiwix menus\nen-matheducators.stackexchange.com_en_all.json {u'media_count': u'8839', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'matheducators.stackexchange.com_en_all_2019-02', u'article_count': u'2504', u'zim_date': u'2019-02-01', u'size': u'81.0M'}\nen-gis.stackexchange.com_en_all.json {u'media_count': u'160233', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'gis.stackexchange.com_en_all_2019-01', u'article_count': u'111233', u'zim_date': u'2019-01-31', u'size': u'2.26G'}\nde-wikibooks_de_all_novid.json {u'media_count': u'36428', u'menu_item': u'', u'language': u'de', u'tags': u'wikibooks;novid', u'file_name': u'wikibooks_de_all_novid_2019-07', u'article_count': u'31240', u'zim_date': u'2019-07-24', u'size': u'1.63G'}\nen-outdoors.stackexchange.com_en_all.json {u'media_count': u'15201', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'outdoors.stackexchange.com_en_all_2019-02', u'article_count': u'5211', u'zim_date': u'2019-02-03', u'size': u'238M'}\nen-ted_en_global_issues.json {u'media_count': u'999', u'menu_item': u'', u'language': u'en', u'tags': u'', u'file_name': u'ted_en_global_issues_2018-07', u'article_count': u'498', u'zim_date': u'2018-07-23', u'size': u'12.2G'}\nen-wikivoyage_en_all_novid.json {u'media_count': u'57482', u'menu_item': u'en-wikivoyage_en_all_novid', u'language': u'en', u'tags': u'wikivoyage;novid', u'file_name': u'wikivoyage_en_all_novid_2019-06', u'article_count': u'29074', u'zim_date': u'2019-06-03', u'size': u'776M'}\nde-wikipedia_de_all_novid.json {u'media_count': u'2743417', u'menu_item': u'', u'language': u'de', u'tags': u'wikipedia;novid;_ftindex', u'file_name': u'wikipedia_de_all_novid_2018-11', u'article_count': u'2235081', u'zim_date': u'2018-11-06', u'size': u'33.0G'}\nen-physics.stackexchange.com_en_all.json {u'media_count': u'192686', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'physics.stackexchange.com_en_all_2019-02', u'article_count': u'134830', u'zim_date': u'2019-02-03', u'size': u'2.31G'}\nen-reverseengineering.stackexchange.com_en_all.json {u'media_count': u'21652', u'menu_item': u'en-reverseengineering_stackexchange_com_en_all', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'reverseengineering.stackexchange.com_en_all_2019-02', u'article_count': u'6010', u'zim_date': u'2019-02-04', u'size': u'181M'}\nde-wikivoyage_de_all_novid.json {u'media_count': u'49084', u'menu_item': u'', u'language': u'de', u'tags': u'wikivoyage;novid', u'file_name': u'wikivoyage_de_all_novid_2019-06', u'article_count': u'16748', u'zim_date': u'2019-06-03', u'size': u'692M'}\nen-ham.stackexchange.com_en_all.json {u'media_count': u'10161', u'menu_item': u'en-ham_stackexchange_com_en_all', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'ham.stackexchange.com_en_all_2019-01', u'article_count': u'2656', u'zim_date': u'2019-01-31', u'size': u'125M'}\nen-security.stackexchange.com_en_all.json {u'media_count': u'157899', u'menu_item': u'en-security_stackexchange_com_en_all', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'security.stackexchange.com_en_all_2019-02', u'article_count': u'52420', u'zim_date': u'2019-02-07', u'size': u'0.99G'}\nen-chemistry.stackexchange.com_en_all.json {u'media_count': u'63278', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'chemistry.stackexchange.com_en_all_2019-01', u'article_count': u'29111', u'zim_date': u'2019-01-29', u'size': u'543M'}\nen-phet_en.json {u'media_count': u'91', u'menu_item': u'', u'language': u'en', u'tags': u'', u'file_name': u'phet_en_2019-06', u'article_count': u'73', u'zim_date': u'2019-06-05', u'size': u'41.4M'}\nsw-wikipedia_sw_all_novid.json {u'media_count': u'33954', u'menu_item': u'', u'language': u'sw', u'tags': u'', u'file_name': u'wikipedia_sw_all_novid_2019-03', u'article_count': u'48895', u'zim_date': u'2019-03-18', u'size': u'406M'}\nen-sustainability.stackexchange.com_en_all.json {u'media_count': u'5547', u'menu_item': u'', u'language': u'en', u'tags': u'_ftindex', u'file_name': u'sustainability.stackexchange.com_en_all_2018-04', u'article_count': u'6419', u'zim_date': u'2018-04-21', u'size': u'64.8M'}\nen-interpersonal.stackexchange.com_en_all.json {u'media_count': u'20421', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'interpersonal.stackexchange.com_en_all_2019-01', u'article_count': u'3088', u'zim_date': u'2019-01-31', u'size': u'139M'}\nen-cogsci.stackexchange.com_en_all.json {u'media_count': u'17299', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'cogsci.stackexchange.com_en_all_2019-01', u'article_count': u'6130', u'zim_date': u'2019-01-29', u'size': u'140M'}\nen-wikihow_en.json {u'media_count': u'723923', u'menu_item': u'', u'language': u'en', u'tags': u'', u'file_name': u'wikihow_en_2016-11', u'article_count': u'69943', u'zim_date': u'2016-11-29', u'size': u'12.3G'}", "stdout_lines": ["Updating kiwix menus", "en-matheducators.stackexchange.com_en_all.json {u'media_count': u'8839', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'matheducators.stackexchange.com_en_all_2019-02', u'article_count': u'2504', u'zim_date': u'2019-02-01', u'size': u'81.0M'}", "en-gis.stackexchange.com_en_all.json {u'media_count': u'160233', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u' [... cut ...] 139M'}", "en-cogsci.stackexchange.com_en_all.json {u'media_count': u'17299', u'menu_item': u'', u'language': u'en', u'tags': u'stackexchange;_ftindex', u'file_name': u'cogsci.stackexchange.com_en_all_2019-01', u'article_count': u'6130', u'zim_date': u'2019-01-29', u'size': u'140M'}", "en-wikihow_en.json {u'media_count': u'723923', u'menu_item': u'', u'language': u'en', u'tags': u'', u'file_name': u'wikihow_en_2016-11', u'article_count': u'69943', u'zim_date': u'2016-11-29', u'size': u'12.3G'}"]}

tim-moody, 2019-08-04:
> I downloaded the above RACHEL/OER2Go module then renamed/moved its ZIM file from /library/www/html/modules/en-wikihow/data/content/en-wikihow.zim to /library/zims/content/wikihow_en_2016-11.zim

good

I copy en-wikihow.json to en-wikihow_zim.json
add/change the lines 

  "intended_use": "zim",
  "zim_name": "wikihow_en",

I place en-wikihow_zim into the "menu_items_1" list at /library/www/html/home/menu.json
I verified that https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-wikihow_en.json is installed at /library/www/html/js-menu/menu-files/menu-defs/en-wikihow_en.json

I ran iiab-make-kiwix-lib

> 
It now works! Showing up on IIAB home page http://box leading to the ZIM content just like http://iiab.me/kiwix/wikihow_en_2016-11/

cd /opt/iiab/iiab-admin-console; ./install works fine

> 
FYI renaming wikihow_en_2016-11.zim to en-wikihow_en_2016-11.zim did not work (http://box fails to display it if you do that).

so why do it

tim-moody, 2019-08-04:
N.B. /library/www/html/common/assets/zim_version_idx.json  has
  "wikihow_en": {
    "media_count": "723923",
    "menu_item": "en-wikihow",
    "language": "en",
    "tags": "",
    "file_name": "wikihow_en_2016-11",
    "article_count": "69943",
    "zim_date": "2016-11-29",
    "size": "12.3G"
  }
so even size and article count, etc. are available for footnotes

holta, 2019-08-08:
> - FYI renaming `wikihow_en_2016-11.zim` to `en-wikihow_en_2016-11.zim` did not work (http://box fails to display it if you do that).

This seems to have been a big mistake on my part.  The correct ZIM file is in fact `wikihow_en_2016-11.zim` -- there should be no `en-` on the left.

holta, 2019-12-21:
Ref: #1902 "en-wikihow missing from http://box.lan/admin > Install Content > Get OER2Go(RACHEL) Modules [publish metadata standards for ZIM files published to/indexed at Archive.org]"