iiab/iiab #1925: 'internetarchive' fails to install on 1st pass -- forcing a 2nd pass -- why?

holta, 2019-08-13:
@jvonau @mitra42 this failure has been reproduced on both OS's Ubuntu 18.04 and Raspbian Lite on RPi 4 -- any idea why this is happening?

> ASK [internetarchive : Run yarn install to get needed modules (CAN TAKE ~15 MINUTES)] ***
00h00m00s 0/0: : fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["yarn", "add", "@internetarchive/dweb-archive", "@internetarchive/dweb-mirror"], "delta": "0:00:01.003249", "end": "2019-08-13 16:13:57.339924", "msg": "non-zero return code", "rc": 1, "start": "2019-08-13 16:13:56.336675", "stderr": "ERROR: [Errno 2] No such file or directory: 'add'", "stderr_lines": ["ERROR: [Errno 2] No such file or directory: 'add'"], "stdout": "", "stdout_lines": []}

It's possible a regression was introduced by @jvonau's PR #1921 and my PR #1924 was not enough to fix it in the end?

holta, 2019-08-13:
@jvonau smoke-tested PR #1921 as root using `./runrole internetarchive` and yet it does not work to fresh-install the 'internetarchive' role.  He asks if this is possibly a `sudo` issue similar to...?

#1714 "MySQL database password-changing failure during fresh install of IIAB on Ubuntu 19.04 [and 19.10 pre-releases?]"

mitra42, 2019-08-13:
I'm not sure , but there here is my hunch ... 
There is an ancient package called 'yarn' which apt knows about, in @jvonau's PR I see the `apt-get update` is removed and I suspect that caused it to install the wrong yarn, which is something completely different and doesn't understand 'add'. 

I encountered this before - and warn about it in https://github.com/internetarchive/dweb-mirror/blob/master/INSTALLATION-raspberrypi.md which also says how to get rid of the bad package if its inadvertently installed.

holta, 2019-08-13:
Thanks @mitra42 for the possible explanation.

Am re-opening this until we've confirmed this is under control...

holta, 2019-08-13:
PR #1926 did not help the situation, just as PR #1924 always failed to improve the situation.

@jvonau: there is a deeper flaw in PR #1921 that I hope you can help us recover from overnight!

holta, 2019-08-14:
PR #1927 finally seems to have solved it, yay!

As tested on an Ubuntu 18.04 VM anyway &mdash; going forward let's keep an eye on RPi to be sure!

closed 2019-08-14 by holta

mitra42, 2019-08-14:
Ansible certainly seems to be one of those black-arts !

holta, 2019-08-14:
Is Ansible to blame?

@jvonau can explain later, but this section (that he'd thought was optional) turned out to be critical:

- name: "Yarn | Update APT cache"	
  apt:	
    update_cache: yes

mitra42, 2019-08-14:
Oh - I'm not saying ansible is to blame, just that until you understand its obviously seriously hard to translate requirements into ansible roles.

holta, 2019-08-14:
> PR #1927 finally seems to have solved it, yay!
> 
> As tested on an Ubuntu 18.04 VM anyway — going forward let's keep an eye on RPi to be sure!

BIG-sized test (fresh install) successful on Raspbian Lite on 1GB RPi 4...keeping us on track to release IIAB 7.0

jvonau, 2019-08-14:
Just one of those [commits](https://github.com/iiab/iiab/pull/1921/commits/001e00eb26246eba8eb8669bae101f9414301738) that I should not of pushed when tired....

PR #1926 (2019-08-13): Partial revert of PR #1921, attempting to get 'internetarchive' to install on 1st pass
Does this fix #1925, even if we don't understand why?