iiab/iiab #1931: yarn upgrade fails on RPi 4

holta, 2019-08-16:
@jvonau @mitra42 this is sounding serious...but then running it twice somehow bypassed the issue...do you understand this?

> TASK [internetarchive : Update pre-existing install: yarn upgrade] *************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["yarn", "upgrade"], "delta": "0:27:08.061869", "end": "2019-08-16 11:25:00.213648", "msg": "non-zero return code", "rc": -9, "start": "2019-08-16 10:57:52.151779", "stderr": "warning package.json: No license field\nwarning No license field\nwarning No license field\nwarning No license field\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > gun > text-encoding@0.7.0: no longer maintained\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > gun > emailjs > emailjs-mime-codec > text-encoding@0.7.0: no longer maintained\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi-pino > @hapi/hoek@6.2.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).\nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > ipld-ethereum > ethereumjs-block > merkle-patricia-tree > level-ws > xtend > object-keys@0.4.0: \nwarning @internetarchive/dweb-archive > @internetarchive/dweb-transports > wolkjs > @trust/webcrypto > text-encoding@0.6.4: no longer maintained\nwarning \"@internetarchive/dweb-archive > @internetarchive/ia-components@0.0.0-alpha.327\" has unmet peer dependency \"bookreader@git://github.com/internetarchive/bookreader.git#^4.2.2\".\nwarning \"wrtc@0.2.1\" is missing a bundled dependency \"node-pre-gyp\". This should be reported to the package maintainer.", "stderr_lines": ["warning package.json: No license field", "warning No license field", "warning No license field", "warning No license field", "warning @internetarchive/dweb-archive > @internetarchive/dweb-transports > gun > text-encoding@0.7.0: no longer maintained", "warning @internetarchive/dweb-archive > @internetarchive/dweb-transports > gun > emailjs > emailjs-mime-codec > text-encoding@0.7.0: no longer maintained", "warning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > hapi-pino > @hapi/hoek@6.2.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).", "warning @internetarchive/dweb-archive > @internetarchive/dweb-transports > ipfs > ipld-ethereum > ethereumjs-block > merkle-patricia-tree > level-ws > xtend > object-keys@0.4.0: ", "warning @internetarchive/dweb-archive > @internetarchive/dweb-transports > wolkjs > @trust/webcrypto > text-encoding@0.6.4: no longer maintained", "warning \"@internetarchive/dweb-archive > @internetarchive/ia-components@0.0.0-alpha.327\" has unmet peer dependency \"bookreader@git://github.com/internetarchive/bookreader.git#^4.2.2\".", "warning \"wrtc@0.2.1\" is missing a bundled dependency \"node-pre-gyp\". This should be reported to the package maintainer."], "stdout": "yarn upgrade v1.17.3\n[1/4] Resolving packages...\n[2/4] Fetching packages...\ninfo There appears to be trouble with your network connection. Retrying...\ninfo fsevents@1.2.9: The platform \"linux\" is incompatible with this module.\ninfo \"fsevents@1.2.9\" is an optional dependency and failed compatibility check. Excluding it from installation.\n[3/4] Linking dependencies...\n[4/4] Rebuilding all packages...", "stdout_lines": ["yarn upgrade v1.17.3", "[1/4] Resolving packages...", "[2/4] Fetching packages...", "info There appears to be trouble with your network connection. Retrying...", "info fsevents@1.2.9: The platform \"linux\" is incompatible with this module.", "info \"fsevents@1.2.9\" is an optional dependency and failed compatibility check. Excluding it from installation.", "[3/4] Linking dependencies...", "[4/4] Rebuilding all packages..."]}
>
> PLAY RECAP *********************************************************************
127.0.0.1                  : ok=517  changed=143  unreachable=0    failed=1    skipped=185  rescued=0    ignored=0

jvonau, 2019-08-16:
Yea there is too much running for 1GB to support.

mitra42, 2019-08-16:
Most of these error messages are warnings from dependencies, I've reported them to the individual package maintainers, and tracking the process in https://github.com/internetarchive/dweb-mirror/issues/234 but nothing really I can do about them, and yarn doesn't give a way to turn them off so that you can see real problems. 

Reading through the full list, I'm aware of all of them until the one that caused the problem `There appears to be trouble with your network connection.` which suggests that either your network really had problems or I think @jvonau is suggesting that the network issue is really a memory issue ?

mitra42, 2019-08-27:
Could someone who is having this problem try something which helped with memory issues installing our server on an OrangePi running Armbian. ... 

In internetarchive/tasks/main.yml I want the command to be: 
```
yarn config set child-concurrency 1 && yarn upgrade
```
(That's valid shell, I'm not sure if its valid in ansible)
This restricts it to rebuilding one dependency at a time instead of the default which is 4. I'm guessing that it won't slow the overall time significantly on 4GB machines, but might be the difference between success and failure on 1GB ones.

mitra42, 2019-08-28:
Ok - I was trying to address whoever posted the 1GB RPI report above but I see now the post was you, but that might have been somethng you received from someone else ...

mitra42, 2019-08-28:
It also needs some help from @jvonau I think as to how to rework that one-line of bash into whatever ansible needs (I don't think the "command" field in ansible can have multiple commands changed together like this).

holta, 2019-09-03:
@mitra42: use the Ansible's [shell](https://docs.ansible.com/ansible/latest/modules/shell_module.html) routine command instead of its [command](https://docs.ansible.com/ansible/latest/modules/command_module.html) routine, so that a full line of bash should execute.  Thanks to @jvonau for the reminder!

holta, 2019-09-30:
Not sure if this is still an issue (?) but it's important, so I'm moving it to the IIAB 7.1 milestone.

mitra42, 2019-09-30:
Understood - I did this but am doing a significant refactor of the installer to make it install quicker, by removing the dependencies.   I got the raw version to run on my Mac last night, next step to try on RPI then do the edits necessary on IIAB - I would guess another few days only - what timeline is 7.1 on ?

holta, 2019-09-30:
> Understood - I did this but am doing a significant refactor of the installer to make it install quicker, by removing the dependencies. I got the raw version to run on my Mac last night, next step to try on RPI then do the edits necessary on IIAB - I would guess another few days only -

Good news, thanks for the update.

> what timeline is 7.1 on ?

I would like IIAB 7.1 to be a more rapid release, that's actually distributed before end of year -- i.e. Q4 2019, by reducing the scope of changes to core needs like 1-click installs including content (that many implementers in the developing world can now use, with rapid Internet in their capital cities...)

Unlike IIAB 7.0 which grew to contain many different things over 7 months!

mitra42, 2019-10-04:
PR is pushed for this #1982

closed 2019-10-04 by holta