iiab/iiab #1934: Document Nextcloud

aidan-fitz, 2019-08-23:
TIL (in https://github.com/iiab/iiab/issues/1759#issuecomment-524187956) that Nextcloud is a lot like an open source substitute for G Suite. We should add a README documenting what Nextcloud does and why customers might want it.

Related: #1548

holta, 2019-08-23:
Can we start by expanding this 2-sentence description of Nextcloud?

> Nextcloud puts your data at your fingertips, under your control. Store your documents, calendar, contacts and photos on your local server.

As is currently offered to Internet-in-a-Box implementers here:

> "What services (IIAB apps) are suggested during installation?"
in http://FAQ.IIAB.IO Item 9

holta, 2019-09-06:
Completed, please review PR #1942 !

Specifically: https://github.com/iiab/iiab/blob/master/roles/nextcloud/README.md

And related PRs: #1941 #1943

closed 2019-09-06 by holta