iiab/iiab #1958: 3 new map packs covering {Spanish, French, Arabic}-speaking regions of the world

holta, 2019-09-24:
3 new OSM Vector Map Packs are needed, that include "full zoom" detail across…

- the [~30 countries](https://en.wikipedia.org/wiki/File:Hispanophone_global_world_map_language_2.svg) where Spanish is spoken:
  - This one might be as simple as 2 bounding boxes merged together, hopefully.
- the [~30 countries](https://en.wikipedia.org/wiki/Geographical_distribution_of_French_speakers) where French is spoken:
  - NOTE: this includes countries like Algeria and Morocco that do not *officially* speak French.
  - _One option may be to add map zoom detail for {Quebec, Madagascar, Belgium, Switzerland and France} to the large French-speaking region of West and Central Africa._
- the [~25 countries](https://en.wikipedia.org/wiki/List_of_countries_where_Arabic_is_an_official_language#Countries_with_Arabic_as_a_national_language_or_a_recognized_minority_language) where Arabic is spoken:
  - This one might be as simple as 1 single bounding box hopefully.

These are not singular bounding box rectangles unfortunately, so we'll discuss how best to make this properly happen during upcoming IIAB Thursday calls (http://minutes.iiab.io).

ASIDE: the above 3 are much more urgent than https://github.com/iiab/iiab/issues/1170 ("Haiti Diaspora Map Pack: HT + DR + Cuba + S. Florida + Greater NYC (+ NJ?) + Boston + Montreal + Sao Paulo/Brazil + Santiago/Chile + Bahamas (+ Paris + Atlanta + Philly?)") for reasons that I'll explain during upcoming IIAB calls (http://minutes.iiab.io).

holta, 2019-09-24:
TARGET: these 3 new Map Packs should be (around!) 10-to-12GB (or possibly less) so that they fit on 64GB microSD cards especially, alongside Wikipedia and KA Lite in that same regional language.

REASON: viral mass-distribution of IIAB (via @jvonau's auto-created ISO's using pi-gen & similar) suddenly becomes possible &mdash; as soon as high-quality content suites in multinational languages like {Spanish, French, Arabic} are baked in &mdash; as newbies perpetually (and very reasonably) demand.

KEY INSIGHT: IIAB is understood _(and adored!)_ when it comes with fully working content, that can then be further customized by implementers with high-bandwidth connections (and an Ethernet cord ideally ;)

holta, 2019-10-01:
@georgejhunt: Nick Doiron (@mapmeld) asks if he might help us move the map of all Arabic-speaking countries forward:

- A simple bounding-box could well be sufficient for the ~25 countries where Arabic is spoken, as seen by [Wikipedia's green](https://en.wikipedia.org/wiki/List_of_countries_where_Arabic_is_an_official_language#Countries_with_Arabic_as_a_national_language_or_a_recognized_minority_language) ?
- More than just one rectangle (bounding box) is required for the ~30 countries that speak [Spanish](https://en.wikipedia.org/wiki/File:Hispanophone_global_world_map_language_2.svg) and [French](https://en.wikipedia.org/wiki/Geographical_distribution_of_French_speakers), but let's start with the easy one!

PS Nick will be away from Boston for a bit, but returning well in time for our Nov 7-11 IIAB Summit with http://wikiconference.org and @avni etc!

holta, 2019-10-17:
1) @georgejhunt can you help me generate the "Hispanic Americas" region (@avni agrees to this exact title) with the following parameters?

   ```
   "title": "South Asia",
   "perma_ref": "en-osm-omt_south_asia", 
   "seq": 25,    # betw Central Am (20) and South Am (30)
   
   "east": "-18.0",
   "north": "33.0",    # was 25.0
   "south": "-84.0",
   "west": "-180.00",    # was -179.0
   ```

   In coming days if you have time?

   Above syntax per http://download.iiab.io/content/OSM/vector-tiles/maplist/assets/regions.json

   I'd assume the final size for the "Hispanic Americas" Map Pack will be about 9-to-10GB ?  (Given the current "Central America" Map Pack occupies 3.77 GB on installation, and the current / greatly overlapping "South America" occupies 6.60 GB.)

   Later we can try merging in Spain and Equatorial Guinea (in Africa) from openstreetmaptiles.com if that turns out to be easy.  If which case the Map Pack could be renamed to "Spanish-speaking Regions".  Western Sahara ([global Spanish map](https://en.wikipedia.org/wiki/File:Hispanophone_global_world_map_language_2.svg)) has some Spanish speakers but it does not seem that populated.

2) [Arabic-Speaking Countries/Regions](https://en.wikipedia.org/wiki/List_of_countries_where_Arabic_is_an_official_language#Countries_with_Arabic_as_a_national_language_or_a_recognized_minority_language) can be sufficiently represented by the [existing 10.56 GB "Africa" Map Pack](http://iiab.me/maps/maplist/) for now &mdash; however I'm still going to talk to Nick Doiron who wants to improve on this with me when he returns from Mongolia etc next week.

3) Finally the [French-speaking regions](https://en.wikipedia.org/wiki/Geographical_distribution_of_French_speakers) are painfully scattered around the world (see map) so let's do a dress rehearsal in coming weeks (e.g. when many of us are in Boston!) merging stuff like Quebec+Haiti+West Africa+Madagascar+France+Belgium+Switzerland from https://openmaptiles.com/downloads/planet/ so that I learn how to do this!  Trying to get that bundled up into ~10 GB just like the above two.

_PS @georgejhunt would you know why https://openmaptiles.com/downloads/dataset/osm/ shows all of 67.81 GB for osm-2019-10-14-v3.10-planet.mbtiles (https://openmaptiles.com/downloads/tileset/osm/) ?  I'm asking if this appears abnormally large...and what the reasons might be (?) given that your "World" Map Pack currently ends up as only 53.17 GB &mdash; including satellite photo across 11 levels of zoom!  (932M /library/www/osm-vector-maps/satellite_z0-z9_v3.mbtiles)_

georgejhunt, 2019-10-18:
>    Pack <http://iiab.me/maps/maplist/> for now — however I'm still going
> your "World" Map Pack currently ends up as only 53.17 GB — including

*osm-2019-10-14-v3.10-planet.mbtiles costs $1000. What we are using is free
-- but it's 2017 vintage*

holta, 2019-10-18:
> *osm-2019-10-14-v3.10-planet.mbtiles costs $1000. What we are using is free -- but it's 2017 vintage*

Ok, understood.  As https://openmaptiles.com/downloads/tileset/osm/ says, the free version is:

> From: 2017-07-03 / Version: v3.6.1
Size: 51.01 GB / Format: MBTiles (PBF)

holta, 2019-10-21:
"Spanish-speaking Regions" Map Pack is now in testing, thanks to @georgejhunt, who writes:

> Tested on a rpi4, min local_vars.
> 1. Change "hidden" to "test" in vars/default_vars.yml line 413
[ see "spanish_speaking_regions" stanza near bottom of http://download.iiab.io/content/OSM/vector-tiles/maplist/test/assets/regions.json ]
> 2. ./runrole osm-vector-maps
> 3. reboot
> 4. check box in admin/install-content
> 5. click "install selected map regions"
> 6. To create menu item for new map, run "iiab-update-map"

avni, 2019-10-21:
> "Spanish-speaking Regions" Map Pack is now in testing, thanks to @georgejhunt, who writes:

Amazing!

holta, 2019-10-22:
@georgejhunt you have the following line twice: (in 2 different stanzas, near the end of http://download.iiab.io/content/OSM/vector-tiles/maplist/test/assets/regions.json)

```
      "perma_ref": "en-osm-omt_south_america", 
```

This looks wrong...can you fix this?

holta, 2019-11-16:
> "Spanish-speaking Regions" Map Pack is now in testing, thanks to @georgejhunt, who writes:

@georgejhunt can you make this new region available now, to everyone installing IIAB?  i.e. using Admin Console -> Install Content -> Get Map Region ...that appears similar to http://iiab.me/maps/maplist/

PS Avni prefers the name "Hispanic Americas" but if you prefer "Spanish-speaking Regions" that's fine too.

holta, 2019-11-16:
> PS Avni prefers the name "Hispanic Americas" but if you prefer "Spanish-speaking Regions" that's fine too.

FYI Wikipedia prefers to call it "Latin America"

holta, 2019-11-18:
I didn't realize Spain is a part of George's new Map Pack...i.e. this is the 1st Map Pack that is more than just a bounding box!

So we decided on today's call (http://minutes.iiab.io) to keep this Map Pack's name as "Spanish-speaking Regions" ...and George will publish it in coming days so that every IIAB implementer will be able to easily install it going forward.

@georgejhunt: will both of its included rectangles/regions (Latin America & Spain) appear as green rectangles in Admin Console's visual map picker?