iiab/iiab #1960: IIAB's 1-line installer should *require* (a recent version of?) Raspbian Buster

holta, 2019-09-25:
Consider a school that recently installed IIAB onto a 2017 version of Raspbian from almost 2 years ago, and then asked for support again and again and again... :/

holta, 2019-10-18:
During yesterday's IIAB weekly call (http://minutes.iiab.io) we discussed several ways to implement this:
- Check the timestamp on a file like /etc/issue and refuse to install if it's too old?
- Contents of /etc/os-release could also be checked but unclear exactly how to parse its contents.
- Easiest is probably to act on /etc/debian_version &mdash; and then explaining that IIAB no longer supports OS's based on version of Debian < 10.x.  FYI here's what /etc/debian_version shows on various OS's...
  - Raspbian Buster shows "10.1"
  - Debian 10.1 shows "10.1"
  - Debian 9.11 shows "9.11" **(likely we should disallow these going forward / soon!)**
  - Ubuntu 18.04 and 19.10 shows "buster/sid"
  - Ubuntu 16.04 shows "stretch/sid" **(likely we should disallow these going forward / soon!)**

jvonau, 2020-04-29:
first touch point is 'iiab' so this really should addressed in iiab-factory

jvonau, 2020-09-07:
#2501

holta, 2020-09-12:
Related:
- #2497 "Concerning 1 line install on Raspberry pi 3b+ [kolibri.service failed to start due to Ansible 2.9.12; 'ansible_userspace_architecture' is undefined; WiFi flaky due to RaspiOS kernel 5.4.51-v7+ #1333 ?]"
- iiab/iiab-factory#133 "add kernel warning for RaspOS"

jvonau, 2021-07-21:
implemented

holta, 2021-07-21:
Good enough, yes!

closed 2021-07-21 by holta