iiab/iiab #1964: Admin Console removes comments from local_vars.yml

holta, 2019-09-26:
This bug is not a blocker preventing the release of [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes), but as mentioned in PR #1962 it definitely should be fixed in coming weeks if at all possible.  Specifically: Admin Console has accidentally introduced a bug whereby things in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) like...

```
# Unmaintained
# osm_install: False
# osm_enabled: False

# Unmaintained
# docker_install: False
# docker_enabled: False
```

Become...

```
# Unmaintained

# Unmaintained
```

(This happens after use of Admin Console, e.g. to enable/disable services.)

My personal opinion is that implementers' comments should be guaranteed to remain untouched in all circumstances.

jvonau, 2019-10-05:
after running ICO to see the full extent of the difference try 
diff -u /opt/iiab/iiab/vars/local_vars_medium.yml /etc/iiab/local_vars.yml

jvonau, 2019-10-08:
http://paste.debian.net/1105380/ of above diff

holta, 2019-10-08:
> http://paste.debian.net/1105380/ of above diff

Above page disappears in 7 days.  I've reposted it here (disappears in 1 year) so we don't lose it:

https://paste.ubuntu.com/p/5pSMPGmBCf/

holta, 2019-10-09:
My belief is that Admin Console should not modify any comments of any kind, ever, in /etc/iiab/local_vars.yml

For those who don't read diffs, here are just a few of the many comments (beginning with the '#' character) that Admin Console is currently deleting from local_vars.yml:

- all explanatory URL's appear to be deleted, presumably because they contain the character ':'
- all English sentences are deleted, on lines where they happen to contain the character ':'
- explanatory documentation is deleted, when it (very intentionally) refers to actual example variable usage, e.g.

```
# kiwix_install: True is REQUIRED, if you install IIAB's Admin Console
```

**Where English sentences remain, they become incoherent when comment-out variables on nearby lines suddenly disappear.**

Separately: stale comments are accidentally being added back into local_vars.yml, in the place of the latest explanations that are silently disappearing.

Let's avoid all the above and similar problems.  For all these reasons, I believe we need to continue allowing implementers full control over their own comments, and not try to programmatically change they human-generated explanations, mangling the intent and comments that are very intentionally placed there as contextual and explanatory reminders.

tim-moody, 2019-10-09:
None of the items you mention was put there by an implementer; all were put there by you. As far as I am aware, no implementer has ever added a comment. local_vars.yml is a configuration file. Put documentation in documentation. 

If you comment out kalite_enable: True, there is no way for Admin Console to set it to False without uncommmenting, or adding a line at the bottom, which was not desired I thought.

There is no reasonable way to distinguish between 

# xxx_install: True is a really good thing to do
and
# xxx_install: True

And the bottom line is that no variables in this configuration file have the wrong values as a result of this process, only comments that should not be there in the first place.

Comments in the configuration file create a huge maintenance problem in that they must be maintained in every variant of local vars, min, med, and big, not to mention medical, which doesn't even make the list. Further, most variables get copied from defaults to variants of local vars, so the user has trouble telling how each configuration differs and this is made even worse by an overwhelming set of comments.

Furthermore, install and enable variables should have two states, True or False. But allowing commented variables adds a third state, undefined, another unnecessary complication.

holta, 2019-10-10:
> adding a line at the bottom, which was not desired I thought

That's precisely what should happen.

The /usr/bin/iiab-support command does exactly this, for exactly these kinds of reasons &mdash; preserving implementer/operator control over their own /etc/iiab/local_vars.yml file.

For 2 years people have been reading and benefitting from carefully crafted explanations while editing local_vars.yml, explaining their initial options on how to set up IIAB, that I've invested many hundreds of hours in, after consulting with dozens of others on how to explain these options in simple language.

Randomly removing these coaching/comments (those that happen to contain colons, censoring URLs, etc) defies reason; nobody will suggest this with a straight face.

Similarly sending comments into faraway documentation (removing essential context & helpfulness) is counterproductive and not a serious suggestion in the least.

In the end what implementers want and need is context & control as explanatory comments guide their initial IIAB configuration, including their own comments and those generated by volunteers / field implementers / whoever came before them, no matter who that happens to be in the past or future is irrelevant &mdash; the crux is that IIAB's sometimes painful complexity is becoming increasingly much more approachable and understandable every day due to explanations of this sort.

(Perhaps it goes without saying: but these critical explanatory setup comments are exactly the same kind of contextual options that can and increasingly should inform Admin Console selections, as whatever [re]configuration needs become increasingly proven and popular, such that a GUI becomes evident and increasingly required...)

holta, 2020-01-30:
Know problem, that we're hoping to solve in coming months, even if worst case after IIAB 7.1 is released.

holta, 2020-04-19:
Change would happen herein:
https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/iiab-cmdsrv3.py#L2669-L2814

And especially here in this "if commented out" section:
https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/iiab-cmdsrv3.py#L2685-L2701

tim-moody, 2020-04-19:
This should be combined with a planned change to the backend function such that a subset of local vars can be saved, for example only to enable a single service.

tim-moody, 2020-04-21:
I have a rewrite with the following caveat. All comments are preserved as is. There is no attempt to find a commented out variable which has its value changed and uncomment it. The new value will simply be appended.

tim-moody, 2020-04-21:
> My belief is that Admin Console should not modify any comments of any kind, ever, in /etc/iiab/local_vars.yml

if there is general agreement to this we can proceed.

holta, 2020-04-21:
There is general agreement from @jvonau and @georgejhunt.

If you have a rewrite as mentioned [4-5 hours ago](https://github.com/iiab/iiab/issues/1964#issuecomment-617201168) that sounds great.

holta, 2020-04-26:
See https://github.com/iiab/iiab-admin-console/pull/305 "functions to write local vars"

holta, 2020-04-26:
> See [iiab/iiab-admin-console#305](https://github.com/iiab/iiab-admin-console/pull/305) "functions to write local vars"

Now merged: thanks @tim-moody & thanks all for testing Admin Console & [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) pre-releases in general...in preparation for release hopefully very soon, ideally around 2020-05-01!

holta, 2020-04-28:
Great news this appears to be solve according to https://github.com/iiab/iiab/pull/2368#issuecomment-620607502

closed 2020-04-28 by holta