iiab/iiab #1975: HowTo Videos

georgejhunt, 2019-10-01:
- Adam's description of the 10 year old computer labs in Oaxaca, in disrepair, started me on a train of thought. 
- I'd like to create a collection of training videos, that an enterprising teacher, or student might use. I'd like the collection to be under 500MB, and part of the standard IIAB install.
- I started trying to learn how to make them. But as I nosed around, I kept finding stuff that was better than the what I was creating.
- So far, I've created only one video, with 2 or 3 others in various stages of completion. My Idea of what should be next, keeps changing.
- I've created a structured menu at http://iiab.net/info/videos, and another menu, based upon Tim's code, at http://iiab.net/videos.
- The viewer permits the user to select the language for the closed captions, and also has the spoken text translations under selectable buttons. (adding a new language translation turns out to be pretty easy -- though I'm not sure about right to left languages)
- My collection is already over 500MB. But I think I can get the six ka-lite videos down to 50% of their current size.
- So I think we need discussion about whether we want to put them all up on youtube under our own branding. The ka-lite videos are already there (but without the french and spanish closed caption translations).
- Maybe, as an alternative, we link to the videos that are already up at youtube, and just upload the ones we create ourselves. -- Maybe the only thing we really need is a playlist.
- Can we somehow avoid the complexity of replicating the kiwix catalog, and downloading interface in the Admin Console, if we continue to move into making more videos? And we're not willing to have the whole 500MB chunk selectable via local_vars.
- Are the google translated closed captions good enough? Yes there are encode/decode glitches that are pretty easy to fix.

holta, 2019-10-01:
Thanks @georgejhunt as there's a lot happening around the 1st week of December with a teacher's workshop in Mexico, that will depend on exactly these kinds of videos &mdash; thanks to @thotmx and @avni especially &mdash; the more actionable (AKA simpler) these videos are the better!

Both kinds of vids are needed, if there's time!
1) Learning Games depicting how to get to know the most fun/interesting/eye-opening parts of IIAB, "Raspbian With Desktop", browse literacy, information-seeking literacy/curiosity, etc (:
2) IT-oriented vids for the most advanced educators/students wanting to dig deeper, bringing things like http://FAQ.IIAB.IO to life with visceral videos, even if command-line sophistication cannot be expected in most of these schools for now!

holta, 2019-10-02:
I've added a "Tech Skills Videos" link at http://FAQ.IIAB.IO #10 ("What technical documentation exists?") to:

http://iiab.net/info/videos/

holta, 2020-09-12:
@georgejhunt are there any new videos that should be published to http://iiab.net/videos in coming weeks?

(And/or to http://iiab.net/info/videos/ as nec!)