iiab/iiab #1980: Big install failing in step 4 - No package matching 'dansguardian' is available"

mitra42, 2019-10-04:
### Expected behavior and actual behavior.
I'm seeing a failure of the "big" install, with ` No package matching 'dansguardian' is available"` 

### Steps to reproduce the problem.

I'm running the standard install process, and selecting "Big" 

It fails on dansguardian, same if I repeat it. 

### Specifications like the version of the IIAB, operating system version, or hardware details.

This is running on an IIAB, against the master.  The only thing I've changed was to control-C out of the first install run, then replace the internetarchive role with one that I'm trying (and failing) to test, but internetarchive is in Stage 9 and the failure is in Stage 4, so I'm pretty sure its not that change. 

I also notice #1373 also talks about danguardian causing problems, but that was closed a long time ago.

holta, 2019-10-04:
Thank you @mitra42

Can you run `iiab-diagnostic` (or `sudo iiab-diagnostics` if nec) ?

Then send us the resulting output in a pastebin URL?

holta, 2019-10-04:
1) Do you know what's possibly trying to install DansGuardian ?

   FYI it's extremely unusual for anybody to install this package in 2019, which is no longer part of Buster etc &mdash; especially when BIG-sized IIAB has those dansguardian_install and dansguardian_enabled variables very intentionally set to false in https://github.com/iiab/iiab/blob/master/vars/local_vars_big.yml#L156-L158

2) @mitra42 in general plz do identify your exact OS if poss!

   e.g. as shown by the date and "stage2" or "stage5" in /etc/rpi-issue

   e.g. stage2 means "Raspbian Lite", and stage5 means "Raspbian With Desktop"

mitra42, 2019-10-04:
dansguardian was turning up true in my vars. But I also saw a problem where somehow I had what appears to be an old version in /etc/iiab/local_vars.yml despite having new versions in /opt/iiab/iiab/vars/local_*_vars.yml . I can't see where that came from since it was a clean install, but its possible I broke something banging my head against trying to get it to run against my fork of the repo (github.com/mitra42/iiab#mitra42) rather than against github.com/iiab/iiab#master

holta, 2019-10-04:
Yes, odd since we haven't seen DansGuardian installed or enabled in a very long time.

In fact DansGuardian's last stable release was more than a decade ago, so it's very likely both dansguardian variables will be deprecated/removed from `/etc/iiab/local_vars.ym` in very short order as part of the IIAB 7.1 release cycle.

_Ok, do post the results of `iiab-diagnostics` or `sudo iiab-diagnostics` (i.e. the pastebin URL) when you can !_

mitra42, 2019-10-04:
I think I'm too far past that point for it to be useful but here goes ...
```

Accelerate troubleshooting by collecting your IIAB diagnostics into 1 file:

   sudo iiab-diagnostics
   sudo iiab-diagnostics PATH/FILE1 PATH/FILE2 ...

Can you provide a short public nickname: (no spaces!) foo

Compiling diagnostics...

  0. Header/Hashes/OS

  1. Files specially requested: (for "iiab-diagnostics PATH/FILE1 PATH/FILE2")

  2. Regular Files:

     /etc/iiab/openvpn_handle
     /.iiab-image
     /etc/iiab/iiab.env
     /etc/iiab/iiab.ini
     /etc/iiab/local_vars.yml
     /etc/iiab/config_vars.yml
     /etc/resolv.conf
     /etc/network/interfaces
     /usr/bin/iiab-gen-iptables

  3. Content of Directories, 1-level deep:

     /etc/network/interfaces.d
     /etc/systemd/network
     /etc/NetworkManager/system-connections
     /etc/netplan

  4. Output of Commands:

     uname -a  # Linux kernel
     free  # RAM memory
     lscpu  # CPU details
     df -h  # Disk usage
     lsblk  # Partition mount points
     blkid  # Mount point details
     ip addr  # Network interfaces
     ifconfig  # Network interfaces (old view)
     brctl show  # Bridge for LAN side
     netstat -rn  # Routing table
     netstat -natp  # Ports/Services in use
     iptables-save  # Firewall rules
     systemctl status dnsmasq  # Is dnsmasq Ok?
     journalctl -u dnsmasq  # dnsmasq log

  5. Log Files: (last 100 lines of each)

     /opt/iiab/iiab/iiab-install.log
     /opt/iiab/iiab/iiab-network.log
     /opt/iiab/iiab/iiab-debug.log
     /opt/iiab/iiab-admin-console/admin-install.log

COMPLETE! To share this on the web (92K, 1489 lines) run:

pastebinit < /etc/iiab/diag/7.1_raspbian-10_2019-10-05_08:53:34_AEST_foo

```

holta, 2019-10-04:
> pastebinit < /etc/iiab/diag/7.1_raspbian-10_2019-10-05_08:53:34_AEST_foo

@mitra42 can you run the final line above as requested?

mitra42, 2019-10-04:
```
/usr/bin/pastebinit:42: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5
  release = platform.linux_distribution()[0].lower()
/usr/bin/pastebinit:413: DeprecationWarning: pasteURLopener style of invoking requests is deprecated. Use newer urlopen functions/methods
  url_opener = pasteURLopener()
http://paste.debian.net/1104571/
```

holta, 2019-10-05:
@jvonau and I took a look at http://paste.debian.net/1104571 and both dansguardian variables are set to False there, so we cannot reproduce this.

From what we can tell this means that a very stale local_vars.yml triggered this 16h ago, and as such that this will not affect others doing fully fresh/clean installs going forward.

And as a result that this issue can be closed for now.

closed 2019-10-05 by holta

holta, 2019-10-05:
PR https://github.com/iiab/iiab/pull/1983 has been merged to try to warn implementers (more loudly) away from enabling DansGuardian by accident.