iiab/iiab #1986: Install Apache modules upfront, so {Kiwix, Kolibri, Node-RED, etc} playbooks become self-sufficient, so short URL's like http://box/nodered work immediately

holta, 2019-10-06:
As of now, http://box/nodered does not work until the Kiwix (or Kolibri?) playbook is later run.  Others (Kolibri until recently) suffer the very same problem.

The needed code is here...installing 4 Apache modules so URL's work:
https://github.com/iiab/iiab/blob/master/roles/kiwix/tasks/kiwix_install.yml#L74-L83

Or possibly this single Apache is sufficient?
https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/main.yml#L97-L101

QUESTION: can we fix this by installing the essential Apache modules right away in roles/httpd [during Stage 3](https://github.com/iiab/iiab/blob/master/roles/3-base-server/tasks/main.yml#L6-L10) ?

holta, 2019-10-06:
Apache modules proxy_http is clearly necessary to get http://box/kolibri to work!

But does anybody know which-if-any-of-the-3-others (below) are truly necessary for http://box/kiwix, http://box/nodered and others?

- proxy
- proxy_html
- rewrite

jvonau, 2019-10-07:
The matter of which modules to enable would depend on what the apache config files contain, ie ProxyPass, ReWrite.... Looks like kiwix is always installed, but each playbook should contain what is required to make it work with only stage3 installed. Not against stage3 but if you move that code the later roles should have the command removed to have the code run faster without it present.

holta, 2019-10-07:
@jvonau please review PR #1990 if possible ("Install common Apache modules in roles/httpd, for {http://box/kiwix, http://box/kolibri, http://box/nodered, etc}").

PR #1990 (2019-10-07): Install common Apache modules in roles/httpd, for {http://box/kiwix, http://box/kolibri, http://box/nodered, etc}
Resolves #1986, helping more playbooks' short URL's work immediately, i.e. right after running each IIAB app/service's Ansible playbook.

e.g. http://box/kiwix, http://box/kolibri, http://box/nodered, etc.
closed 2019-10-07 by holta