iiab/iiab #1987: internetarchive install failed on RPi 3: "libsecret-1-dev : Depends: libglib2.0-dev"

holta, 2019-10-06:
@mitra42 can you help with this error?

> "The following packages have unmet dependencies:", " libsecret-1-dev : Depends: libglib2.0-dev but it is not going to be installed"

As attempted during a BIG-sized install of IIAB 7.1 / master using http://d.iiab.io onto Raspbian Lite on Raspberry Pi 3:

> TASK [internetarchive : Install packages needed by Distributed Web] ************
fatal: [127.0.0.1]: FAILED! => {"cache_update_time": 1570402490, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"      install 'libsecret-1-dev'' failed: E: Unable to correct problems, you have held broken packages.\n", "rc": 100, "stderr": "E: Unable to correct problems, you have held broken packages.\n", "stderr_lines": ["E: Unable to correct problems, you have held broken packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n libsecret-1-dev : Depends: libglib2.0-dev but it is not going to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " libsecret-1-dev : Depends: libglib2.0-dev but it is not going to be installed"]}

holta, 2019-10-07:
@mitra42 

@jvonau solved this.  It turns out that an a priori installation of Calibre (on RPi) will mess up installation of internetarchive, due to Calibre's incorporation of dependencies from Debian's 'testing' not 'stable' repo (i.e. what will become Debian Bullseye instead of Buster).

So I'll assume this is a non-issue and close it...while moving the Calibre section of Stage 9 to the very final position, in a separate PR now in coming minutes...

closed 2019-10-07 by holta

mitra42, 2019-10-07:
Ok great - because I'd have expected libsecret-1-dev to install any dependencies it needed. I haven't tested on Raspbian Lite.

holta, 2019-10-07:
> Ok great - because I'd have expected libsecret-1-dev to install any dependencies it needed. I haven't tested on Raspbian Lite.

Resolved by PR #1988 now merged!

PR #1988 (2019-10-07): Move Calibre & Calibre-Web installs to VERY END of iiab-install's 9 stages
Resolves #1987