iiab/iiab #1993: kiwix install fails as part of install of current master

tim-moody, 2019-10-08:
TASK [kiwix : Enable & Restart 'kiwix-serve' service] ****************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service kiwix-serve: Job for kiwix-serve.service failed because a fatal signal was delivered to the control process.\nSee \"systemctl status kiwix-serve.service\" and \"journalctl -xe\" for details.\n"}

tim-moody, 2019-10-08:
fails again on rerun of install

tim-moody, 2019-10-08:
but the services is actually running

tim-moody, 2019-10-08:
ran a third time and it succeeded

tim-moody, 2019-10-08:
-- A start job for unit kiwix-serve.service has begun execution.
--
-- The job identifier is 2362.
Oct 08 21:16:11 box.lan systemd[1]: kiwix-serve.service: Control process exited, code=killed, status=11/SEGV
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit kiwix-serve.service has exited.
--
-- The process' exit code is 'killed' and its exit status is 11.
Oct 08 21:16:11 box.lan systemd[1]: kiwix-serve.service: Failed with result 'signal'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit kiwix-serve.service has entered the 'failed' state with result 'signal'.
Oct 08 21:16:11 box.lan systemd[1]: Failed to start Provides the Kiwix ZIM File Server.
-- Subject: A start job for unit kiwix-serve.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit kiwix-serve.service has finished with a failure.
--
-- The job identifier is 2362 and the job result is failed.
Oct 08 21:16:16 box.lan systemd[1]: kiwix-serve.service: Service RestartSec=5s expired, scheduling restart.
Oct 08 21:16:16 box.lan systemd[1]: kiwix-serve.service: Scheduled restart job, restart counter is at 1.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- Automatic restarting of the unit kiwix-serve.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Oct 08 21:16:16 box.lan systemd[1]: Stopped Provides the Kiwix ZIM File Server.
-- Subject: A stop job for unit kiwix-serve.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit kiwix-serve.service has finished.
--
-- The job identifier is 2420 and the job result is done.
Oct 08 21:16:16 box.lan systemd[1]: Starting Provides the Kiwix ZIM File Server...
-- Subject: A start job for unit kiwix-serve.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit kiwix-serve.service has begun execution.
--
-- The job identifier is 2420.
Oct 08 21:16:16 box.lan systemd[1]: Started Provides the Kiwix ZIM File Server.
-- Subject: A start job for unit kiwix-serve.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit kiwix-serve.service has finished successfully.
--
-- The job identifier is 2420.
root@raspberrypi:/opt/iiab/iiab# systemctl status kiwix-serve
● kiwix-serve.service - Provides the Kiwix ZIM File Server
   Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
   Active: active  since Tue 2019-10-08 21:16:16 BST; 1min 21s ago
  Process: 32088 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/ (code=exited, status=0/SUCCESS
 Main PID: 32089 (kiwix-serve)
    Tasks: 5 (limit: 4035)
   Memory: 1.6M
   CGroup: /system.slice/kiwix-serve.service
           └─32089 /opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml --urlRootLocation=/kiwix/

tim-moody, 2019-10-08:
the question is what killed it early in the process

holta, 2019-10-08:
FYI @tim-moody is running Raspbian Lite 2019-09-26 on a RPi 4 with 2GB RAM.

jvonau, 2019-10-09:
status=11/SEGV suggests segmentation fault, that would be in the binary provided by upstream.

tim-moody, 2019-10-09:
I'll raise an issue.

holta, 2019-10-09:
Almost identical problem happened to someone else last week, also with kiwix-tools 3.0.1-6 as released in IIAB 7.0

> I'll raise an issue.

Thanks @tim-moody 

What's the upstream tkt when you can?

tim-moody, 2019-10-09:
kiwix/kiwix-tools#341

tim-moody, 2019-10-09:
opened kiwix/kiwix-tools#341

holta, 2019-10-10:
Ideas:

1) @jvonau suggests we use Ansible's `ignore_errors: yes` for now, until kiwix-tools is fixed.

   Reason: kiwix-serve is eventually started (by its systemd unit file, after several attempts) so the initial failure to start can safely be ignored it seems.

2) @jvonau suggests adding `ExecStopPost=/bin/sleep 10` under the `ExecStart=` line in [/etc/systemd/system/kiwix-serve.service](https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/kiwix-serve.service.j2) &mdash; and then testing by running `systemtl daemon-reload` and then `systemctl restart kiwix-serve`

3) @tim-moody asks if kiwix-tools 2.0.0 or 2.1.0 might work better than 3.0.1-6 ??

4) Several are asking if the very recent Raspbian 2019-09-26 [published around 2019-10-01] or its subsequent apt updates (or a new kernel) might be underlying this problem...

holta, 2019-10-11:
Root cause might be this upstream Raspbian issue: raspberrypi/linux#3271 (thx @tim-moody)

jvonau, 2019-10-11:
Think raspberrypi/linux#3279 is the root cause due to raspberrypi/linux@c0ccb4d

holta, 2019-10-11:
The fix is in &mdash; running `rpi-update` to get pre-release kernel 4.19.79-v7l+ (instead of 4.19.75-v7l+ released yesterday and again today!) somehow does the trick: https://github.com/raspberrypi/linux/issues/3271#issuecomment-541250165

But it might take a few days to be published/mainlined by the Raspberry Pi Foundation, so that everyone will be able to fix things by running...

```
apt update && apt -y dist-upgrade && reboot
```

holta, 2019-10-13:
@jvonau points out we can watch for new kernel packages here:

http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/

(Or run `apt update`)

jvonau, 2019-10-29:
We should really not waste potential users time by forcing a broken kernel to be installed as part of 'iiab' https://github.com/iiab/iiab-factory/pull/84

holta, 2019-10-29:
I've widely published community warnings (on mailing lists etc) that it's as simple as running `rpi-update` to fix the kernel problem -- in the interim until Raspbian finally gets their new kernel mainlined into apt / etc -- hopefully very soon now!

Thanks to @jvonau who provided this link so we can track that upcoming release, which should appear here:
http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/?C=M;O=D

Recap: this issue is annoying but under control, which an easy-as-cake workaround that even non-technical implementers find wonderfully easy.  But perhaps things like iiab/iiab-factory#84 can message this warning even more broadly, if that's easy to do?

holta, 2019-10-29:
If we automate this warning, it should probably say something like:

> Please now run 'rpi-update' to install an experimental Raspbian kernel/firmware at your own risk.  This is needed because of ABC.  After reboot, please then do XYZ...

tim-moody, 2019-10-29:
my concern is whether we should trust this patch for production environments. if not what is the alternative?

holta, 2019-10-29:
> my concern is whether we should trust this patch for production environments. if not what is the alternative?

Yep.  The patch (just run `rpi-update`) has been working for almost ~3 weeks so far, but having it blessed as an official Raspbian apt update (or [published Raspbian release](https://www.raspberrypi.org/downloads/raspbian/)) would be better.

holta, 2019-10-29:
FWIW the `rpi-update` critical patch is helping a lot of people out there...hopefully the Rasp Pi Fdtn will take notice and finally push it making it more official soon:

https://github.com/weaveworks/weave/issues/3717#issuecomment-547545459

jvonau, 2019-10-30:
The big unknown is what happens when apt finally updates a kernel on a machine where rpi-update was run, 'should be fine' (famous words to be burnt by) but my crystal ball is foggy. I'd rather run an older working kernel than do the 'apt update'/rpi-update dance. My general take is rpi-update should never be used unless asked to by an upstream developer as a debugging tool for their end of the code, it's an easy to used tool for them to deploy patches for testing. This now introduces another variable in that what was installed yesterday can differ with what will be installed today. Given the fix appeared in '1273' but now '1274' appeared in rpi-update with new code and potential regressions to debug as this method is a fast moving target. I'd prefer not to force the full dist-upgrade and install a broken kernel. 

> my concern is whether we should trust this patch for production environments. if not what is the alternative? 

Funny you should ask, we do have ready made images I did up for 7.0 dated Oct 8, frozen in time before the kernel issue cropped up. Those images can also be used with master (much better once installed branch hit master) and I'm working with Adam for the needed code to 'iiab' for one question y|n to make that happen.

holta, 2019-11-05:
@tim-moody observed this same kiwix problem on Ubuntu 19.10 (32-bit) on RPi 4:

> TASK [kiwix : Enable & Restart 'kiwix-serve' service] ****************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kiwix-serve: Job for kiwix-serve.service failed because a fatal signal was delivered causing the control process to dump core.\nSee \"systemctl status kiwix-serve.service\" and \"journalctl -xe\" for details.\n"}

Is the very latest Linux kernel necessary for 19.10 as well as for Raspbian?

jvonau, 2019-11-14:
#2025 is forcing the use of rpi-update

holta, 2019-11-14:
> http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/?C=M;O=D

A reminder that kernel packages available via apt can be tracked above.

In addition, new Raspbian releases can be tracked here:
https://downloads.raspberrypi.org/raspbian_lite/images/?C=M;O=D

holta, 2019-11-19:
As mentioned on https://github.com/raspberrypi/linux/issues/3271#issuecomment-555531053

...the next version of Raspbian is likely...

> waiting for something new to show up on https://github.com/raspberrypi/firmware/commits/stable/

holta, 2020-01-22:
Finally a new kernel appeared 3+ months later in Raspbian `apt` update process:

```
root@box:~# uname -a
Linux box.lan 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux
```

Also confirmed by:
http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/?C=M;O=D)

...so I'm closing this ticket (hurrah the `rpi-update` command is no longer necessary at this time!)

PS quite likely new Raspbian ISO / .img images will be published here shortly in coming weeks if not days: https://www.raspberrypi.org/downloads/raspbian/ (for those who don't like to wait for apt updates :)

closed 2020-01-22 by holta