iiab/iiab #1999: Gitea fails on Raspberry Pi

holta, 2019-10-10:
@aidan-fitz can you help us investigate why Gitea is failing on Raspberry Pi?

FYI reverting [roles/gitea/defaults/main.yml#L11](https://github.com/iiab/iiab/blob/master/roles/gitea/defaults/main.yml#L11) to 1.9.3 or 1.9.2 or 1.9.1 or 1.9.0 does not help.

Specifically, this is an example of what fails on IIAB 7.1/master (http://d.iiab.io) on Raspbian Lite 2019-09-26 on Raspberry Pi 4:

```
cd /opt/iiab/iiab
./runrole gitea
systemctl status gitea
```

(And consequently of course, http://box/gitea also fails!)

Kernel is:

```
root@box:~# uname -a
Linux box.lan 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
```

Are there any specific log files you that you can mention to help debug this?

PS all above works on Ubuntu 18.04 in comparison.

holta, 2019-10-10:
Here is an example of Gitea dying immediately (about 1 second later) after its systemd service is started:

```
root@box:~# systemctl start gitea.service

root@box:~# systemctl status gitea.service
● gitea.service - Gitea (Git with a cup of tea)
   Loaded: loaded (/etc/systemd/system/gitea.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-10-10 16:28:38 EDT; 10ms ago
 Main PID: 18707 (gitea-1.9.4)
    Tasks: 0 (limit: 1599)
   Memory: 0B
   CGroup: /system.slice/gitea.service

Oct 10 16:28:38 box.lan systemd[1]: Started Gitea (Git with a cup of tea).
Oct 10 16:28:38 box.lan systemd[1]: gitea.service: Main process exited, code=killed, status=11/SEGV
Oct 10 16:28:38 box.lan systemd[1]: gitea.service: Failed with result 'signal'.

root@box:~# systemctl status gitea.service
● gitea.service - Gitea (Git with a cup of tea)
   Loaded: loaded (/etc/systemd/system/gitea.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: signal) since Thu 2019-10-10 16:28:42 EDT; 431ms ago
  Process: 18709 ExecStart=/library/gitea/bin/gitea-1.9.4 web -c /etc/gitea/app.ini (code=killed, signal=SEGV)
 Main PID: 18709 (code=killed, signal=SEGV)
```

holta, 2019-10-10:
@jvonau suspects this is related to #1993 and kiwix/kiwix-tools#341

Which would mean that Raspbian 2019-09-26 (released October 1st) or one of its libraries...is somehow the underlying cause ?

holta, 2019-10-11:
Thanks @tim-moody who traced this to (likely root cause) upstream Raspbian issue: https://github.com/raspberrypi/linux/issues/3271

jvonau, 2019-10-11:
Think https://github.com/raspberrypi/linux/issues/3279 is the root cause due to https://github.com/raspberrypi/linux/commit/c0ccb4da26bc9eca366f2a437b9d35a749fdde15

holta, 2019-10-11:
The fix is in &mdash; running `rpi-update` to get pre-release kernel 4.19.79-v7l+ (instead of 4.19.75-v7l+ released yesterday and again today!) somehow does the trick: https://github.com/raspberrypi/linux/issues/3271#issuecomment-541250165

But it might take a few days to be published/mainlined by the Raspberry Pi Foundation, so that everyone will be able to fix things by running...

```
apt update && apt -y dist-upgrade && reboot
```

holta, 2019-10-17:
Keeping #1993 open until final resolution, but this is under control.

closed 2019-10-17 by holta