iiab/iiab #2003: CMDSRV requires more than python-systemd -> python3-systemd to run on Ubuntu 19.10

holta, 2019-10-14:
Now that Python 2 support is gone from Ubuntu 19.10 etc (https://launchpad.net/ubuntu/+source/python-systemd/234-3) what else is needed to make CMDSRV run below?

CONTEXT: building off PR #2002, we have 1 core remaining issue to get IIAB running on Ubuntu 19.10 that is being released this week on 2019-10-17.

> 4:09 PM
@tim-moody any recommendations?
>
> TASK [cmdsrv : Install packages for cmdsrv] ************************************
changed: [127.0.0.1] => (item=libzmq3-dev)
changed: [127.0.0.1] => (item=python-cracklib)
ok: [127.0.0.1] => (item=pkg-config)
changed: [127.0.0.1] => (item=sqlite3)
changed: [127.0.0.1] => (item=libapache2-mod-wsgi)
 [WARNING]: Could not find aptitude. Using apt-get instead
>
> failed: [127.0.0.1] (item=python-systemd) => {"ansible_loop_var": "item", "changed": false, "item": "python-systemd", "msg": "No package matching 'python-systemd' is available"}

> 4:59 PM
Does this mean yet another case of python-xyz to python3-xyz (i.e. python-systemd -> python3-systemd) on Line 13 of https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/tasks/main.yml#L13 ?
>
> root@box:~# apt -a list python-systemd python3-systemd
Listing... Done
python3-systemd/eoan,now 234-3 amd64 [installed]
>
> python3-systemd/eoan 234-3 i386

> 5:05 PM
Close but no cigar?  I tried the above python3-systemd, and got much further, but then it failed here:
>
> TASK [cmdsrv : Enable iiab-cmdsrv service] *************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service iiab-cmdsrv: Job for iiab-cmdsrv.service failed because the control process exited with error code.\nSee \"systemctl status iiab-cmdsrv.service\" and \"journalctl -xe\" for details.\n"}
root@box:~# systemctl status iiab-cmdsrv
>
> ● iiab-cmdsrv.service - Provides the IIAB Command Server
   Loaded: loaded (/etc/systemd/system/iiab-cmdsrv.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-10-14 17:03:20 EDT; 4min 27s ago
  Process: 7048 ExecStart=/opt/admin/cmdsrv/iiab-cmdsrv --daemon (code=exited, status=1/FAILURE)
>
> Oct 14 17:03:20 box.lan systemd[1]: Starting Provides the IIAB Command Server...
Oct 14 17:03:20 box.lan iiab-cmdsrv[7048]: Traceback (most recent call last):
Oct 14 17:03:20 box.lan iiab-cmdsrv[7048]:   File "/opt/admin/cmdsrv/iiab-cmdsrv", line 19, in <module>
Oct 14 17:03:20 box.lan iiab-cmdsrv[7048]:     from systemd import journal
Oct 14 17:03:20 box.lan iiab-cmdsrv[7048]: ImportError: No module named systemd
Oct 14 17:03:20 box.lan systemd[1]: iiab-cmdsrv.service: Control process exited, code=exited, status=1/FAILURE
Oct 14 17:03:20 box.lan systemd[1]: iiab-cmdsrv.service: Failed with result 'exit-code'.
Oct 14 17:03:20 box.lan systemd[1]: Failed to start Provides the IIAB Command Server.

holta, 2019-10-30:
Python 3 umbrella tkt https://github.com/iiab/iiab/issues/1810#issuecomment-548020264 mentions:

> mod_wsgi for apache can only be python 2 OR 3, not both. it is used by
captive portal
cmdsrv
pathagar

jvonau, 2019-11-10:
This issue won't go away while 'from systemd import journal' is in roles/cmdsrv/files/iiab-cmdsrv and a python2 shabang present. Apt has no way to supply this python2 library for cmdsrv to use.

holta, 2019-12-12:
@tim-moody @jvonau is this issue now resolved?

tim-moody, 2019-12-12:
yes

holta, 2019-12-21:
Confirmed with a fresh install of http://download.iiab.io on Ubuntu Server 19.10

i.e. http://10.8.0.42/admin

closed 2019-12-21 by holta