iiab/iiab #2005: Transition from Node.js 10 to 12 as it becomes LTS on 2019-10-21

holta, 2019-10-15:
@m-anish @llaske @mitra42 do you forsee any problems upgrading Sugarizer, Node-RED and "Internet Archive Offline" etc from Node.js 10 to 12?

As Node.js 12 becomes LTS starting Monday in 6 days from now &mdash; please see:

- https://nodejs.org/en/about/releases/
- https://nodejs.org/en/download/releases/

Aside: the new Ubuntu 19.10 apt installs Node.js v10.15.2 and npm 5.8.0, both of which are a bit dated, but just happen to be the easiest way to test this OS for now &mdash; given that 19.10's official OS release is ~48h away (on Thur 2019-10-17) &mdash; whereupon Node.js's [official/recommended curl-based installation technique](https://github.com/nodesource/distributions#installation-instructions) should begin to work:

```
# Run as root

curl -sL https://deb.nodesource.com/setup_12.x | bash -
apt-get install -y nodejs
```

mikkokotila, 2019-10-15:
I think Node 10 is totally ok for now. Unless there is a specific reason to move to 12 already.

mitra42, 2019-10-15:
I have seen no problems with v12 all of our code is happy with v10 or v12 so I think you could upgrade when you get confirmation from the other packages. 

What I did in one of my previous PR's was change the `node` code in the `internetarchive` role to be happy with either 10.x or 12.x . 

I think (but can't be positive I'm correctly remembering) that I made that change because an installation broke. Which I believe was because the nodejs role gets the latest support release and so probably automatically gets 12.x now.
```
- name: FAIL (STOP INSTALLING) IF nodejs_version is not set to 10.x or 12.x
  fail:
    msg: "Internet Archive install cannot proceed, as it currently requires Node.js 10.x or 12.x, and your nodejs_version is set to {{ nodejs_version }}.  Please check the value of nodejs_version in /opt/iiab/iiab/vars/default_vars.yml and possibly also /etc/iiab/local_vars.yml"
  when: internetarchive_install and (nodejs_version != "10.x") and (nodejs_version != "12.x")
```

mitra42, 2019-10-15:
Looking at Sugarizer it doesnt appear to be installing node (so I hope it only installs after internetarchive or something else that does)

mitra42, 2019-10-15:
Looks like both sugarizer and node-red just rely on any nodejs being installed, neither of them check the version (like internetarchive does), that's a bit unsafe but maybe they work with the old (32-bit) ones like node8.

mikkokotila, 2019-10-15:
For Node-Red [12 is still marked as tentative](https://nodered.org/docs/faq/node-versions).

mikkokotila, 2019-10-15:
What I missed I guess here is the context to why make the move right now. Because how many patching takes place with new LTS on almost any software, it seems empirically sound to avoid switching at the earliest possible moment unless there is a strong reason for doing it. Node 10 should be just fine for a while still, and there is nothing "out-dated" about it. The web runs on it.

holta, 2019-10-15:
Inevitably IIAB get complaints when we don't support LTS releases, of whatever it happens to be.

So it's not unhealthy for us to start kicking the tires of Node.js 12 six months after its release, as it's promoted to LTS status this coming Monday, to make sure it's genuinely available and workable with certain IIAB's -- if only so that IIAB isn't accused of restricting developers/implementers options.

ASIDE: we also need to keep our eye on and prepare for April 2020 when Node.js 10 will no longer be LTS.  Effectively we're forced to support Node.js 10 and 12 during this 6-month transition period...in order to give devs options...such is the price we pay for supporting [so many different IIAB Apps/Services](http://wiki.laptop.org/go/IIAB/FAQ#What_services_.28IIAB_apps.29_are_suggested_during_installation.3F) (:

mitra42, 2019-10-15:
@holta - you need to add "pbx" to your list, it looks like it is specifically required v10.x in the install. 

I'd suggest an upgrade with potential to backout if any problems. Node's upgrades have been relatively good in my experience, because their odd number releases are the unstable ones, but of course problems are possible. 

This is useful https://medium.com/@nodejs/introducing-node-js-12-76c41a1b3f3f

The only part which caused concern for me was 
```
The increment in minimum compiler and libc requirements also increments minimums in supported platforms. Platforms using glibc (most platforms other than macOS and Windows) must now include a minimum version of 2.17. Common Linux platforms compatible with this version include Enterprise Linux 7 (RHEL and CentOS), Debian 8 and Ubuntu 14.04. Binaries available from nodejs.org will be compatible with these systems. Users needing to compile their own binaries on systems not natively supporting GCC 6 may need to use a custom toolchain. Even though Node.js 12.0.0 may compile with older compilers, expect the Node.js 12 codebase (including V8) to rapidly adopt C++ features supported by GCC 6 during the pre-LTS timeframe.

Windows minimums remain the same as Node.js 11, requiring at least Windows 7, 2008 R2 or 2012 R2 and a minimum compiler of Visual Studio 2017. macOS users needing to compile Node.js will require a minimum of Xcode 8 and Node.js binaries made available on nodejs.org will only support a minimum of macOS 10.10 “Yosemite”
```
which implies there COULD be problems if we have native modules that try and compile, AND any of the platforms we are doing that on dont support GCC6. 

Looking at a couple of the features, especially defining variables in classes instead of constructors, I'd expect code to move to being node v12.x reasonably quickly, so we might have no choice before long.

holta, 2019-10-15:
> @holta - you need to add "pbx" to your list, it looks like it is specifically required v10.x in the install.

@m-anish can you look into this?  We have time to get to know Node.js 12 before Node.js 10 loses its LTS status in April 2020, but again let's be prepared (:

mitra42, 2019-10-15:
I'll take responsibility to keep internetarchive compatible with 10.x or 12.x from now until you make the switch. (I believe it works on node12 but can't check at the moment as I'm offline)

llaske, 2019-10-17:
@holta I've just tested Sugarizer Server v1.1.1 with node v12.11.1 and everything seems to work.

@mitra42 Sugarizer Server works with node v6 to v10 from now. But you're right, it will be better to check the node version before launch. I've filled a [bug](https://github.com/llaske/sugarizer-server/issues/201) on it.

holta, 2019-12-16:
Please also see the Node.js 10.x -> 12.x notes for this upcoming Thursday's IIAB call (2019-12-19) at http://minutes.iiab.io near the bottom of 2.a. (currently this is item ix. but it may move around!)

(@jvonau raises the concern that the [nodesource.com process for installing Node.js](https://github.com/nodesource/distributions#installation-instructions) introduces Python 2 onto systems where that might not be wanted!?)

holta, 2019-12-16:
jvonau writes: "Once python2 is fully removed by either debian or ubuntu the nodesource install of nodejs will break on that distro if the ppa is not corrected before then.'

holta, 2019-12-26:
Implemented by PR #2075

closed 2019-12-26 by holta