iiab/iiab #2013: Can python3-geojson be made to work, as required by OS's like Ubuntu 19.10+? [osm-vector-maps]

holta, 2019-10-20:
Thanks @georgejhunt for clarifying when you can!

Refs:
PR #2002 
PR #2011

jvonau, 2019-11-19:
I don't think there is any python code that uses geojson directly and can be removed as pylint returns unused-import on /usr/bin/iiab-update-maps
> W:  4, 0: Unused FeatureCollection imported from geojson (unused-import)
W:  4, 0: Unused Point imported from geojson (unused-import)
W:  4, 0: Unused Polygon imported from geojson (unused-import)
W:  4, 0: Unused Feature imported from geojson (unused-import)
W:  5, 0: Unused import geojson (unused-import)

Commenting out those imports doesn't seem to affect running it but I don't have a data set to test with.

holta, 2019-11-22:
Recap: https://github.com/iiab/iiab/pull/2042#issuecomment-557616995 is another report installing an experimental version of IIAB (http://d.iiab.io/install-test.txt) onto Ubuntu 19.10, that ran into python-geojson package issues.

(Presumably we need to transition to python3-geojson instead?)

georgejhunt, 2019-11-22:
> Recap: #2042 (comment)

I agree with Jerry. I don't think python*-geojson needs to be installed. It
is used during map generation, but not deployment.

holta, 2019-12-09:
@georgejhunt is there any workaround possible here?

When installing http://d.iiab.io onto Ubuntu Server 19.10, it fails as follows:

> TASK [console : Install packages for console] **********************************
ok: [127.0.0.1] => (item=libapache2-mod-authnz-external)
ok: [127.0.0.1] => (item=pwauth)
failed: [127.0.0.1] (item=python-geojson) => {"ansible_loop_var": "item", "changed": false, "item": "python-geojson", "msg": "No package matching 'python-geojson' is available"}

holta, 2019-12-10:
> @georgejhunt is there any workaround possible here, that would allow Admin Console to install?
> 
> (When installing http://d.iiab.io onto Ubuntu Server 19.10, it fails as follows...)
> 
> > TASK [console : Install packages for console] **********************************
> > ok: [127.0.0.1] => (item=libapache2-mod-authnz-external)
> > ok: [127.0.0.1] => (item=pwauth)
> > failed: [127.0.0.1] (item=python-geojson) => {"ansible_loop_var": "item", "changed": false, "item": "python-geojson", "msg": "No package matching 'python-geojson' is available"}

@tim-moody has experienced the exact same problem installing IIAB onto 64-bit Ubuntu 19.10 for Raspberry Pi 4.

Let's try to cook up up (any) kind of workaround in coming weeks now to clear this important roadblock:

> https://pypi.org/project/geojson/https://pypi.org/project/geojson/ says the one installed with pip is compatible with python 3

tim-moody, 2019-12-10:
Think @jvonau  already discovered that there is no package, only pip. But I don't know if pip3 will install python-geojson. pip will, but can we do that in ansible under python 3.

tim-moody, 2019-12-10:
btw with python-geojson installed on 19.10 on rpi maps run

tim-moody, 2019-12-10:
https://docs.ansible.com/ansible/latest/modules/pip_module.html says the executable can be specified, so pip should be possible

tim-moody, 2019-12-10:
but it needs to be installed in 2-common or somewhere

tim-moody, 2019-12-10:
actually  pip3 install geojson also works

jvonau, 2019-12-10:
> I agree with Jerry. I don't think python*-geojson needs to be installed. It is used during map generation, but not deployment.

Based on the above feedback the import of geojson was disabled in 99ce889308d228151ccb69cf01dbe98b3f6fd6d9, but has been replace by the refactoring done to enable [iiab-update-map3.py](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/scripts/iiab-update-map3.py). I don't see an import of geojson in either [iiab_adm.py](https://github.com/iiab/iiab-admin-console/blob/master/roles/common/templates/adm_lib.py) or [iiab_lib.py](https://github.com/iiab/iiab/blob/master/roles/2-common/templates/iiab_lib.py) perhaps the package can be omitted altogether? BTW the old version worked without geojson installed, that is what install-test.txt tested and successfully ran to competition.

jvonau, 2019-12-10:
https://github.com/iiab/iiab-admin-console/pull/270

holta, 2019-12-10:
@georgejhunt can I close this tkt, if test-installing http://download.iiab.io now works onto Ubuntu 19.10 ?

holta, 2019-12-10:
Installation of osm-vector-maps Map Packs no longer works.  Uncovered by @tim-moody's testing:

> I get 404 on all the pbf and jpeg assets

> [Central America Map Pack] didn't work on ub 18.04 either

> no complaints about install in adm cons, but no map shows (has same 404s as on 19.10)

> this worked a couple of days ago

> as expected apt install python-geojson made no difference

tim-moody, 2019-12-10:
urls like http://192.168.3.92/osm-vector-maps/en-osm-omt_central_america_2017-07-03_v0.23/tileserver.php/detail/1/1/0.pbf
return 404
as do http://192.168.3.38/osm-vector-maps/en-osm-omt_central_america_2017-07-03_v0.23/tileserver.php/satellite/2/0/2.jpeg

holta, 2019-12-10:
FYI @georgejhunt and others are working on this&mdash;excerpt:

> GH: I think I just overlooked writing a nginx config file for the php component of osm.  

> TM: 2019/12/10 16:26:24 [error] 802#802: *4 open() "/library/www/html/common/css/bootstrap.min.css.map" failed (2: No such file or directory)
>
> /library/www/osm-vector-maps/maplist/main.js.map" failed (2: No such file or directory)

> GH: When I've been trying to debug stuff, the failure of gets for *.map was a debugging thing that the debug console was adding

> TM: it's in the apache log

> GH: I need to add a location clause in the nginx config file for /maps(.*)\.php similar to the treatment for admin-console. I think it's in my court

> TM: like WordPress:
>
>location /wordpress {
  proxy_pass http://127.0.0.1:8090/wordpress;
}
 location ~ /wordpress/.*\.php$ {
>
>    proxy_set_header X-Real-IP  $remote_addr;
    proxy_set_header X-Forwarded-For $remote_addr;
    proxy_set_header Host $host;
    proxy_pass http://127.0.0.1:8090;
>
>}

> GH: exactly

> JV: that uses apache to support proxy pass... 

> GH: But this is proxying to apache, at this point my first inclination to to do it with fastcgi php

> TM: go for it
>
> like adm cons
>
> location ~ /admin/(.*)\.php$ {
    alias /opt/admin/console/$1.php;
    proxy_set_header X-Real-IP  $remote_addr;
    proxy_set_header X-Forwarded-For $remote_addr;
    proxy_set_header Host $host;
    fastcgi_pass php;
    fastcgi_index index.php;
    fastcgi_split_path_info ^(.+\.php)(/.+)$;
    fastcgi_param   SCRIPT_FILENAME    $document_root$fastcgi_script_name;
    fastcgi_param   SCRIPT_NAME        $fastcgi_script_name;
    include fastcgi_params;
}

> JV: yup that is the one I was looking for.. 

> GH: The location clause will need to match on both maps|osm-vector-maps/(.*)\.php

holta, 2019-12-10:
@georgejhunt please LMK if #2013 can be closed (if the just-merged fix PR #2058 "Include a php clause in NGINX conf" is confirmed solid?)

holta, 2019-12-21:
Can this be closed now?!

closed 2019-12-21 by georgejhunt