iiab/iiab #2021: Nextcloud stores its data in /opt/ instead of /library/

m-anish, 2019-10-28:
### Expected behavior and actual behavior.
I had experienced this on the IIAB in Zanskar so I am creating an issue so that it can be checked against latest master.

Nextcloud creates and stores user data in `/opt/` rather than `/library/`. For IIAB setups that have these directories in different partitions, this presents a problem. Even though there is a nextcloud directory in `/library` (IIRC), it still stores its data in a directory in `/opt`

holta, 2019-11-01:
Thanks @m-anish!

Can I assign this tkt to you to investigate solution(s)?

holta, 2019-12-20:
@m-anish can you investigate if this is still a serious issue?

m-anish, 2020-01-09:
I'll fix this as part of the ongoing nginx port.

holta, 2020-02-18:
> I'll fix this as part of the ongoing nginx port.

Thanks!  Resolved by PR #2255 that built upon @m-anish's PR #2119

closed 2020-02-18 by holta