iiab/iiab #2022: Apache warnings.

jvonau, 2019-10-30:
### Expected behavior and actual behavior.
A warning free restart.

Oct 30 00:00:01 box.lan apachectl[27096]: [Wed Oct 30 00:00:01.608252 2019] [alias:warn] [pid 27100] AH00671: The Alias directive in /etc/apache2/sites-enabled/010-iiab.conf at line 273 will probably never match because it overlaps an earlier Alias.
Oct 30 00:00:01 box.lan apachectl[27096]: [Wed Oct 30 00:00:01.609414 2019] [alias:warn] [pid 27100] AH00671: The ScriptAlias directive in /etc/apache2/sites-enabled/010-iiab.conf at line 297 will probably never match because it overlaps an earlier ScriptAlias.
Oct 30 00:00:01 box.lan apachectl[27096]: [Wed Oct 30 00:00:01.610011 2019] [alias:warn] [pid 27100] AH00671: The ScriptAlias directive in /etc/apache2/sites-enabled/awstats.conf at line 43 will probably never match because it overlaps an earlier ScriptAlias.
Oct 30 00:00:01 box.lan apachectl[27096]: [Wed Oct 30 00:00:01.610547 2019] [alias:warn] [pid 27100] AH00671: The Alias directive in /etc/apache2/sites-enabled/munin24.conf at line 1 will probably never match because it overlaps an earlier Alias.
Oct 30 00:00:01 box.lan apachectl[27096]: [Wed Oct 30 00:00:01.610828 2019] [alias:warn] [pid 27100] AH00671: The ScriptAlias directive in /etc/apache2/sites-enabled/munin24.conf at line 9 will probably never match because it overlaps an earlier ScriptAlias.
Oct 30 00:00:01 box.lan apachectl[27096]: [Wed Oct 30 00:00:01.611197 2019] [alias:warn] [pid 27100] AH00671: The Alias directive in /etc/apache2/sites-enabled/osm.conf at line 2 will probably never match because it overlaps an earlier Alias.
Oct 30 00:00:01 box.lan systemd[1]: Reloaded The Apache HTTP Server.

### Steps to reproduce the problem.

restart apache2

### Specifications like the version of the IIAB, operating system version, or hardware details.

general I would think.

jvonau, 2019-11-02:
#877

jvonau, 2019-11-03:
It's a bug, you are supplying some apache2 config files that duplicate some directives.

holta, 2019-11-03:
> It's a few bugs, [IIAB is] supplying some apache2 config files that duplicate some directives.

Can you list these duplicates?

jvonau, 2019-11-03:
osm.conf:Alias /maps /library/www/html/modules/en-osm-omt-min/
> cd /etc/apache2/sites-enabled/
grep Alias *
osm.conf:Alias /maps /library/www/html/modules/en-osm-omt-min/
osm.conf:Alias /osm /library/www/html/modules/en-osm-omt-min/
osm-vector-maps.conf:Alias /map /library/www/osm-vector-maps
osm-vector-maps.conf:Alias /maps /library/www/osm-vector-maps
osm-vector-maps.conf:Alias /osm-vector-maps /library/www/osm-vector-maps

I'll go with /maps as one...

tim-moody, 2019-11-03:
the warnings are just the tip of the iceberg. iiab has directives that go back to OLPC XS, still there because of the practice of copying all defaults into the configuration file so no one cat tell what directive belongs there and what is a possibly obsolete directive from somewhere else. I tried to sort this out several years ago and gave up.

jvonau, 2019-11-03:
Think then we need to reconcile what is supplied into sites-available, looks like awstats is complaining about ScriptAlias that would be cgi related.
> pi@box:/etc/apache2/sites-available $ grep ScriptAlias *
010-iiab.conf:# ScriptAlias: This controls which directories contain server scripts.
010-iiab.conf:# ScriptAliases are essentially the same as Aliases, except that
010-iiab.conf:# The same rules about trailing "/" apply to ScriptAlias directives as to
010-iiab.conf:ScriptAlias /cgi-bin/ "/library/www/cgi-bin/"
010-iiab.conf:# "/var/www/cgi-bin" should be changed to whatever your ScriptAliased
awstats.conf:ScriptAlias /awstats /usr/lib/cgi-bin
awstats.conf:ScriptAlias /cgi-bin/ /usr/lib/cgi-bin/
cmd-service.conf:WSGIScriptAlias /cmd-service /opt/admin/console/cmd-service.wsgi
munin24.conf:ScriptAlias /munin-cgi/munin-cgi-graph /usr/lib/munin/cgi/munin-cgi-graph

Full list of Alias
>pi@box:/etc/apache2/sites-available $ grep Alias *
010-iiab.conf:   ServerAlias box.lan
010-iiab.conf:# Aliases: Add here as many aliases as you need (with no limit). The format is 
010-iiab.conf:# Alias fakename realname
010-iiab.conf:Alias /icons/ "/usr/share/httpd/icons/"
010-iiab.conf:# ScriptAlias: This controls which directories contain server scripts.
010-iiab.conf:# ScriptAliases are essentially the same as Aliases, except that
010-iiab.conf:# The same rules about trailing "/" apply to ScriptAlias directives as to
010-iiab.conf:# Alias.
010-iiab.conf:ScriptAlias /cgi-bin/ "/library/www/cgi-bin/"
010-iiab.conf:# "/var/www/cgi-bin" should be changed to whatever your ScriptAliased
010-iiab.conf:# We use Alias to redirect any /error/HTTP_<error>.html.var response to
010-iiab.conf:#   Alias /error/include/ "/your/include/path/"
010-iiab.conf:Alias /error/ "/library/www/error/"
022-moodle.conf:Alias /moodle /opt/iiab/moodle
admin-console.conf:Alias /admin "/opt/admin/console"
awstats.conf:ScriptAlias /awstats /usr/lib/cgi-bin
awstats.conf:Alias /awstats-icon/ /usr/share/awstats/icon/
awstats.conf:Alias /awstatsicons/ /usr/share/awstats/icon/
awstats.conf:Alias /awstatsclasses/ /usr/share/java/awstats/
awstats.conf:ScriptAlias /cgi-bin/ /usr/lib/cgi-bin/
cmd-service.conf:WSGIScriptAlias /cmd-service /opt/admin/console/cmd-service.wsgi
content_dir.conf:Alias /content /library/www/html/local_content
content_dir.conf:Alias /local_content /library/www/html/local_content
content_dir.conf:Alias /USB /library/www/html/local_content
content_dir.conf:Alias /usb /library/www/html/local_content
dokuwiki.conf:Alias /wiki /library/dokuwiki
elgg.conf:Alias /elgg /opt/elgg
lokole.conf:Alias /lokole/static /library/lokole
mediawiki.conf:Alias /mediawiki /library/mediawiki-1.33.1
munin24.conf:Alias /munin /var/cache/munin/www
munin24.conf:ScriptAlias /munin-cgi/munin-cgi-graph /usr/lib/munin/cgi/munin-cgi-graph
nextcloud.conf:Alias /nextcloud /opt/nextcloud
osm.conf:Alias /maps /library/www/html/modules/en-osm-omt-min/
osm.conf:Alias /osm /library/www/html/modules/en-osm-omt-min/
osm-vector-maps.conf:Alias /map /library/www/osm-vector-maps
osm-vector-maps.conf:Alias /maps /library/www/osm-vector-maps
osm-vector-maps.conf:Alias /osm-vector-maps /library/www/osm-vector-maps
wordpress.conf:Alias /wordpress /library/wordpress

holta, 2019-11-03:
The [above-mentioned warning](https://github.com/iiab/iiab/issues/2022#issue-514889612) mentions these 6 lines... (from these 4 files)
<br>

> Alias /icons/ "/usr/share/httpd/icons/"
> ...
> ScriptAlias /cgi-bin/ "{{ doc_base }}/cgi-bin/"    # ScriptAlias /cgi-bin/ "/library/www/cgi-bin/"

from https://github.com/iiab/iiab/blob/master/roles/httpd/templates/010-iiab.conf.j2#L279-L303

> ScriptAlias /cgi-bin/ /usr/lib/cgi-bin/

from https://github.com/iiab/iiab/blob/master/roles/awstats/templates/apache.conf#L43

> Alias /munin /var/cache/munin/www
> ...
> ScriptAlias /munin-cgi/munin-cgi-graph /usr/lib/munin/cgi/munin-cgi-graph

from https://github.com/iiab/iiab/blob/master/roles/munin/templates/munin24.conf.j2#L1-L9

> Alias /maps /library/www/html/modules/en-osm-omt-min/

from https://github.com/iiab/iiab/blob/master/roles/httpd/files/osm.conf#L2

closed 2020-01-30 by jvonau