iiab/iiab #2035: fresh IIAB install on Raz4 does not load home page correctly

kodepharm, 2019-11-12:
### Expected behavior and actual behavior.
i am new to IIAB so i am not sure if this is the right place to ask for help. installed raspian and ran the one line command to install IIAB with option 3 for content. everything completed fine but when i visit box.lan/home none of the menu items show only a power off option is presented but if i type in box.lan/maps or box.lan/books it does load those applications. 

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.
the latest IIAB, raspian 10 buster, raspberry 4

holta, 2019-11-12:
1) @kodepharm can I ask if you ran `rpi-update` to deal with Raspbian's recent kernel issue described at #1993 ?

Here is a recommended approach, prior to installing IIAB:
```
sudo su -
apt update
apt -y dist-upgrade
rpi-update
reboot
```

2) Can you run `sudo iiab-diagnostics` and reply with the pastebin URL to give us details?

tim-moody, 2019-11-12:
sounds like the install mostly went OK if you see the poweroff button. (green background, blue header?)

the menu is entirely in javascript. what browser are you on and have you turned javascript off?

on firefox or chrome press F12 and look for javascript errors

kodepharm, 2019-11-12:
@holta 

1)i ran all other commands except "rpi-update"

2) http://paste.debian.net/1115947/

holta, 2019-11-12:
> 1. i ran all other commands except "rpi-update"  ... what would happen i ran that command now?
> 2. http://paste.debian.net/1115947/

1) @kodepharm the above shows you're running kernel 4.19.81-v7l+, so running `rpi-update` is "magically" no longer necessary!  (Likely due to a very recent `apt` update that I was not aware of.)

2) Can you reply to @tim-moody [above](https://github.com/iiab/iiab/issues/2035#issuecomment-553021347) ?

kodepharm, 2019-11-12:
@tim-moody 
i added a screenshot of the home page hopefully you can see it. javascript is on and i have tried on 3 browsers(chrome, firefox, safari) same thing. inspecting the code doesn't seem to give me much lead.  

[image]

kodepharm, 2019-11-13:
@kodepharm  @tim-moody  before i do a fresh re-install i wanted to see if there was any new suggestions i could try?  thanks

tim-moody, 2019-11-13:
@kodepharm at the very least the css is missing, which makes me think the javascript is also missing. F12 should have revealed that.

Before doing a complete install I suggest
```
cd /opt/iiab/iiab-admin-console
./install

```

tim-moody, 2019-11-13:
Actually, if you did your install  before Nov 6 you need to do

```
cd /opt/iiab/iiab-admin-console
git pull
./install

There was a bug that caused css and js to be missing.

kodepharm, 2019-11-13:
@tim-moody thanks that fixed my issue. so new IIAB installs will not have this issue i am assuming.

holta, 2019-11-13:
> @tim-moody thanks that fixed my issue. so new IIAB installs will not have this issue i am assuming.

Thanks @kodepharm -- I'm closing this issue now if indeed it's taken care of.  Please contact me/us directly if you have more questions and/or want to be more involved with other global/implementer communities using Internet-in-a-Box, Thanks!

My email address is here:
http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F

holta, 2019-11-14:
Hi again @kodepharm !  Others who install Raspbian are not getting kernel 4.19.81+ as you did.  (Others are getting kernel 4.19.75, wishing they had what you have!)

- Can you reconfirm that you and your colleagues never ran `rpi-update` ?
- Can you explain how you obtained & installed Raspbian originally?

Thanks if possible!

kodepharm, 2019-11-14:
@holta  i did not run rip-update and i got the Raspbian lite image from the Raspbian site recommended in the IIAB how to docs.

holta, 2019-12-21:
@tim-moody can you clarify recent progress on the "nothing appears except for Power Off" issue, and close this ticket ideally?

holta, 2019-12-21:
> @tim-moody can you clarify recent progress on the "nothing appears except for Power Off" issue, and close this ticket ideally?

@kodepharm here's a possible fix being worked on as we speak:

https://github.com/iiab/iiab-admin-console/pull/275 "race condition processing page before loading it"

holta, 2019-12-23:
> @kodepharm here's a possible fix being worked on as we speak:
> 
> [iiab/iiab-admin-console#275](https://github.com/iiab/iiab-admin-console/pull/275) "race condition processing page before loading it"

Can this #2035 be closed now?

kodepharm, 2019-12-23:
please close it.

closed 2019-12-23 by holta