iiab/iiab #2040: http://box/usb behaves differently on IIAB 7.0 built on A. [ Raspberry Pi Desktop (for PC and Mac)]  compared to B. Raspbian Buster Lite

shanti-bhardwa, 2019-11-14:
### Expected behavior and actual behavior.
A. is downloaded from https://www.raspberrypi.org/downloads/raspberry-pi-desktop/ and has the Raspbian Desktop.
B. is downloaded from  https://www.raspberrypi.org/downloads/raspbian/ as Raspbian Buster Lite which only has the command line interface.

The Desktop software A mounts the USB device to /media/pi as a default and you can use File Manager to explore the file system on the device.

IIAB  B expects the device to be mounted on /media/usb0 to usb7.

So you  cannot see the content from USB or load modules or zim files using  Admin Command Server when these file exist on the  external device under  /library.

from Manage Content > Find USB device you can see the device as mounted on /media/pi/<name of device or partition label> and list of zims files amd modules.

If you try to downlaod from the mounted USB device, you get the error COPY-OER2GO-MODULE failed or COPY-ZIMs Failed with invalid Source.

This is fair because IIAB is looking to use /media/usbx and Desktop uses /media/pi.

I can go into the Terminal and umount the device from /media/pi and the mount it on /media/usb3 say.

Then you can see the new mount point from Admin Console and  download the content from the Admin Console and it works fine.

But publishing the  content via USB from the Home page does not work.

### Steps to reproduce the problem.

Insert external USB device into PC running A.

### Specifications like the version of the IIAB, operating system version, or hardware details.

IIAB 7.0 on A. & B, installed in both case using the 1 line install script

curl d.iiab.io/6.7/install.txt | sudo bash

holta, 2019-11-14:
@georgejhunt can you take a quick look at this?

At a glace, I'd summarize it as "http://box/usb behaves differently on Raspbian-for-PC-and-Mac" (as compared to Raspbian on actual Raspberry Pi hardware).

holta, 2019-11-14:
Earlier today, @jvonau suggested the issue might be...?

> the timing of the udev rules that are used by udisk2/usbmount

jvonau, 2019-11-15:
@shanti-bhardwa There has been a change in upstream naming of the option in the system unit file [here.](https://github.com/systemd/systemd/pull/9153) Could you as root edit /etc/systemd/system/systemd-udevd.service changing PrivateMounts=yes to become
PrivateMounts=no, save the file, reboot and retest please.

holta, 2019-11-15:
Side note that IIAB supports installation onto these 3 Linux OS distributions at this time:
- Raspbian on Raspberry Pi 3/4
- Ubuntu 18.04 on x86_64
- Debian 10 on x86_64

( See https://github.com/iiab/iiab/wiki/IIAB-Platforms )

Whereas Raspbian for x86_64 / PC / Mac is not supported at this time, but if later appropriate, it could be added to the associated list (in link above) of experimentally supported OS's that are "theoretically possible, but may require extensive babysitting to get right".

shanti-bhardwa, 2019-11-15:
on A. Re-tested with editing /etc/systemd/system/systemd-udevd.service changing:
PrivateMounts=yes to become PrivateMounts=no

Does NOT fix the issue.

shanti-bhardwa, 2019-11-15:
where can I download 
Ubuntu 18.04 on x86_64 and Debian 10 on x86_64
and do they have a desktop as well

shanti-bhardwa, 2019-11-15:
Here is comparative out between A. and B.
on A. $ mount | grep /dev/sd 
gives
/dev/sdb2 on /media/pi/REPO-rootfs type ext4 (rw,nosuid,nodev,relatime,stripe=32753,uhelper=udisks2)

and on B. the same gives
/dev/sdb2 on /media/usb2 type ext4 (rw,nodev,noexec,noatime,nodiratime,sync,stripe=32753)

on A. systemctl status systemd-udevd.service gives
systemd-udevd.service - udev Kernel Device Manager
   Loaded: loaded (/etc/systemd/system/systemd-udevd.service; static; vendor preset: enabled)
   Active: active (running) since Fri 2019-11-15 10:53:30 GMT; 28min ago
     Docs: man:systemd-udevd.service(8)
           man:udev(7)
 Main PID: 908 (systemd-udevd)
   Status: "Processing with 29 children at max"
    Tasks: 1
   Memory: 4.0M
   CGroup: /system.slice/systemd-udevd.service
           └─908 /lib/systemd/systemd-udevd

Nov 15 11:04:38 box.lan systemd-udevd[4820]: 
Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:04:38 box.lan systemd-udevd[4820]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:04:44 box.lan systemd-udevd[4830]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:04:44 box.lan systemd-udevd[4830]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:15:59 box.lan systemd-udevd[6385]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:15:59 box.lan systemd-udevd[6388]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:15:59 box.lan systemd-udevd[6385]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.
Nov 15 11:15:59 box.lan systemd-udevd[6388]: Process '/usr/share/usbmount/usbmount remove' failed with exit code 1.

and on B. the same command gives
status systemd-udevd.service
● systemd-udevd.service - udev Kernel Device Manager
   Loaded: loaded (/etc/systemd/system/systemd-udevd.service; static; vendor pre
   Active: active (running) since Thu 2019-11-07 16:59:30 GMT; 1 weeks 0 days ag
     Docs: man:systemd-udevd.service(8)
           man:udev(7)
 Main PID: 1039 (systemd-udevd)
   Status: "Processing with 30 children at max"
    Tasks: 1
   Memory: 3.5M
   CGroup: /system.slice/systemd-udevd.service
           └─1039 /lib/systemd/systemd-udevd

Nov 07 16:59:30 box.lan systemd[1]: Starting udev Kernel Device Manager...
Nov 07 16:59:30 box.lan systemd[1]: Started udev Kernel Device Manager.
Nov 15 12:19:06 box.lan systemd-udevd[1980]: Process '/usr/sbin/th-cmd --socket 
Nov 15 12:19:06 box.lan systemd-udevd[1975]: Process '/usr/sbin/th-cmd --socket 
Nov 15 12:19:06 box.lan systemd-udevd[1976]: Process '/usr/sbin/th-cmd --socket 
Nov 15 12:19:06 box.lan systemd-udevd[1973]: Process '/usr/sbin/th-cmd --socket

holta, 2019-11-15:
> where can I download
> Ubuntu 18.04 on x86_64 and Debian 10 on x86_64
> and do they have a desktop as well

Ubuntu 18.04 Desktop is your safest bet for now:

http://releases.ubuntu.com/18.04/
https://ubuntu.com/download/desktop

Start with the above.  Also note that experimental support for Ubuntu 19.10 is growing as we speak (especially in terms of IIAB's growing support Python 3).

jvonau, 2019-11-15:
I don't have raspbian desktop installed here, I would need access to a machine (could be arm as the underling os is the same) to work through this issue as 'lite' is working as is here.

holta, 2019-11-15:
@shanti-bhardwa wrote:
> I have tested USB mounts with IIAB 7.0 on VERSION="18.04.3 LTS (Bionic Beaver)" Desktop - "Ubuntu 18.04.3 LTS". It works as expected, with mounts going do /media/usbX within the Desktop and from Home Page and Admin Console. I will carry on testing this desktop.

jvonau, 2019-11-18:
> Nov 15 12:14:50 box PackageKit: daemon start
Nov 15 12:18:22 box vmunix: [1375229.809236] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
Nov 15 12:18:22 box vmunix: [1375229.941772] usb 1-1.2: New USB device found, idVendor=05dc, idProduct=a81d, bcdDevice=11.00
Nov 15 12:18:22 box vmunix: [1375229.941789] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 15 12:18:22 box vmunix: [1375229.941798] usb 1-1.2: Product: USB Flash Drive
Nov 15 12:18:22 box vmunix: [1375229.941808] usb 1-1.2: Manufacturer: Lexar
Nov 15 12:18:22 box vmunix: [1375229.941817] usb 1-1.2: SerialNumber: AADZGSEKNAZUKZDP
Nov 15 12:18:22 box vmunix: [1375229.943356] usb-storage 1-1.2:1.0: USB Mass Storage device detected
Nov 15 12:18:22 box vmunix: [1375229.944639] scsi host0: usb-storage 1-1.2:1.0
Nov 15 12:18:23 box mtp-probe: checking bus 1, device 4: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2" 
Nov 15 12:18:23 box mtp-probe: bus: 1, device: 4 was not an MTP device 
Nov 15 12:18:23 box vmunix: [1375230.265306] usbcore: registered new interface driver uas
Nov 15 12:18:23 box mtp-probe: checking bus 1, device 4: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2" 
Nov 15 12:18:23 box mtp-probe: bus: 1, device: 4 was not an MTP device 
Nov 15 12:18:24 box vmunix: [1375231.409716] scsi 0:0:0:0: Direct-Access     Lexar    USB Flash Drive  1100 PQ: 0 ANSI: 4
Nov 15 12:18:24 box vmunix: [1375231.412145] sd 0:0:0:0: [sda] 15659008 512-byte logical blocks: (8.02 GB/7.47 GiB)
Nov 15 12:18:24 box vmunix: [1375231.412922] sd 0:0:0:0: [sda] Write Protect is off
Nov 15 12:18:24 box vmunix: [1375231.412941] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
Nov 15 12:18:24 box vmunix: [1375231.413626] sd 0:0:0:0: [sda] No Caching mode page found
Nov 15 12:18:24 box vmunix: [1375231.413640] sd 0:0:0:0: [sda] Assuming drive cache: write through
Nov 15 12:18:24 box vmunix: [1375231.417702]  sda: sda1
Nov 15 12:18:24 box vmunix: [1375231.421394] sd 0:0:0:0: [sda] Attached SCSI removable disk
Nov 15 12:18:24 box vmunix: [1375231.442697] sd 0:0:0:0: Attached scsi generic sg0 type 0
Nov 15 12:18:25 box vmunix: [1375232.234334] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
Nov 15 12:18:25 box dbus-daemon[1786]: [session uid=1000 pid=1786] Activating via systemd: service name='org.gtk.vfs.Metadata' unit='gvfs-metadata.service' requested by ':1.7' (uid=1000 pid=1927 comm="pcmanfm --desktop --profile LXDE-pi ")
Nov 15 12:18:25 box dbus-daemon[1786]: [session uid=1000 pid=1786] Successfully activated service 'org.gtk.vfs.Metadata'
Nov 15 12:18:29 box usbmount[6371]: executing command: mount -tvfat -osync,noexec,nodev,noatime,nodiratime dev/sda1 /media/usb0
Nov 15 12:18:29 box usbmount[6371]: executing command: run-parts /etc/usbmount/mount.d
Nov 15 12:18:30 box 70-usb-library: Displaying root directory on /media/usb0.
Nov 15 12:18:30 box 70-usb-library: Found Share Directory /media/usb0.
Nov 15 12:18:30 box 70-usb-library: Creating link to /library/www/html/local_content/USB0.
Nov 15 12:23:12 box PackageKit: daemon quit

holta, 2019-12-26:
@shanti-bhardwa I'm moving this tkt to IIAB 7.2's milestone so we can try to release IIAB 7.1 promptly in coming weeks, ideally in January.

Can you provide any updates here, since 6 week ago?

shanti-bhardwa, 2019-12-27:
Yes, can be moved into 7.2

The issue is around auto mounting USB device.

IIAB mounts the USB device to /media/usbx where x can 0 to 7.

Raspbian desktop mounts  to /media/pi and IIAB then fails to pick it up.

The work around is to manually umount from /media/pi and then mount to 
/media/usbx

Shanti Bhardwa

holta, 2020-03-05:
Please see this related tkt:

#2277 http://box/usb is becoming cluttered with dead dirs USB0, USB1, USB2, USB3, USB4 (etc)

holta, 2020-03-09:
Let's monitor the situation now that PR #2282 is merged &mdash; which might possibly improve things around this race condition?

jvonau, 2020-04-06:
Might be fixed in master add this to the test matrix when evaluating #2266 with desktops installed.

holta, 2020-09-12:
@shanti-bhardwa can we close this ticket now?

Or is additional documentation-or-otherwise required at http://FAQ.IIAB.IO > "4 Can teachers display their own content?" ?

jvonau, 2021-07-20:
close, doc's as needing to disable the GUI automount

closed 2021-07-20 by holta