iiab/iiab #2041: Survey software: KoboToolbox server / ODK Collect

Birdzview, 2019-11-15:
ODK Collect is a serious survey system, open-source and totally free - no paid editions, etc
Compatible with Apache, PHP, MySQL and PostgreSQL
Will also need nginx-certbot

Highly useful for students, educators and humanitarian fieldworkers where no internet is available
Unfortuantely 
[http://support.kobotoolbox.org/en/articles/592393-installing-on-a-local-computer]
[http://support.kobotoolbox.org/en/articles/592378-install-kobotoolbox-on-your-own-servers]
[https://github.com/kobotoolbox/kobo-install]
[https://github.com/wmnnd/nginx-certbot]

holta, 2019-11-15:
See also the earlier discussion of LimeSurvey at #1355

Birdzview, 2020-02-13:
Is ODK Collect server still being considered ? Please

holta, 2020-02-13:
@Birdzview can you please explain more what you're trying to use this for, towards helping others understand and possibly build momentum around this possible approach?

e.g. what specifically are you trying to measure and for what purposes?

Also/or feel free to send this in a private email if that's easier?

Email address herein: http://internet-in-a-box.org/pages/contributing.html

Thanks Much!

Birdzview, 2020-02-13:
Most students in developing countries' schools have mobile phones that can be used to make life easier in the classroom. ODK Collect can be used efficiently to collect assessments in offline classrooms (schools or government cannot afford internet connectivity). It would reduce the workload on teachers with large student numbers per class. 
Doctors in the field have android phones but may not have internet access. ODK Collect can deliver a huge impact on information flow ( think Corona virus)
The ODK Collect mobile app needs a server - difficult to install for most people.
A ready-to-go installation can change so many lives

holta, 2020-02-13:
Likewise @ericnitschke mentions:
- https://msds.tools/2020/01/30/smartphone-data-collection-can-it-work-offline/
- https://github.com/opendatakit/briefcase

holta, 2020-02-13:
@ericnitschke also adds...
- https://odk-demo.readthedocs.io/en/latest/briefcase-vs-aggregate.html