iiab/iiab #2045: Kolibri's new 0.12.9 apt/deb package fails to install

holta, 2019-11-21:
@benjaoming we've confirmed this brand new failure to install Kolibri 0.12.9 on Debian 10.2 and also on Ubuntu.  The running code is here:

https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/main.yml#L48-L54

Do have recommendations below?  And/or should IIAB revert to Kolibri 0.12.8 for now?

> TASK [kolibri : apt install latest Kolibri .deb from https://learningequality.org/r/kolibri-deb-latest (populates /library/kolibri / migrates database, based on params set in /etc/kolibri)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "dpkg --force-confdef --force-confold -i /root/.ansible/tmp/ansible-tmp-1574352150.99-132899473790064/kolibri-deb-latestkaicwiky failed", "stderr": "insserv: script ejabberd-iiab: service ejabberd already provided!\nTraceback (last called is first):\n     kolibri_configure() in /var/lib/dpkg/info/kolibri.postinst:1\n     main() in /var/lib/dpkg/info/kolibri.postinst:456\ndpkg: error processing package kolibri (--install):\n subprocess installed post-installation script returned error exit status 1\nErrors were encountered while processing:\n kolibri\n", "stderr_lines": ["insserv: script ejabberd-iiab: service ejabberd already provided!", "Traceback (last called is first):", "     kolibri_configure() in /var/lib/dpkg/info/kolibri.postinst:1", "     main() in /var/lib/dpkg/info/kolibri.postinst:456", "dpkg: error processing package kolibri (--install):", " subprocess installed post-installation script returned error exit status 1", "Errors were encountered while processing:", " kolibri"], "stdout": "(Reading database ... 310518 files and directories currently installed.)\nPreparing to unpack .../kolibri-deb-latestkaicwiky ...\nUnpacking kolibri (0.12.9-0ubuntu1) over (0.12.8-0ubuntu1) ...\nSetting up kolibri (0.12.9-0ubuntu1) ...\nInstalling new version of config file /etc/default/kolibri ...\nInstalling new version of config file /etc/init.d/kolibri ...\nProcessing triggers for systemd (229-4ubuntu21.22) ...\nProcessing triggers for ureadahead (0.100.0-19.1) ...\n", "stdout_lines": ["(Reading database ... 310518 files and directories currently installed.)", "Preparing to unpack .../kolibri-deb-latestkaicwiky ...", "Unpacking kolibri (0.12.9-0ubuntu1) over (0.12.8-0ubuntu1) ...", "Setting up kolibri (0.12.9-0ubuntu1) ...", "Installing new version of config file /etc/default/kolibri ...", "Installing new version of config file /etc/init.d/kolibri ...", "Processing triggers for systemd (229-4ubuntu21.22) ...", "Processing triggers for ureadahead (0.100.0-19.1) ..."]}

benjaoming, 2019-11-21:
Hi @holta 

Thanks for the feedback!

This is fixed in `0.12.9-0ubuntu2` by @jredrejo - please let us know if that works for you.

We haven't yet understood and explained what has caused the failure to show up now -- but thank you for the trace-back. It may help me and @jredrejo understand why a change from September now causes issues.

holta, 2019-11-21:
> This is fixed in `0.12.9-0ubuntu2` by @jredrejo - please let us know if that works for you.

Where do I get that?

Is that a branch or ideally an apt/deb package with a URL like https://learningequality.org/r/kolibri-deb-latest ?

holta, 2019-11-21:
@benjaoming @jredrejo I'm noticing these 2 URL's but am not sure exactly how to grab an appropriate `0.12.9-0ubuntu2` apt/deb file from either?

https://launchpad.net/~learningequality/+archive/ubuntu/kolibri
https://launchpad.net/~learningequality/+archive/ubuntu/kolibri-proposed

holta, 2019-11-21:
@benjaoming can you help us get access to the problem-free Kolibri 0.12.8 apt/deb file ASAP, so that IIAB installations stop failing?

(Alongside testing/improving on 0.12.9 which we're happy to help with...but we need to make sure we're not leaving production IIAB implementers "dead in the water" all around the planet!)

jredrejo, 2019-11-21:
> @benjaoming @jredrejo I'm noticing these 2 URL's but am not sure exactly how to grab an appropriate `0.12.9-0ubuntu2` apt/deb file from either?
> 
> https://launchpad.net/~learningequality/+archive/ubuntu/kolibri
> https://launchpad.net/~learningequality/+archive/ubuntu/kolibri-proposed

Both contain the same, any of them will be valid.

holta, 2019-11-21:
> Both contain the same, any of them will be valid.

Ok.  I'm still trying to guess/get the new and old .deb URL's if you can advise how I find those...since directory listings are unfortunately turned off.

Should I try evolving this:

https://storage.googleapis.com/le-releases/downloads/kolibri/v0.12.9/kolibri_0.12.9-0ubuntu1_all.deb

...to this?

https://storage.googleapis.com/le-releases/downloads/kolibri/v0.12.9/kolibri_0.12.9-0ubuntu2_all.deb

holta, 2019-11-21:
Thanks to @jvonau who found another URL that looks promising?

http://ppa.launchpad.net/learningequality/kolibri/ubuntu/pool/main/k/kolibri-source/

So I presume we attempt the following?

http://ppa.launchpad.net/learningequality/kolibri/ubuntu/pool/main/k/kolibri-source/kolibri_0.12.9-0ubuntu2_all.deb

Can we also please get a 0.12.8 .deb URL to respect our users & implementers please?  (They did NOT sign up for QA/downtime!)

jredrejo, 2019-11-21:
> Ok. I'm still trying to guess/get the new and old .deb URL's if you can advise how I find those...since directory listings are unfortunately turned off.
>
> Should I try evolving this: [...]

No, the latest does not exist. Official releases for Debian packages are in Launchpad:
https://launchpad.net/~learningequality/+archive/ubuntu/kolibri/+files/kolibri_0.12.9-0ubuntu2_all.deb

Don't use `kolibri_0.12.9-0ubuntu1_all.deb` because that's the one with the bug.

jredrejo, 2019-11-21:
> Thanks to @jvonau who found another URL that looks promising?
>
> http://ppa.launchpad.net/learningequality/kolibri/ubuntu/pool/main/k/kolibri-source/
>
> So I presume we attempt the following? [...]

It's the same we have been mentioning in previous messages. As far as you install `kolibri_0.12.9-0ubuntu2_all_deb` and not `...._0.12.9-0ubuntu1_all.deb` you are safe.

> Can we also please get a 0.12.8 .deb URL to respect our users & implementers please? (They did NOT sign up for QA/downtime!)

Ubuntu Launchpad removes (or hides, I don't know) previous versions after new ones are uploaded. To get 0.12.8 you should go to https://github.com/learningequality/kolibri/releases/download/v0.12.8/kolibri_0.12.8-0ubuntu1_all.deb

holta, 2019-11-21:
> https://launchpad.net/~learningequality/+archive/ubuntu/kolibri/+files/kolibri_0.12.9-0ubuntu2_all.deb

The above installed correctly on Ubuntu 16.04 and we will continue testing on other OS's.

_Will this likely be pushed to become 0.12.10-or-similar in come days?_

> Ubuntu Launchpad removes (or hides, I don't know) previous versions after new ones are uploaded. To get 0.12.8 you should go to https://github.com/learningequality/kolibri/releases/download/v0.12.8/kolibri_0.12.8-0ubuntu1_all.deb

Thanks!  While directory listings are being blocked above, I now see that these URL's can be uncovered by clicking on "Assets" under each version within https://github.com/learningequality/kolibri/releases

jredrejo, 2019-11-21:
> > https://launchpad.net/~learningequality/+archive/ubuntu/kolibri/+files/kolibri_0.12.9-0ubuntu2_all.deb
> 
> The above installed correctly on Ubuntu 16.04 and we will continue testing on other OS's.
> 
> _Will this likely be pushed to become 0.12.10-or-similar in come days?_
> 

Yes, there will likeley be a 0.13.0 in mid December, with a lot of changes and new features

> Thanks! While directory listings are being blocked above, I now see that these URL's can be uncovered by clicking on "Assets" under each version within [...]

Right, I have double checked https://github.com/learningequality/kolibri/releases/edit/v0.12.9 has the good one.

jvonau, 2019-11-21:
think the real issue is ['latest'](https://learningequality.org/r/kolibri-deb-latest) is not pointing to kolibri_0.12.9-0ubuntu2_all.deb where the fix is..

holta, 2019-11-21:
> think the real issue is ['latest'](https://learningequality.org/r/kolibri-deb-latest) is not pointing to kolibri_0.12.9-0ubuntu2_all.deb where the fix is..

@benjaoming indicates they're trying to understand & QA the situation first, before immediately publishing to all:

> We haven't yet understood and explained what has caused the failure to show up now -- but thank you for the trace-back. It may help me and @jredrejo understand why a change from September now causes issues.

holta, 2019-11-21:
@benjaoming @jredrejo can you or someone or something possible notify me when https://learningequality.org/r/kolibri-deb-latest changes to a new underlying URL?

e.g. in the coming hours/days?

(Or in general, if that's miraculously possible, with an email or whatev?)

jredrejo, 2019-11-21:
> think the real issue is ['latest'](https://learningequality.org/r/kolibri-deb-latest) is not pointing to kolibri_0.12.9-0ubuntu2_all.deb where the fix is..

It is now, your messages came in the middle of the platform link updates. Sorry for the inconveniences. We're adding more controls to avoid this in the future, but as you probably know well, no software is free of bugs and humans make mistakes.

jredrejo, 2019-11-21:
> @benjaoming @jredrejo can you or someone or something possibly notify me when https://learningequality.org/r/kolibri-deb-latest changes to a new underlying URL?
> 
> e.g. in the coming hours/days?
> 
> (Or in general, if that's miraculously possible, with an email or whatev?)

@holta even if I make the purpose of notifying you I will probably forget it as the final date is not fixed in the calendar.  Our #kolibri-dev channel in Slack is public: https://app.slack.com/client/T0KT5DC58/CB37UM23A , if you are there you can see in real time all the interactions and preparations for the release.
Also, in the `kolibri-proposed` ppa you can see appearing the beta versions , https://launchpad.net/~learningequality/+archive/ubuntu/kolibri-proposed 

If you have better ideas to make it easier, they are welcome.

benjaoming, 2019-11-22:
@holta @jvonau 

I'm wondering if the Ansible jobs could/should be updated to make Kolibri added as a PPA instead of downloading the .deb.

Consider basing IIAB on the [kolibri](https://launchpad.net/~learningequality/+archive/ubuntu/kolibri) stable PPA because...

1) ...it is easier for end-users to update from and keep automated updates running
2) ...DPKG does have a few mechanisms for pinning package versions, albeit not so great, this can help you if you need a break/hotfix in the future
3) ...always installs the latest version of Kolibri and your upstream Kolibri maintainers "guarantee" to not break any upgrades.
4) The PPA also includes `kolibri-server` which you should perhaps be switching to quite soon?

LMK if you would like to discuss this. Otherwise, I'd just say: Give it a go :+1:

holta, 2019-11-22:
Thanks @benjaoming @jredrejo @jvonau for the insight, and should the PPA approach be necessary, Ansible very recently introduced the ability to add/remove APT repositories (to help us all keep /etc/apt/source* clean?) with PPA support, as seen in the examples herein: https://docs.ansible.com/ansible/latest/modules/apt_repository_module.html

benjaoming, 2019-11-22:
@holta yup, been using that too, and it works very well :)

closed 2019-12-09 by holta

holta, 2020-10-03:
Does anybody have any idea why Kolibri 0.14.3 sometimes needs to be reinstalled on IIAB on Raspberry Pi OS and Ubuntu 20.04.1 ?

The problem is that http://box/kolibri shows "502 Bad Gateway" only hours/days after it was freshly installed.

We don't have much info but an example is here: 
https://github.com/iiab/iiab/issues/1737#issuecomment-702973271

Both people found this workaround seems to do the job:

```
cd /opt/iiab/iiab
sudo ./runrole --reinstall kolibri
```

(But we'd really like to know why!)

jvonau, 2020-10-03:
502's are service not running or dns like errors, I suspect something like #1932 related, kolibri starts before the database or slow to start with channels present.

jredrejo, 2020-10-03:
> Does anybody have any idea why Kolibri 0.14.3 sometimes needs to be reinstalled on IIAB on Raspberry Pi OS and Ubuntu 20.04.1 ?
> 
> The problem is that http://box/kolibri shows "502 Bad Gateway" only hours/days after it was freshly installed.
> 
> We don't have much info but an example is here:
> [#1737 (comment)](https://github.com/iiab/iiab/issues/1737#issuecomment-702973271)
> 
> Both people found this workaround seems to do the job:
> 
> ```
> cd /opt/iiab/iiab
> sudo ./runrole --reinstall kolibri
> ```
> 
> (But we'd really like to know why!)

Hi @holta , there are lot of missing details in this question to be able to answer:

- are you using `kolibri` or `kolibri-server`
- if using just kolibri, the error is a server error, so `.kolibri/logs` should have the answer, can you attach them here?
- if using kolibri-server the problem could be also in nginx or redis, could you paste the output of executing `systemctl status kolibri-server`
- Also, if instead of reinstalling you restart the service, does it get fixed too? because the only reason I can think of reinstalling fixing it is because it restart the service

holta, 2020-10-03:
Hi @jredrejo 

> * are you using `kolibri` or `kolibri-server`

kolibri.

Is kolibri-server even viable as part of https://github.com/iiab/iiab/tree/master/roles/kolibri ?

Specifically @benjaoming helped us a ton over the years, setting up this 11-step installation of Kolibri here:
https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/install.yml

Is there a need to change anything there with Kolibri 0.14+ ?

> restart the service

Indeed, timeout during start of `kolibri.service` appears to be the problem: #2555

This proposed fix has been tested on a 1GB RAM RPi 4: PR #2556

PR #2046 (2019-12-09): Explain how to find alternative Kolibri apt / .deb installers (e.g. when 0.12.9 fails to install)
Resolves #2045 so that BIG-sized IIAB installs stop breaking (arose as a result of Kolibri 0.12.9's broken release, whose apt .deb installer was published off of https://learningequality.org/download/ sometime in the last 24h).