iiab/iiab #2056: http://box/wiki (DokuWiki) erroneously redirects to http://127.0.0.1:8090/wiki/

holta, 2019-12-09:
FYI testing occurred after installing BIG-sized http://d.iiab.io on Debian 10.2 VM 10.8.0.10

holta, 2019-12-09:
@m-anish & All:

If this is too hard to fix we should say so promptly in the coming week or so &mdash; and deprecate DokuWiki entirely.

holta, 2019-12-28:
After much discussion during our recent calls (http://minutes.iiab.io) it's now time to begin deprecating DokuWiki.  Given its lack of use, and our overall need to focus on high-quality offerings _(that are broadly tested and broadly used)_.

holta, 2019-12-29:
fyi the deprecation of DokuWiki will happen shortly, thanks to PR #2086 with a related PR.

tim-moody, 2019-12-29:
short term dokuwiki_url: /wiki can be changed to dokuwiki_url: /dokuwiki in default vars

holta, 2019-12-31:
Subject line changed as a result of merging PR #2086, which attempts to rassign DokuWiki to http://box/dokuwiki (if it was working anyway!)

holta, 2020-01-01:
> short term dokuwiki_url: /wiki can be changed to dokuwiki_url: /dokuwiki in default vars

FYI the above part is now complete, thanks to PR #2086.

holta, 2020-01-04:
Initial deprecation of DokuWiki begun, in these 2 places:
- PR #2098 "Begin deprecating DokuWiki, per #1470
- http://FAQ.IIAB.IO -> Item 9. = "What services (IIAB apps) are suggested during installation?"

m-anish, 2020-01-09:
I agree with deprecating dokuwiki

closed 2020-01-30 by holta