iiab/iiab #2057: http://box/gitea displays 404 and broken images

holta, 2019-12-09:
Regardless whether visiting http://10.8.0.10/gitea/ or http://192,168.0.162/gitea/

Presumably this regression is a result of IIAB's transitioning to NGINX?

FYI testing occurred after installing BIG-sized http://d.iiab.io on Debian 10.2 VM 10.8.0.10

Refs: PR #2042 PR #2052 #2055 #2056

tim-moody, 2019-12-30:
The page tries to load css and js from gitea/vendor, but the directory does not exist. I doubt this is nginx's fault.

holta, 2019-12-30:
> The page tries to load css and js from gitea/vendor, but the directory does not exist. I doubt this is nginx's fault.

FYI @jvonau & @aidan-fitz, can you confirm?

holta, 2020-01-02:
@aidan-fitz I'm now installing a BIG-sized IIAB on Debian 10.2 (10.8.0.30 = 162-deb102-svr-BIG-0102) to monitor if Gitea 10.2 might possibly offer any improvement w.r.t. 404's and broken images, do you know?

holta, 2020-01-02:
@aidan-fitz @jvonau @tim-moody Gitea 0.10.2 did not improve the situation (e.g. fresh install today of http://10.0.8.30/gitea on Debian 10.2).

What else could be causing all these 404 / broken image failures?

tim-moody, 2020-01-02:
go back to the last one that worked and install it in the nginx environment.

holta, 2020-01-03:
> go back to the last one that worked and install it in the nginx environment.

That didn't help.  See http://10.8.0.30/gitea using 2019-09-07's Gitea 1.9.3 as used by IIAB 7.0

What NGINX files need to be tweaked here to deal with these 404's?

closed 2020-01-03 by holta

holta, 2020-01-03:
CLARIF: this appears to now be solved, thanks to @tim-moody and PR #2095 !

PR #2091 (2020-01-02): Gitea 1.10.0 -> 1.10.2
@aidan-fitz: Might miraculously fix #2057?
PR #2095 (2020-01-03): Fix Gitea 404s fix with NGINX (#2057)
Fixes #2057