iiab/iiab #2062: Runrole regression

georgejhunt, 2019-12-11:
At a minimum, ./runrole should check vars/default_vars.yml, as well as /etc/iiab/local_vars.yml.

I cannot change the config of nginx, without running ./iiab-install --reinstall.

jvonau, 2019-12-12:
For local testing just add nginx_install: True and nginx_enabled: True to your local_vars.yml file to use runrole. 
You could also try running ./iiab-configure which would be the equivalent of using the admin-console's 'save configuration' & 'install configured options' in place of ./iiab-install --reinstall.

holta, 2019-12-12:
Whatever's decided, let's please document these practices within http://FAQ.IIAB.IO !

jvonau, 2019-12-12:
or add nginx_install: True and nginx_enabled: True to the 3 local_vars.yml files

holta, 2019-12-12:
Under discussion during today's http://minutes.iiab.io call...

@jvonau is trying to avoid the counter being reset when `./runrole` invokes one of the 9-10 aggregate roles: (stages)

https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide#ansible

@tim-moody suggests `./runrole` include a `--force` flag for people who really want to run one of those 9-10 aggregate roles / stages.

holta, 2019-12-12:
Call Summary: @jvonau is mulling over a redesign, so power users (advanced implementers etc) aren't impeded by overstrict checking.

holta, 2019-12-27:
Ref:

#1557 ./runrole options

holta, 2020-01-03:
ASIDE: "./runrole gitea" used to just work, e.g. to install a newer or older version etc!

Now one also needs to remove the line "gitea_installed: True" from /etc/iiab/iiab_state.yml

(What's the best way to bring back operator simplicity here?)

Ref: #2057

holta, 2020-01-03:
> ASIDE: "./runrole gitea" used to just work, e.g. to install a newer or older version etc!
> 
> Now one also needs to remove the line "gitea_installed: True" from /etc/iiab/iiab_state.yml

Instead of the above, one can apparently run: `./runrole --reinstall gitea`

holta, 2020-01-16:
PR #2126 "runrole - remove installed lockout" resolves half of this (overstrict checking)

_...but does [./runrole](https://github.com/iiab/iiab/blob/master/runrole) now allow for dashes-in-roles-names ?_

holta, 2020-01-23:
@jvonau explains that this is solve, as can be confirmed by running...

```
sudo su -
cd /opt/iiab/iiab
./runrole all-vars
```

closed 2020-01-23 by holta