iiab/iiab #2064: Master is broken by incorrect /etc/hosts

georgejhunt, 2019-12-15:
On clean install of lite via one liner using local-min.
A client accessing via http://box.lan gets an error when connected via wifi.
/etc/hosts:
127.0.0.1     box.lan localhost.localdomain localhost box box box.lan
::1      localhost ip6-localhost ip6-loopback
ff02::1     ip6-allnodes
ff02::2     ip6-allrouters

127.0.1.1   raspberrypi

A client is not directed to 172.18.96.1 as she should be
~                                                                               
~

holta, 2019-12-15:
Likely due to PR #2051.

georgejhunt, 2019-12-15:
uncommenting L668 in /etc/dnsmasq.conf does not fix the problem for me.
I'm stuck.

holta, 2019-12-15:
> uncommenting L668 in /etc/dnsmasq.conf does not fix the problem for me.

What is L668 ?

In any case, I've asked @jvonau to look into this as soon as he can.

georgejhunt, 2019-12-15:
I think 2065 fixes it

PR #2066 (2019-12-16): Alternative /etc/hosts.dnsmasq fix for PR #2051
Concise alternative to PR #2065, to fix #2064, arising from PR #2051.
closed 2019-12-16 by holta